micropython/micropython #15580: examples/network/https_client.py + cloudflare.com = OSError: 97

CardealRusso, 2024-08-01:
### Port, board and/or hardware

Unix

### MicroPython version

Any

### Reproduction

https://raw.githubusercontent.com/micropython/micropython/master/examples/network/https_client.py

If i use cloudflare.com in place of google.com o get OSError: 97

### Observed behaviour

. Any tips?

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-08-02:
It's because the `getaddrinfo()` is returning an IPv6 address as the first one.

To force IPv4 use:
```
ai = socket.getaddrinfo("cloudflare.com", 443, socket.AF_INET)
```

closed 2024-08-02 by CardealRusso

dpgeorge, 2024-08-03:
I've improved the examples in #15588, so they work with both IPv4 and IPv6.