micropython/micropython #15602: ESP32-S3 FH4R2 v0.2 No WiFi

PsuFan, 2024-08-05:
### Port, board and/or hardware

Custom Made ESP32-S3 Circuit Board (lol)

### MicroPython version

v1.23.0_4M_FLASH

### Reproduction

from network import WLAN
sta = WLAN(0)
sta.active(1)
sta.connect('xxx','xxx')
sta.ifconfig()

### Observed behaviour

('0.0.0.0', xxxx)

### Additional Information

Hey, I was wondering if you guys could point me at anything to try to get the WiFi connecting. I tried the ESP32-S3 generic and octal spiram builds which both had bootup warnings and wifi init failures before finding the correct 4MB flash version. This cleared the boot errors and WiFi init errors but connect does nothing. To be clear, this is a custom circuit board with a custom WiFi antenna circuit, its entirely possible I did not connect it properly. However, sta.scan() does return all of my available networks. I believe my signal is in the -75 ballpark. I have tried to enable logging in boot.py but no additional serial was printed when trying to connect. I also attempted to see if it was working with Arduino but have not been able to get it to boot. I just get the final line: entry 0x_______ and nothing after until it eventually reboots.

### Code of Conduct

Yes, I agree

createdbyjurand, 2024-08-18:
Tried on `ESP32_GENERIC_S3-FLASH_4M-20240808-v1.24.0-preview.181.g91f4a6b9e.bin` but still wifi does not connect to network :(

PsuFan, 2024-08-18:
@createdbyjurand how do you get this debug output??

createdbyjurand, 2025-07-02:
Fix: 
```python
wlan.active(True)
wlan.config(txpower=15)
```