micropython/micropython #15615: Native code decorator should be best-effort

darconeous, 2024-08-08:
### Port, board and/or hardware

MacOS X (Apple Silicon)

### MicroPython version

MicroPython v1.23.0 on 2024-05-31; darwin [GCC 4.2.1] version

### Reproduction

Try running the following code on a macOS device running Apple Silicon using a version of Micropython built for Apple Silicon:

```
@micropython.native
def test_fun():
    print("Hello, World!")
```

### Expected behaviour

I understand that native code execution is not supported on Apple Silicon, but I would expect to still be able to generally execute code even if marked up with native decorators (in order words, I would expect them to be ignored in that case).

### Observed behaviour

When trying to test this code locally on my MacBook Pro M2, all functions with the decorator `@micropython.native` end up generating the following exception:

```
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: invalid micropython decorator
>>> 
```

### Additional Information

I understand that the `@micropython.viper` decorator has different behavior and might not be a good candidate for being ignored, so we may want different behavior there. Perhaps an opt-in for allowing it to be ignored? Something like this:

```
@micropython.viper(optional=true)
def test_fun():
    print("Hello, World!")
```

This would force anyone wanting to test code like this to attest that the behavior would be the same.

### Code of Conduct

Yes, I agree

keenanjohnson, 2024-11-18:
I agree that this would be useful! Especially for things like unit tests or code that you would like to run in CI or on developer machines where performance isn't really a concern, but performance might be a concern on the actual MCU target.

peterhinch, 2024-11-18:
Suppressing the exception for Viper would achieve little because typical Viper code would fail. This is because Viper does not conform to Python semantics (e.g. pointers).

Suppressing this exception for Native is contentious. When developing cross-platform code it's useful to know that, on a given platform, it won't run as intended.

darconeous, 2024-11-18:
> Suppressing the exception for Viper would achieve little because typical Viper code would fail. This is because Viper does not conform to Python semantics (e.g. pointers).

I'm not sure what *your* typical viper code looks like, but in many (but not all) of my cases it works just fine. But in any case, you'd definitely want it to be opt-in-optional only. Really, it isn't that important, just wanted to bring it up in case it resonated with anyone.

> Suppressing this exception for Native is contentious. When developing cross-platform code it's useful to know that, on a given platform, it won't run as intended.

If that is a big concern, it could be made to only be optional explicitly, so that it always fails unless you explicitly specify that it is optional.

peterhinch, 2024-11-18:
The following workround is an option:
```py
if unsupported_arch:
    from collections import namedtuple
    bar = namedtuple('bar', ('native',))
    micropython = bar(native = lambda x : x)

@micropython.native
def foo():
    print("Hello, World.")

foo()
```

keenanjohnson, 2024-11-18:
That is an interesting workaround @peterhinch

darconeous, 2024-11-18:
> The following workround is an option:
> 
> ```python
> if unsupported_arch:
>     from collections import namedtuple
>     bar = namedtuple('bar', ('native',))
>     micropython = bar(native = lambda x : x)
> 
> @micropython.native
> def foo():
>     print("Hello, World.")
> 
> foo()
> ```

Won't that break the rest of the `micropython` import?