micropython/micropython #15623: RP2350 / Pico 2 machine.deepsleep results in crash

geerlingguy, 2024-08-08:
### Port, board and/or hardware

Pico 2 / RP2350

### MicroPython version

MicroPython v1.24.0-preview.201.g269a0e0e1 on 2024-08-09; Raspberry Pi Pico2 with RP2350

### Reproduction

1. Install latest Firmware from https://micropython.org/download/RPI_PICO2/
2. Implement `machine.deepsleep(5000)` in your code
3. Run code.

### Expected behaviour

Expected Pico 2 to enter `deepsleep` mode for 5 seconds, then continue to next bit of code.

### Observed behaviour

Upon reaching `deepsleep`, the LED on my Pico blinks when it reaches that part of the code, but Thonny's console outputs:

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (ConnectionError: read failed: [Errno 6] Device not configured).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
Unable to connect to /dev/cu.usbmodem83101: [Errno 2] could not open port /dev/cu.usbmodem83101: [Errno 2] No such file or directory: '/dev/cu.usbmodem83101'
```

Contents of `backend.log` after the Shell displays that error:

```
12:00:29.977 [MainThread] INFO    thonny: Thonny version: 4.1.4
12:00:29.977 [MainThread] INFO    thonny: cwd: /Users/jgeerling
12:00:29.977 [MainThread] INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python', '-s', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem83101', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:00:29.977 [MainThread] INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
12:00:29.977 [MainThread] INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem83101', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:00:29.977 [MainThread] INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
12:00:29.977 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
12:00:29.978 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/cu.usbmodem83101', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
```

### Additional Information

Full code example:

```python
# Lightsleep MicroPython example.
from machine import Pin
import time
led = Pin(25, Pin.OUT)

while True:
    led.toggle()
    time.sleep_ms(5000)
    led.toggle()
    machine.deepsleep(5000)
```

On the original Pico, `deepsleep` was, I believe, an alias to `lightsleep`. On Pico 2, it looks like there are deeper sleep states that may be implemented and an internal RTC that can be used to wake the chip... is it possible that either the `deepsleep` option could differ on the Pico 2, or should it just remain an alias to `lightsleep`?

Either way, the behavior here is different than `lightsleep`—which also breaks right now, see #15622.

### Code of Conduct

Yes, I agree

felixdoerre, 2024-08-10:
As implemented on the rp2 port, `deepsleep` is just a `lightsleep` followed by a `reset` (https://github.com/micropython/micropython/blob/master/ports/rp2/modmachine.c#L221). That will definitely result in the USB device getting disconnected and re-enumerated when the device is brought back up, which is in line with the errors that thonny is showing. So for me this looks as expected, and I believe the behavior is identical to that on a pico 1. I am also no sure where you see a "crash" here.

> Expected Pico 2 to enter deepsleep mode for 5 seconds, then continue to next bit of code.

This is not the expected behavior of deepsleep. As per documentation (https://docs.micropython.org/en/latest/library/machine.html):

> A deepsleep may not retain RAM or any other state of the system (for example peripherals or network interfaces). Upon wake execution is resumed from the main script, similar to a hard or power-on reset.

geerlingguy, 2024-08-10:
@felixdoerre - On the Pico 1, at least in my brief testing a couple days ago, deepsleep behaved the same and I didn't have to unplug/re-plug the Pico to bring it back up in Thonny. Just had to press 'Stop' and it would pick up the MicroPython prompt again.

DrKRR, 2024-11-11:
As commented by Mr. Geerlingguy, deepsleep() is not working with Raspberry Pi Pico 2 board. Anybody solved this problem?
As such in the Datasheet says there is Always- On Timer associated with Ring Oscillator or External Crystal Oscillator which makes Dormant mode to be used. Of course, there is no MicroPython function associated with it. How to solve ?

davidhubbard, 2024-12-11:
Seems the lowest power consumption can be achieved with 5V supplied to VSYS ([link](https://github.com/raspberrypi/pico-examples/issues/530#issuecomment-2329417549))

The lowest power measurement I found was for a lithium battery 4.11V at 150.8uA [link](https://github.com/raspberrypi/pico-examples/issues/530#issuecomment-2330842210)

matsobdev, 2024-12-12:
As far as I understand that custom example of `powman`, it restarts from flash (when BOOT0..3 set to 0), like you would just power up the board. [Here](https://forums.raspberrypi.com/viewtopic.php?p=2272490#p2272217) is idea of disabling flash and couple posts later an execution. But with flash powered down, as expected (if boots from flash to start over) it was unable to get back up again after sleep is over. Using "normal" sleep or dormant, you could revive flash sending `0xAB` command or you might just leave it powered-down since might not be required, ex. `copy_to_ram` binary. But for `powman` when BOOT0..3 = 0, you need to send it while flash is configured, so `boot_stage2`. Bootrom seems not to contain that instruction. First you need:
```
target_compile_definitions(project_name PRIVATE
	PICO_EMBED_XIP_SETUP=1
)
```
to use it first. RP2350 doesn't require it normally, but now it is the only option to turn on flash. Secondly, need to patch [boot2_w25q080.S](https://github.com/raspberrypi/pico-sdk/pull/2130/files#diff-24f58e911cac8c4890ba95cbb33a7e6ae359eff2a7efede9bb5b0b945778ef1d)