micropython/micropython #15626: asyncio: sleep_ms terminates early

peterhinch, 2024-08-09:
### Port, board and/or hardware

STM32, RP2

### MicroPython version

MicroPython v1.24.0-preview.31.g3c8089d1b on 2024-06-06; Raspberry Pi Pico with RP2040

### Reproduction

Pasted the following at the REPL:
```py
import asyncio
import time

async def do_nothing():
    while True:
        await asyncio.sleep_ms(0)

async def test():
    asyncio.create_task(do_nothing())
    for i in range(1, 55):
        start = time.ticks_us()
        await asyncio.sleep_ms(i)
        t = time.ticks_diff(time.ticks_us(), start)
        tt = 1000 * i  # Target time in us
        print(f"{i} {tt} {t} δ = {t - tt} μs")

asyncio.run(test())
```

### Expected behaviour

Expected the value of δ always to be a positive integer.

On STM32:
```
1 1000 990 δ = -10 μs
2 2000 1687 δ = -313 μs
3 3000 2845 δ = -155 μs
4 4000 3676 δ = -324 μs
5 5000 4838 δ = -162 μs
6 6000 5668 δ = -332 μs
7 7000 6834 δ = -166 μs
8 8000 7662 δ = -338 μs
9 9000 8821 δ = -179 μs
10 10000 9652 δ = -348 μs
11 11000 10817 δ = -183 μs
12 12000 11645 δ = -355 μs
13 13000 12805 δ = -195 μs
14 14000 13635 δ = -365 μs
15 15000 14803 δ = -197 μs
16 16000 15796 δ = -204 μs
17 17000 16622 δ = -378 μs
18 18000 17784 δ = -216 μs
19 19000 18788 δ = -212 μs
20 20000 19778 δ = -222 μs
21 21000 20606 δ = -394 μs
22 22000 21767 δ = -233 μs
23 23000 22773 δ = -227 μs
24 24000 23764 δ = -236 μs
25 25000 24756 δ = -244 μs
26 26000 25751 δ = -249 μs
27 27000 26757 δ = -243 μs
28 28000 27748 δ = -252 μs
29 29000 28739 δ = -261 μs
30 30000 29734 δ = -266 μs
31 31000 30741 δ = -259 μs
32 32000 31731 δ = -269 μs
33 33000 32722 δ = -278 μs
34 34000 33718 δ = -282 μs
35 35000 34728 δ = -272 μs
36 36000 35716 δ = -284 μs
37 37000 36704 δ = -296 μs
38 38000 37868 δ = -132 μs
39 39000 38713 δ = -287 μs
40 40000 39700 δ = -300 μs
41 41000 40689 δ = -311 μs
42 42000 41851 δ = -149 μs
43 43000 42697 δ = -303 μs
44 44000 43684 δ = -316 μs
45 45000 44838 δ = -162 μs
46 46000 45668 δ = -332 μs
47 47000 46683 δ = -317 μs
48 48000 47835 δ = -165 μs
49 49000 48656 δ = -344 μs
50 50000 49818 δ = -182 μs
51 51000 50667 δ = -333 μs
52 52000 51819 δ = -181 μs
53 53000 52639 δ = -361 μs
54 54000 53801 δ = -199 μs
```

### Observed behaviour

On STM32 δ is negative by up to 400μs. On RP2 it is usually positive but occasionally negative by up to 450μs.

### Additional Information

This arose in https://github.com/orgs/micropython/discussions/15594#discussioncomment-10253183.

I doubt this is a practical problem but it does seem a surprising result.

### Code of Conduct

Yes, I agree

vshymanskyy, 2024-08-10:
On ESP32-C3:
```log
MPY: soft reboot
1 1000 473 δ = -527 μs
2 2000 2014 δ = 14 μs
3 3000 3363 δ = 363 μs
4 4000 4235 δ = 235 μs
5 5000 4410 δ = -590 μs
6 6000 6223 δ = 223 μs
7 7000 6450 δ = -550 μs
8 8000 8257 δ = 257 μs
9 9000 9233 δ = 233 μs
10 10000 9345 δ = -655 μs
11 11000 10285 δ = -715 μs
12 12000 11358 δ = -642 μs
13 13000 13325 δ = 325 μs
14 14000 14268 δ = 268 μs
15 15000 14399 δ = -601 μs
16 16000 16257 δ = 257 μs
17 17000 16398 δ = -602 μs
18 18000 18228 δ = 228 μs
19 19000 18269 δ = -731 μs
20 20000 19298 δ = -702 μs
21 21000 20412 δ = -588 μs
22 22000 22235 δ = 235 μs
23 23000 22380 δ = -620 μs
24 24000 24218 δ = 218 μs
25 25000 24413 δ = -587 μs
26 26000 26220 δ = 220 μs
27 27000 26363 δ = -637 μs
28 28000 27277 δ = -723 μs
29 29000 28427 δ = -573 μs
30 30000 30263 δ = 263 μs
31 31000 31312 δ = 312 μs
32 32000 31325 δ = -675 μs
33 33000 32259 δ = -741 μs
34 34000 33333 δ = -667 μs
35 35000 34290 δ = -710 μs
36 36000 35311 δ = -689 μs
37 37000 36382 δ = -618 μs
38 38000 38196 δ = 196 μs
39 39000 38413 δ = -587 μs
40 40000 40255 δ = 255 μs
41 41000 40288 δ = -712 μs
42 42000 41366 δ = -634 μs
43 43000 43208 δ = 208 μs
44 44000 43470 δ = -530 μs
45 45000 45240 δ = 240 μs
46 46000 46237 δ = 237 μs
47 47000 46363 δ = -637 μs
48 48000 47347 δ = -653 μs
49 49000 49339 δ = 339 μs
50 50000 50307 δ = 307 μs
51 51000 51352 δ = 352 μs
52 52000 52303 δ = 303 μs
53 53000 52307 δ = -693 μs
54 54000 53353 δ = -647 μs
```

It's interesting that δ seems to stay between +-1 ms

andrewleech, 2024-08-10:
> It's interesting that δ seems to stay between +-1 ms

Ah, yes that makes sense! 

The `asyncio.sleep_ms(timeout)` function is  essentially just saying "wake me up at `ticks_add(ticks_ms(), timeout)`.

https://github.com/micropython/micropython/blob/91f4a6b9e902c066c98bda3799e01c8c6c0783ea/extmod/asyncio/core.py#L57

Of note, it's not done as `ticks_add(ticks_us(), timeout * 1000)`... it only has ms resolution.

In other words, if `sleep_ms(1)` is started half way through a ticks_ms interval, the wake-up will be in half a ms.

peterhinch, 2024-08-12:
On ideal (infinitely fast) hardware this would result in -1 < δ <= 0 ms (always early). I guess hardware limitations bias this 1ms of uncertainty towards positive values. Interestingly the Unix build somehow circumvents this restriction:
```py
1 1000 888 δ = -112 μs
2 2000 1904 δ = -96 μs
3 3000 2989 δ = -11 μs
4 4000 3986 δ = -14 μs
5 5000 4996 δ = -4 μs
6 6000 5998 δ = -2 μs
7 7000 6993 δ = -7 μs
8 8000 7997 δ = -3 μs
9 9000 8998 δ = -2 μs
10 10000 9995 δ = -5 μs
11 11000 10994 δ = -6 μs
12 12000 11996 δ = -4 μs
13 13000 12997 δ = -3 μs
14 14000 13997 δ = -3 μs
15 15000 14995 δ = -5 μs
16 16000 15997 δ = -3 μs
17 17000 16997 δ = -3 μs
18 18000 17992 δ = -8 μs
19 19000 18996 δ = -4 μs
20 20000 19997 δ = -3 μs
21 21000 20996 δ = -4 μs
22 22000 21995 δ = -5 μs
23 23000 22994 δ = -6 μs
24 24000 23996 δ = -4 μs
25 25000 24998 δ = -2 μs
26 26000 25994 δ = -6 μs
27 27000 26995 δ = -5 μs
28 28000 27994 δ = -6 μs
29 29000 28996 δ = -4 μs
30 30000 29999 δ = -1 μs
31 31000 30995 δ = -5 μs
32 32000 31997 δ = -3 μs
33 33000 32997 δ = -3 μs
34 34000 33996 δ = -4 μs
35 35000 34997 δ = -3 μs
36 36000 35996 δ = -4 μs
37 37000 36998 δ = -2 μs
38 38000 37997 δ = -3 μs
39 39000 38995 δ = -5 μs
40 40000 39995 δ = -5 μs
41 41000 40998 δ = -2 μs
42 42000 41996 δ = -4 μs
43 43000 42996 δ = -4 μs
44 44000 43996 δ = -4 μs
45 45000 44998 δ = -2 μs
46 46000 45999 δ = -1 μs
47 47000 46993 δ = -7 μs
48 48000 47997 δ = -3 μs
49 49000 48999 δ = -1 μs
50 50000 49991 δ = -9 μs
51 51000 50996 δ = -4 μs
52 52000 51995 δ = -5 μs
53 53000 52996 δ = -4 μs
54 54000 53997 δ = -3 μs
>>> 
```

washley, 2025-07-01:
Sorry if this isn't actually related to `time.sleep_ms`, but I am more generally seeing consistently early wakes with `time.sleep_ms` and `time.sleep` on RP2 (pico 2w), but `time.sleep_us` is not exhibiting this.

`sleep`:
```
>>> for _ in range(0, 10): a = time.ticks_us() ; time.sleep(0.002) ; b = time.ticks_us() ; print(b - a) ;
... 
1900
1767
1883
1868
1882
1894
1918
1885
1902
1901
```

`sleep_ms`:
```
>>> for _ in range(0, 10): a = time.ticks_us() ; time.sleep_ms(2) ; b = time.ticks_us() ; print(b - a) ;
... 
1957
1782
1917
1881
1895
1883
1903
1902
1905
1899
```

`sleep_us`:
```
>>> for _ in range(0, 10): a = time.ticks_us() ; time.sleep_us(2000) ; b = time.ticks_us() ; print(b - a) ;
... 
2035
2033
2022
2022
2020
2021
2022
2021
2022
2027
```

This is with the latest micropython release.