micropython/micropython #15627: BLE GATT descriptor set to DESCRIPTOR_READ (0x01) returns Error Code: Read Not Permitted (0x02)

Heerkog, 2024-08-09:
### Port, board and/or hardware

esp32

### MicroPython version

This issue started showing from MicroPython v1.20 onwards.
V1.23 is confirmed to still have the issue. 
V1.19 is confirmed to NOT have the issue.

### Reproduction

I'm using my [MicroPythonBLEHID library](https://github.com/Heerkog/MicroPythonBLEHID/blob/master/hid_services.py).

On line 588, among others, I define the following HID service description:
```
        self.HIDS = (
            UUID(0x1812),
            (
                (UUID(0x2A4A), F_READ),
                (UUID(0x2A4B), F_READ),
                (UUID(0x2A4C), F_WRITE),
                (UUID(0x2A4D), F_READ_NOTIFY, ((UUID(0x2908), DESCRIPTOR_READ),)),
                (UUID(0x2A4E), F_READ_WRITE),
            ),
        )
```
Where `DESCRIPTOR_READ = 0.x01`, as specified [here](https://docs.silabs.com/bluetooth/6.2.0/bluetooth-stack-api/sl-bt-gattdb-descriptor-properties).

### Observed behaviour

The value `0x01` of the report reference `UUID(0x2908)` returns a `Rcvd Error Response - Read Not Permitted, Handle: 0x0030 (Human Interface Device: Report: Report Reference)` message when intercepted with Pcap/Wireshark, causing a driver failure.

### Additional Information

When I set the descriptor to `0.x02`, the issue is resolved. However, this is the write flag, and not read, and therefore wrong. My current version of the library uses this value to 'make it work', but this is incorrect.

### Code of Conduct

Yes, I agree

dlech, 2024-08-09:
The Client Characteristic Configuration needs to be writable in order to be able to enable notifications.

Heerkog, 2024-08-09:
> The Client Characteristic Configuration needs to be writable in order to be able to enable notifications.

Thank you for your reply. Are you saying I need to do the following?

```
ATT_F_READ = 0x01
ATT_F_WRITE = 0x02
ATT_F_READ_WRITE = ATT_F_READ | ATT_F_WRITE
```

```
        self.HIDS = (
            UUID(0x1812),                                          # Human Interface Device
            (
                (UUID(0x2A4A), F_READ),                            # HID information
                (UUID(0x2A4B), F_READ),                            # HID report map
                (UUID(0x2A4C), F_WRITE),                           # HID control point
                (UUID(0x2A4D), F_READ_NOTIFY, (                    # HID report
                    (UUID(0x2908), ATT_F_READ),                    # HID reference 
                    (UUID(0x2902), ATT_F_READ_WRITE),)),           # Client Characteristic Configuration
                (UUID(0x2A4E), F_READ_WRITE),                      # HID protocol mode
            ),
        )
```

I'm unsure why defining Client Characteristic Configuration would give a read error on the HID reference though.

dlech, 2024-08-09:
Ah, I didn't notice that 0x2908 was not the UUID for CCCD. I guess the CCCD gets implemented automatically when the NOTIFY flag is set on a characteristic?

Anyway, not really relevant to the issue, sorry for the noise.

Heerkog, 2024-08-10:
Just to check, I have added CCCD and the issue persists.

Once I allow `ATT_F_WRITE`, `ATT_F_READ_WRITE`, or `ATT_F_READ_WRITENORESPONSE` on `UUID(0x2908)` HID reference, the issue is solved. However, this characteristic descriptor should be `ATT_F_READ` only.