micropython/micropython #15630: memory allocation ESP32's

kdschlosser, 2024-08-10:
### Port, board and/or hardware

ESP32-S3 N16R8

### MicroPython version

1.23.0

### Reproduction

`micropython.mem_info()`, `gc.mem_alloc()` and `gc.mem_free()` are not returning the expected values. I am not 100% sure what is going on. I see in the documentation something about using a split heap but I am not able to locate anywhere in the MicroPython code any of the `multi_heap_*` functions of the ESP-IDF being used. malloc and other stdlib memory functions are not going to allocate memory or free memory from an esp multi heap.

I don't know if this is a bug or possibly an issue with my understanding of what is going on.... The numbers are not even close to what they should be for having an MCU that has 8mb of external SPIRAM.

### Expected behaviour

For the numbers to match up properly.

### Observed behaviour

```
>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 127936, used: 126336, free: 1600, max new split: 69632
 No. of 1-blocks: 2167, 2-blocks: 365, max blk sz: 282, max free sz: 86
```

```
>>> gc.mem_alloc()
55792
```

```
>>> gc.mem_free()
93616
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

vshymanskyy, 2024-08-11:
Could not reproduce on TTGO T7S3, which uses the same module:
[image]

all works as expected. what I can suggest:
- Check the boot log to see if PSRAM was detected (probably it wasn't in your case)
- Try both OCTAL SPI and a reqular build
- Try using a different module with the same chip.

I think you should really start posting such questions to https://github.com/orgs/micropython/discussions, the community will help. And if it turns out to be a bug, the discussion can easily be converted to a bug report. Cheers!

vshymanskyy, 2024-08-11:
Same on MicroPython v1.23.0 on 2024-06-02; Generic ESP32S3 module with Octal-SPIRAM.
**But, if I use the non-OCTAL version of MPY, the PSRAM is not detected iand it gives me:
```log
>>> gc.mem_free()
248000
```

kdschlosser, 2024-08-11:
I just actually got it sorted out last night. It was a matter of the octal spiram,.

closed 2024-08-11 by kdschlosser