micropython/micropython #15651: Portenta C33 USB regression.

andrewleech, 2024-08-13:
### Port, board and/or hardware

Renesas port, Portenta C33

### MicroPython version

Micropython nightly builds after v1.23

### Reproduction

Use DFU to load micropython firmware.

### Expected behaviour

USB serial / repl should be available.

### Observed behaviour

USB does not enumerate.

### Additional Information

As reported in https://github.com/micropython/micropython/pull/14462#issuecomment-2255280313

### Code of Conduct

Yes, I agree

andrewleech, 2024-08-13:
Hi @iabdalkader I wanted to capture this regression here in its own separate issue for tracking, I feel it really should be resolved one way or another before the next release. 

I don't personally feel confident that just a crystal/clock issue could cause the fault in combination with https://github.com/micropython/micropython/commit/2d33071b171e2de4a3e04f48d1f19b5bbe87c663 
I think it's more likely a bug I've introduced in the USB HS vs FS configuration.

One of the primary goals of my set of stdio consolidate commits was to enable the runtime USB `machine. USBDevice` support on more ports, so I'm loath to suggest just reverting that commit as that blocks progressing this on the renesas port. 

I don't have any renesas high speed devices to test & fix with however.

Did you get any feedback on whether the newer builds do work on production boards though?

iabdalkader, 2024-08-13:
> I don't personally feel confident that just a crystal/clock issue could cause the fault in combination with [2d33071](https://github.com/micropython/micropython/commit/2d33071b171e2de4a3e04f48d1f19b5bbe87c663)
> I think it's more likely a bug I've introduced in the USB HS vs FS configuration.

I think it really might just be the bad crystal there, it also had something to do with USB maybe it derived some clock from it. I wouldn't revert anything until we can test it.

> Did you get any feedback on whether the newer builds do work on production boards though?

Not yet, everyone is on vacation including me and I don't have a second C33 to test with, but will check again.

iabdalkader, 2024-08-13:
Everything is working on a production board, USB and WiFi:
```
WiFi Connected  ('192.168.86.244', '255.255.255.0', '192.168.86.1', '192.168.86.1')
Year:2024 Month:8 Day:13 Time: 9:19:38
```

iabdalkader, 2024-08-13:
I can test this when I get back in two weeks: https://github.com/micropython/micropython/pull/15550

andrewleech, 2024-08-13:
That all sounds great thanks! I still find it curious it worked before the change on your board but not after, but I'll take any positive test results! 

Thanks for the feedback and enjoy the rest of your time off!

closed 2024-08-27 by andrewleech