micropython/micropython #15654: Build embed with C modules isn't working

volodymyr-zvarun-globallogic, 2024-08-13:
### Port, board and/or hardware

embed

### MicroPython version

MicroPython v1.22.2 on 2024-08-13; linux [GCC 4.2.1] version

### Reproduction

1. Modify `examples/embedding/main.c` by adding `import cexample` into second example
2. From `examples/embedding` 
```
run make -f micropython_embed.mk USER_C_MODULES=<micropython_path>/examples/usercmodule
```

### Expected behaviour

Expected to start without errors

### Observed behaviour

Traceback (most recent call last):
  File "<stdin>", in <module>
ImportError: module not found

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

jonnor, 2024-09-01:
This could be a failure either at build time or a module loading time. Do you see the module being built?
Do a clean and check the build log with make V=1

Gadgetoid, 2024-09-02:
If I try to build using the same instructions I get the error:

```
../usercmodule/cppexample/examplemodule.h:2:10: fatal error: 'py/runtime.h' file not found
#include "py/runtime.h"
```

If the USER_C_MODULES path is invalid it will just fail silently. Fwiw I was using:

```
make -f micropython_embed.mk USER_C_MODULES=../usercmodule
```

Gadgetoid, 2024-09-02:
With these changes I got C/C++ modules working... in a limited capacity. It's clear the embed port makes no effort to support these whatsoever since presumably you'd handle C functionality in your application code anyway. But since ulab is a (very good) C module, and might conceivably be wanted in MicroPython's embed port... this could make sense.

(ﾉ◕ヮ◕)ﾉ*:・ﾟ✧. https://github.com/micropython/micropython/compare/master...pimoroni:micropython:feature/embed_c_modules

dpgeorge, 2024-09-03:
The embed port is very new and still needs work to add more features.  See eg #11430.