micropython/micropython #15663: RPI PICO WH - RP2040 WIFI CHIP NEVER ENABLES

realdaveblanch, 2024-08-15:
### Port, board and/or hardware

rp2040

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

### Reproduction

from machine import Pin
from time import sleep
import network
import time

def initialize_wifi():
    wlan = network.WLAN(network.STA_IF)
    
    if not wlan.active():
        print("Activando Wi-Fi...")
        wlan.active(True)
        time.sleep(2)  # Espera para asegurarse de que se activa

    # Verificar el estado después de intentar activar
    if wlan.active():
        print("Wi-Fi está activo.")
    else:
        print("Wi-Fi no está activo. Verifica el hardware o el firmware.")
    
    return wlan

def connect_wifi(wlan, ssid, password):
    if not wlan.active():
        print("Wi-Fi no está activo, activando...")
        wlan.active(True)
        time.sleep(2)
    
    wlan.connect(ssid, password)
    print(f'Conectando a {ssid}...')
    
    start_time = time.time()
    while time.time() - start_time < 20:  # Esperar hasta 20 segundos
        if wlan.isconnected():
            print("Conectado.")
            print("Detalles de la red:", wlan.ifconfig())
            return True
        time.sleep(1)
    
    print("No se pudo conectar.")
    return False

# Configuración de pines
pines = [Pin(13, Pin.OUT), Pin(14, Pin.OUT), Pin(12, Pin.OUT), Pin(15, Pin.OUT)]

def encender_apagar_pines():
    # Encender pines secuencialmente
    for pin in pines:
        print(pin)
        pin.value(1)  # Encender el pin
        sleep(1)      # Esperar 1 segundo 
        pin.value(0)  # Apagar el pin
        sleep(1)      # Esperar 1 segundo

try:
    encender_apagar_pines()
    # Inicializar y probar Wi-Fi
    wlan = initialize_wifi()
    result = connect_wifi(wlan, 'ADAMO-24C0', 'XMSD6O2FN2E6O6')
    print("Resultado de la conexión:", result)
except KeyboardInterrupt:
    # Apagar todos los pines al terminar
    for pin in pines:
        pin.value(0)
    print("Programa terminado")

### Expected behaviour

It never enables wifi connection, even tho i tried different rpi pico w, and the same micropython version 1.23.0.

wlan.active always says False, it used to work.....

doesn't even try to enable the wifi chip

### Observed behaviour

The error is always the same, EPERM, and before trying to connect wlan.active() returns false every single time.
The code was tested on 3 rpi pico wh and no way to enable the wifi chip

Traceback (most recent call last):
  File "<stdin>", line 58, in <module>
  File "<stdin>", line 28, in connect_wifi
OSError: [Errno 1] EPERM

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

realdaveblanch, 2024-08-15:
after nuking all of them and reinstalling 1.23.0 the chip enables but doesn't make requests... not even ntp

closed 2024-08-15 by realdaveblanch

dpgeorge, 2024-08-15:
Firmware 1.23.0 definitely works well on Pico W.

If you are having problems then it's most likely either a hardware issue, or your local network.

realdaveblanch, 2024-08-15:
@dpgeorge memory issues... oh my god...., now it works