micropython/micropython #15670: Binary operations on undefined variables crash the native emitter.

agatti, 2024-08-16:
### Port, board and/or hardware

unix

### MicroPython version

MicroPython v1.24.0-preview.206.ge9814e987.dirty on 2024-08-16; linux [GCC 14.2.1] version

### Reproduction

Run the following code:

```python
@micropython.native
def f():
    a += 0  # Or anything else, really
f()
```

### Expected behaviour

A `NameError` exception should be raised.  (CPython raises `NameError: name 'a' is not defined`)

### Observed behaviour

This is the backtrace of the crash when running the interpreter under GDB:

```
Program received signal SIGSEGV, Segmentation fault.
mp_obj_get_type (o_in=o_in@entry=0x0) at ../../py/obj.c:61
61              return o->type;
(gdb) bt
#0  mp_obj_get_type (o_in=o_in@entry=0x0) at ../../py/obj.c:61
#1  0x0000555555591333 in mp_binary_op (op=MP_BINARY_OP_INPLACE_ADD, lhs=0x0, rhs=0x1) at ../../py/runtime.c:630
#2  0x00007ffff7fc0053 in ?? ()
#3  0x00007ffff7a06440 in ?? ()
#4  0x00007ffff7fc006d in ?? ()
#5  0x00007fffffff9610 in ?? ()
#6  0x0000000000000003 in ?? ()
#7  0x00007fffffff9760 in ?? ()
#8  0x0000000000000000 in ?? ()
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

mendenm, 2024-12-30:
Is anyone working on this bug?  I can confirm it still exists:

```
@micropython.native
def rgb_hsv():
    a = 120 - a    
x = rgb_hsv()
```

crashes.  It took me a long time to find the typo in my code that was hard-hanging the machine.  In the real code, `a = 120 - a` was supposed to be `a = 120 - b`.

Version: MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico with RP2040

agatti, 2024-12-30:
I'm still (slowly) working on a more general fix for this issue but it will take a while as it changes a quite a bit of things inside the emitter/compiler code.

In the meantime your best bet is to double down on static analysis to catch these issues at "compile" time.

peterhinch, 2024-12-30:
Or test the code before adding the decorator.

mendenm, 2024-12-30:
@peterhinch Test the code?  Never!  ;-)