micropython/micropython #15673: Wrong size reported while the memory allocation failed.

xlla, 2024-08-17:
### Port, board and/or hardware

esp32

### MicroPython version

MicroPython v1.20.0-729-g82bf0aa19-dirty on 2024-08-17; M5Coreink with ESP32-PICO-D4

### Reproduction

while execute lv display driver create function
`self.disp_drv = lv.display_create(self.width, self.height)`

width and height is same , equal to 200.

### Expected behaviour

expected to print correct allocating bytes.

### Observed behaviour

`MemoryError: memory allocation failed, allocating 1073659464 bytes`

### Additional Information

As I search the error message inside repo, I found the exception was throw from m_malloc_fail function, at there, size_t variable was cast to uint, maybe uint is smaller than size_t, so it is overflow.
can we use %zu for size_t instead of cast to uint?

### Code of Conduct

Yes, I agree

dpgeorge, 2024-08-18:
> while execute lv display driver create function `self.disp_drv = lv.display_create(self.width, self.height)`
> 
> width and height is same , equal to 200.

This function is not defined in this repository.  Can you point to where it is defined?

> maybe uint is smaller than size_t, so it is overflow.
> can we use %zu for size_t instead of cast to uint?

MicroPython has it's own `printf` routine which does not support `%zu` formatting.  So we can't use this.

But I don't think that's the issue.  It's unlikely that the function is trying to allocate such a large number of bytes, if the width and height are only 200.

The number 1073659464 = 0x3ffebe48 could be a pointer, or could be a small-int value that was incorrectly converted.

Carglglz, 2024-08-18:
@dpgeorge 

this is from MicroPython LVGL bindings and I think it is related to root pointers not being handled properly on soft reset see https://github.com/lvgl/lv_binding_micropython/issues/343

dpgeorge, 2024-08-19:
OK, thanks for the info @Carglglz .  I think we can close this issue then.

xlla, 2024-08-23:
hi @Carglglz ,  after apply [fix(init/deinit): Properly init/deinit lvgl module](https://github.com/Carglglz/lv_binding_micropython/commit/b0d87e9676bb2afeae0c3b23b426b43eeecd3202) , I got this compile error,
<pre>
In file included from /Volumes/Data/git/lv_micropython/ports/esp32/build-M5STACK_COREINK/lv_espidf.c:20:
/Volumes/Data/git/lv_micropython/ports/esp32/build-M5STACK_COREINK/lv_espidf.c:20511:50: error: 'lvgl_mod___init___obj' undeclared here (not in a function); did you mean 'mp_mdns_init_mpobj'?
     { MP_ROM_QSTR(MP_QSTR___init__), MP_ROM_PTR(&lvgl_mod___init___obj) },
                                                  ^~~~~~~~~~~~~~~~~~~~~
/Volumes/Data/git/lv_micropython/py/obj.h:348:24: note: in definition of macro 'MP_ROM_PTR'
 #define MP_ROM_PTR(p) (p)
                        ^
/Volumes/Data/git/lv_micropython/ports/esp32/build-M5STACK_COREINK/lv_espidf.c:20512:49: error: 'lvgl_mod___del___obj' undeclared here (not in a function)
     { MP_ROM_QSTR(MP_QSTR___del__), MP_ROM_PTR(&lvgl_mod___del___obj) },
                                                 ^~~~~~~~~~~~~~~~~~~~
/Volumes/Data/git/lv_micropython/py/obj.h:348:24: note: in definition of macro 'MP_ROM_PTR'
 #define MP_ROM_PTR(p) (p)
                        ^
[1473/1828] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/__/lv_mp.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [deploy] Error 2
</pre>

should we put these export statement inside macro LV_OBJ_T judgement ?
<pre>
#ifdef LV_OBJ_T
    { MP_ROM_QSTR(MP_QSTR___init__), MP_ROM_PTR(&lvgl_mod___init___obj) },
    { MP_ROM_QSTR(MP_QSTR___del__), MP_ROM_PTR(&lvgl_mod___del___obj) },
#endif // LV_OBJ_T
</pre>

closed 2024-08-25 by xlla