micropython/micropython #15674: SSL Context behaviour various between Raspberry Pi Pico and ESP32 Board

vasili1, 2024-08-17:
### Port, board and/or hardware

esp32 port, RPI_PICO_W port

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Generic ESP32 module with ESP32
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

### Reproduction

This issue is related to certificates generated by AWS IoT. You will need to create an AWS Account to reproduce this exact issue. I'm guessing the issue comes up with other certificates but that is my use case currently.

1. Copy and Paste the code down below in your main.py file
2. In the aws console search for IoT Core
3. Create a new thing: [AWS IoT Core](https://eu-central-1.console.aws.amazon.com/iot/home?region=eu-central-1#/create/provisioning). Choose "Create single thing". On the first step leave everything as is. On the second step specify a name (doesn't really matter which one). Leave everything as is on the third step.
4. A pop up will present you with the generated certificates. Download the Device certificate (first one) and the Private key file (third one).
5. Convert these certificates in DER format with openssl commands down below. You will need to adjust the input file names
6. Upload the certificates to the device
7. Run main.py on an esp32 based board and on a pi pico w

main.py
`import tls

key = open("private.key.der", "rb").read()
cert = open("certificate.der.crt", "rb").read()

context = tls.SSLContext(tls.PROTOCOL_TLS_CLIENT)
context.load_cert_chain(cert, key)`

openssl
`openssl pkcs8 -topk8 -nocrypt -in <...-private.pem.key> -inform PEM -out private.key.der -outform DER
openssl x509 -outform der -in <...-certificate.pem.crt> -out certificate.der.crt`

### Expected behaviour

I expect the main.py to accept the certificates on both platforms (esp32 and pi pico w).

### Observed behaviour

The same certificates and main.py with the above mentioned micropython versions produce different behaviour. The esp32 runs the code without any issues. The pico w on the other hand throws an error `ValueError: invalid key`.

### Additional Information

This is only a section of my code from my use-case. I use these certificate to connect to AWS IoT over MQTT at the end of the day. The actual connection to MQTT as well as receiving and sending messages works on esp32 without issues. The pico w board in the other hand doesn't get past this snippet of code.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-08-18:
Thanks for the very clear bug report.

This could be the same as issue #14371, which was fixed in #14385.

Are you able to try latest firmware for the Pico W and see if the issue is still there?

If the issue persists then it's likely a difference in the mbedtls configuration between the esp32 and rp2 ports.

vasili1, 2024-08-18:
Thank you very much for your fast response!

I can confirm with version `MicroPython v1.24.0-preview.206.ge9814e987 on 2024-08-18; Raspberry Pi Pico W with RP2040` everything works as expected.

closed 2024-08-18 by vasili1

dpgeorge, 2024-08-19:
Thanks for confirming the fix.

I just wanted to point out to the wider audience that this was a great example of a clear, concise and reproducible bug report.  When submitting a report it's easy to overlook and omit simple steps that you may think are inconsequential, but may in the end make all the difference to being able to solve the problem.

In this case it was the `openssl pkcs8 -topk8 ... ` command that caught my attention and linked the problem to #14371.