micropython/micropython #15687: PICO2 internal temperature sensor not accessible at ADC(4)

CGsama, 2024-08-21:
### Port, board and/or hardware

PICO2

### MicroPython version

RPI_PICO2-20240809-v1.24.0-preview.201.g269a0e0e1

### Reproduction

from machine import ADC
temp_sensor = ADC(4)

### Expected behaviour

able to read ADC

### Observed behaviour

ValueError: Pin doesn't have ADC capabilities

### Additional Information

adc channel defined at https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf  section 12.4.6.
tried both 4 and 8 no luck

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-09-11:
This, I think, has since been fixed by https://github.com/micropython/micropython/pull/15619/commits/78b1bf3f84f4716d659b69b97edc61e0fd43a805

In all cases the constant `ADC.CORE_TEMP` should be the right value.

leiwangphd, 2024-09-22:
> This, I think, has since been fixed by [78b1bf3](https://github.com/micropython/micropython/commit/78b1bf3f84f4716d659b69b97edc61e0fd43a805)
> 
> In all cases the constant `ADC.CORE_TEMP` should be the right value.

Hi @Gadgetoid, The problem is still there. "ADC.CORE_TEMP" always returns a constant value of 4. For instance, 

from machine import ADC
print(ADC.CORE_TEMP) # return 4
print(ADC(0).CORE_TEMP) # return 4
print(ADC(4).CORE_TEMP) # error shows "Pin doesn't have ADC capabilities"

leiwangphd, 2024-09-22:
> ### Port, board and/or hardware
> PICO2
>
> ### MicroPython version
> RPI_PICO2-20240809-v1.24.0-preview.201.g269a0e0e1
>
> ### Reproduction
> from machine import ADC temp_sensor = ADC(4)
>
> ### Expected [...]

Hi @CGsama, 
I use RP2350A and receive the same issue as yours. Hopefully, Raspberry PI dev team has captured this thread and escalated the bug fix.

Gadgetoid, 2024-09-22:
This was fixed by https://github.com/micropython/micropython/commit/78b1bf3f84f4716d659b69b97edc61e0fd43a805

The correct form is:

```python
ADC(ADC.CORE_TEMP)
```

Since `ADC.CORE_TEMP` is the *channel* from which to read core temp (this varies between RP2350 and RP2350B), and not the actual temperature. Eg:

```python
>>> from machine import ADC
>>> ADC(4).read_u16()
14227
```

If you need this fix then you'll have to compile your own preview build, since the one available at https://micropython.org/download/RPI_PICO2/ is *old*.

Alternatively you can grab a somewhat newer build from https://github.com/pimoroni/pimoroni-pico-rp2350/releases/tag/v0.0.7  (you'll need [pico2_rp2350-v0.0.7-pimoroni-micropython.uf2](https://github.com/pimoroni/pimoroni-pico-rp2350/releases/download/v0.0.7/pico2_rp2350-v0.0.7-pimoroni-micropython.uf2)) or hope @dpgeorge uploads a newer preview 👀

dpgeorge, 2024-09-23:
Yes, this should be fixed on the latest rp2350 PR.

I have uploaded new firmware to https://micropython.org/download/RPI_PICO2/ which should contain this fix.

leiwangphd, 2024-09-23:
@Gadgetoid - Thank you for the detailed info

@dpgeorge - That is awesome. Thank you for the quick fix!!

jonnor, 2025-03-03:
Appears to have been fixed.

closed 2025-03-03 by jonnor