micropython/micropython #15701: ESP32_GENERIC_C3 binary does not run if compiled with IDF 5.0.4

ricksorensen, 2024-08-23:
### Port, board and/or hardware

ESP32C3, GENERIC_ESP32_C3, SEEED XIAO ESP32C3  IDF 5.0.4

### MicroPython version

[v1.24.0-preview.224.g6c3dc0c0b (2024-08-22) .bin](https://micropython.org/resources/firmware/ESP32_GENERIC_C3-20240822-v1.24.0-preview.224.g6c3dc0c0b.bin)

Started with branch https://github.com/micropython/micropython/commit/0a11832cddbeed57bf0267afa0c833584817312d

### Reproduction

(Presumes use of IDF 5.0.4 for working binary downloads:
```
$strings ~/Downloads/ESP32_GENERIC_C3-20240820-v1.24.0-preview.213.g76dd4facb.bin | grep IDF
[0;32mI (%lu) %s: ESP-IDF:          %s
IDF version : v5.0.4
MicroPython-1.24.0-preview-riscv-IDFv5.0.4-with-newlib4.1.0
```
)
1. For ESP32_GENERIC_C3, Download latest preview version [v1.24.0-preview.224.g6c3dc0c0b (2024-08-22) .bin](https://micropython.org/resources/firmware/ESP32_GENERIC_C3-20240822-v1.24.0-preview.224.g6c3dc0c0b.bin)
2. Write to board:
`esptool --chip esp32c3 --port /dev/ttyACM0 --baud 460800 write_flash -z 0x0 ~/Downloads/ESP32_GENERIC_C3-20240820-v1.24.0-preview.213.g76dd4facb.bin`
3.  Reset board
4. `mpremote` 

### Expected behaviour

Expected `>>>` prompt after carriage return

### Observed behaviour

No response 

### Additional Information

The first revision that fails with IDV 5.0.4 is:
https://github.com/micropython/micropython/commit/0a11832cddbeed57bf0267afa0c833584817312d `esp32: Use the ESP-IDF default esp_console config for ESP32-C3 `

If I use IDF 5.2.2 to compile the most recent development versions the  binary loads and runs successfully.  

### Code of Conduct

Yes, I agree

vshymanskyy, 2024-08-23:
Probably, we can just drop IDF 5.0.4 support

ricksorensen, 2024-08-23:
That is a possible decision - but it would be good to understand what the breaking change is.

One of the difficulties with the Espressif eco-system is the breaking changes and resource usage from the tool set/apis.

agatti, 2024-08-24:
I'm not sure it's an ESP-IDF issue.  I routinely build and run the latest master using 5.0.4 on various no-name boards I have around but I cannot seem to reproduce the issue with the download link you provided either:

```bash
> esptool.py --chip esp32c3 --port /dev/ttyUSB0 --baud 460800 write_flash -z 0x0 ~/Downloads/ESP32_GENERIC_C3-20240822-v1.24.0-preview.224.g6c3dc0c0b.bin
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: ec:da:3b:ab:0f:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0019afff...
Compressed 1681600 bytes to 1016141...
Wrote 1681600 bytes (1016141 compressed) at 0x00000000 in 28.4 seconds (effective 473.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

> ../../tools/mpremote/mpremote.py connect /dev/ttyUSB0
Connected to MicroPython at /dev/ttyUSB0
Use Ctrl-] or Ctrl-x to exit this shell
# I PRESS ENTER HERE
>>>
```

robert-hh, 2024-08-24:
It seems commit:
`0a11832cd esp32: Use the ESP-IDF default esp_console config for ESP32-C3.`
which breaks the v5.04 build. With the commit before that the build works, at that commit it fails. Note that after deploying a new build I  have to manually reset the board (XIAO ESP32C3).

robert-hh, 2024-08-24:
@agatti  You board is different to mine, in that your board uses /dev/ttyUSBx for communication, while the boarfd used by me and @ricksorensen uses /dev/ttyACM0. There were dedicated ESP32C3 variants for both communication types, which seem to be gone.

agatti, 2024-08-24:
@robert-hh On my boards /dev/ttyACM0 shows up only if I attach the USB JTAG serial port exposed on pins 18 and 19:

```
Bus 001 Device 033: ID 1a86:7523 QinHeng Electronics CH340 serial converter  # USB0
Bus 001 Device 035: ID 303a:1001 Espressif USB JTAG/serial debug unit        # ACM0
```

Does your board have a CH340 as well?

robert-hh, 2024-08-24:
At my boards /dev/ttyACM0 shows up, but there is no REPL prompt. And no, there is no CH340. USB is directly connected to the ESP32C3. I get the same device with lsusb with both commit.
`ID 303a:1001 Espressif USB JTAG/serial debug unit`
But before commit 0a11832cd there was the REPL stream at that device.

agatti, 2024-08-24:
I see...  Then I wonder why my boards have an extra USB/UART adapter on board when it's not needed at all, I thought those no-name board manufacturers tried to make them as cheap as possible!

robert-hh, 2024-08-24:
One can indeed have th impression that these vendors do not understand the boards they build. I have two other ESPC3 boards with CH340. But these miss the two transistors for the reset/boot logic. The pads are there, but no transistors and CB shortened.

agatti, 2024-08-24:
Hrm, apparently even Espressif themselves use an external USB to UART adapter in their devkits. https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-devkitm-1/user_guide.html#pin-layout shows they use a CP2102 in there for example.

ricksorensen, 2024-08-24:
I wonder what changed in IDF 5.2.2 to allow that compile to build and run with the same updated board definitions. Is there another system variable that changed?

dpgeorge, 2024-08-27:
I think the issue is that `ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED` is not an option in IDF 5.0.4, whereas it is in IDF 5.2.2.

projectgus, 2024-08-27:
Apologies folks, I could have sworn I verified `ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED` was present in IDF v5.0.4 but clearly I did not check correctly! A PR with a fix is linked above.

> Probably, we can just drop IDF 5.0.4 support

The plan after merging #15523 was to wait a little while and drop support for versions before v5.2.2 if nothing came up. @dpgeorge has just switched nightlies to v5.2.2 now (as of the next build), so it's a bit premature right now but it is the plan unless a serious issue shows up in v5.2.2.

commit 6ad6297e (2024-09-03): esp32: Fix ESP32-C3 USB serial/jtag peripheral pre-IDF 5.1.

Regression in 0a11832cd in IDF 5.0.x where macro
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED is not defined.

With this patch, ESP32-S3 still USB Serial/JTAG incorrectly (now on all
ESP-IDF versions).

Closes #15701

This work was funded through GitHub Sponsors.
PR #15727 (2024-09-03): esp32: Fix ESP32-C3 usb serial/jtag on IDF v5.0.4, clean up native USB-CDC
### Summary

* Closes #15701 which is a regression introduced with #15523 when using IDF v5.0.x.
* Fixes a lower impact regression from the same PR where on ESP-IDF >= 5.1 the ESP32-S3 would come up using the USB Serial/JTAG interface instead of native USB. The obvious fixes for the first regression all made this second regression appear on all ESP-IDF versions.
* Overall the logic for how the USB function is assigned were a bit hard to follow, and depended on some default ESP-IDF config selections.
* There are now two  [... cut]
closed 2024-09-03 by dpgeorge