micropython/micropython #15706: ESP32-S3 sdkconfig custom PID no longer working after commit 0a11832cd (...esp_console config for ESP32-C3)

stephanelsmith, 2024-08-24:
### Port, board and/or hardware

esp32, boards: UM_TINYS3, UM_PROS3, etc...

### MicroPython version

commit 0a11832cd    esp32: Use the ESP-IDF default esp_console config for ESP32-C3.

### Reproduction

Build UM_TINYS3 or any other board that sets ```CONFIG_TINYUSB_DESC_CUSTOM_PID``` in sdkconfig and observe the VID:PID.

### Expected behaviour

Up to commit fbb02d3ae, ```CONFIG_TINYUSB_DESC_CUSTOM_PID``` sets the PID.  0a11832cd onwards the PID is always the default ```VID:PID=303A:1001```. 

### Observed behaviour

The commit describes C3 specific behavior regarding console.  I don't have a C3 on hand so I don't fully understand the specifics of what was being addressed here, but does look like S3 port is affected and reporting this in case this was not intended.  

Would still like to be able to control USB parameters.  Eg. sdkconfig a la:
```
CONFIG_TINYUSB_DESC_CUSTOM_VID=0x303A
CONFIG_TINYUSB_DESC_CUSTOM_PID=0x80D1
CONFIG_TINYUSB_DESC_BCD_DEVICE=0x0100
CONFIG_TINYUSB_DESC_USE_ESPRESSIF_VID=n
CONFIG_TINYUSB_DESC_USE_DEFAULT_PID=n
CONFIG_TINYUSB_DESC_MANUFACTURER_STRING="Unexpected Maker"
CONFIG_TINYUSB_DESC_PRODUCT_STRING="TinyS3"
CONFIG_TINYUSB_DESC_SERIAL_STRING="_ts3_"
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

stephanelsmith, 2024-08-24:
Adding @UnexpectedMaker as this affects some UM boards with custom PID

UnexpectedMaker, 2024-08-24:
Yeah, I noticed the device name was not being set anymore while I was working on my OMGS3 port - but had not had a chance to look into it yet :(

andrewleech, 2024-08-24:
Thanks for the report. That linked change was done to fix some issues where repl on UART was broken on boards like C3, and overall the configuration was simplified in the process. 
Safe to say though breaking the custom pid/vid was unintended though! 

For S2 and S3 they should be fixed again with https://github.com/micropython/micropython/pull/15108 though it's still pending final review. 
I'm not sure how to fix it on non-tinyusb chips though.

UnexpectedMaker, 2024-08-24:
> Thanks for the report. That linked change was done to fix some issues where repl on UART was broken on boards like C3, and overall the configuration was simplified in the process. Safe to say though [...]

Oh, cool, thanks for the update Andrew, and for all of your awesome work on the UART and USB fixes!

stephanelsmith, 2024-08-26:
I'm going to close the issue. I'm enabled.

@andrewleech Greatly appreciate your response and assistance. Looking forward to updated UART/USB integration!

@UnexpectedMaker, recommendation above should work for the custom PIDs. Also appreciate all your great esp32 based hardware!

closed 2024-08-26 by stephanelsmith