micropython/micropython #15746: unknown output along with my own Print statements

jonhp, 2024-08-29:
### Port, board and/or hardware

Lolin D1 mini-pro v2.0.0

### MicroPython version

MicroPython v1.19.1-1061-g20b3470e21 on 2023-03-31; LOLIN D1 MINI with ESP8266
(This is the first use of this board.)

### Reproduction

n/a

### Expected behaviour

[image]
See this screengrab.

### Observed behaviour

See the attached file above. It contains both expected and unexpected output. (I cannot cut/paste a longer selection as it cuts off somewhere in the hex data; don't know if it is thonny, cut, or paste.) I have included the source in the "Additional Information" below. 

### Additional Information

import network
import socket
import json
from machine import Pin
wlan=network.WLAN(network.STA_IF)
wlan.active(True)
if not wlan.isconnected():
  wlan.connect('alionsystems','t3y48x8qckisojh3blbi45a7j')
  while not wlan.isconnected():
    pass
client=socket.socket(socket.AF_INET,socket.SOCK_DGRAM)
host=''
port=50222
client.bind((host,port))
led=Pin(2,Pin.OUT)
toggle=1
while True:
  try:
    data,addr=client.recvfrom(256)
    led.value(toggle)
    toggle = toggle ^ 1
  except OSError:
    continue
  packet=json.loads(data)
  if packet['type']=='obs_st':
    time_stamp=packet['obs'][0][0]  
    temp=packet['obs'][0][7]
    temp=((temp*9)/5)+32
    humd=packet['obs'][0][8]
    print('time stamp=',time_stamp,' temp=',temp,' humd=',humd)
  
``

### Code of Conduct

Yes, I agree

robert-hh, 2024-08-29:
This rubbish in the output of an ESP8266 is normal. The first messages after boot are sent by the erspressif ROM firmware at 74880 Baud. So no MicroPython bug.

jonhp, 2024-08-29:
Hi, and thanks for the info. In my case, this "rubbish" keeps getting outputted at random times. Sometimes every minute; sometimes several minutes apart. So maybe I didn't completely understand your response.

robert-hh, 2024-08-30:
May it be that the board reboots unexpectedly?
About your first post: Enclose code into lines with three backticks ``` to have it properly formatted. See also the hints about code formatting https://github.com/orgs/micropython/discussions/9111.

jonhp, 2024-08-30:
Hi @robert-hh, I did click on the "<>" in the command bar but it apparently didn't do it's thing. I have updated the original post--i hope. If the board is unexpectedly rebooting, how would I know that for sure. And do you know how I would figure out why?

robert-hh, 2024-08-31:
This line:
`ets Jan 8, 2013, rst cause: 4, boot mode:(3,7)` indicates a reboot. And was well the `wdt reset` = watchdog reset. SInce that happens when WiFi activity is ongoing, it could be a power failure. Make sure that you have a good power supply and a  good USB cable.

Since the device reboots, you could try to set the baud rate of your terminal to 74880 while running to see, if the "rubbish" contains more information.

dpgeorge, 2024-09-02:
You could try adding a `time.sleep_ms(10)` at every iteration in the `while True` loop, to give the esp8266's task handler time to process things.  That may help eliminate wdt resets.

jonhp, 2024-09-02:
@dpgeorge thanks for trying to help out. however, the 10ms delay did not fix the problem.
@robert-hh the USB hub providing power can supply up to 2.1A to the port being used. Tried a different USB cable. My com port will not accept a baud setting of 74880 .

projectgus, 2024-09-03:
@jonhp Suggest also trying with the current 1.23 MicroPython release. Do you still see the problem with this version?

projectgus, 2024-09-03:
(Although I agree this looks a lot like a power issue of some kind.)

jonhp, 2024-09-03:
@projectgus I updated to 1.23; did not solve the problem. I will try yet another power source. Jon

```
MicroPython v1.23.0 on 2024-06-02; ESP module with ESP8266
Type "help()" for more information.
>>> 
 ets Jan  8 2013,rst cause:4, boot mode:(3,0)
wdt reset
load 0x40100000, len 31212, room 16 
tail 12
chksum 0xa6
ho 0 tail 12 room 4
load 0x3ffe8000, len 1060, room 12 
tail 8
chksum 0x80
load 0x3ffe8430, len 1124, room 0 
tail 4
chksum 0xcb
csum 0xcb

����n�r��n|�llld`c��|{�l�n��o�
```

jonhp, 2024-09-03:
I have tried 3 different power sources: PC front panel, 7-port anker hub, dell display hub. Yet the problem remains.

projectgus, 2024-09-04:
> @projectgus I updated to 1.23; did not solve the problem. I will try yet another power source. Jon
> 
> ```
> MicroPython v1.23.0 on 2024-06-02; ESP module with ESP8266
> Type "help()" for more information.
> >>> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,0)
> wdt reset

Thanks @jonhp for the update. Do I understand from this log that you booted to the REPL prompt, did nothing at all, and then saw it crashed? If that's the case then it definitely looks like something with the hardware - it may not be the USB side of the power supply but something on the board itself.

jonhp, 2024-09-04:
@projectgus I will see if I have another board and start over. Later until I know something more.

jonhp, 2024-09-05:
Brand new board out of the package. Same problem; maybe even worse.

dpgeorge, 2024-09-05:
> Brand new board out of the package. Same problem; maybe even worse.

Is it the same brand of board as the first one?  If so it could be an issue with the design of that board, or the particular manufacturing run.

If you are using stock firmware obtained from https://micropython.org/download/ESP8266_GENERIC/ and you get a watch-dog reset just sitting at the REPL with no code in boot.py or main.py, then it is highly likely a hardware issue, because that firmware is known to work.

jonhp, 2024-09-05:
They are both Lolin v2.0.0 boards. For this and other reasons, I am going to switch SBCs. So I am going to close this issue out as undetermined.

jonhp, 2024-09-08:
Well, I'm back. Loaded the same python module (and microPython v1.23.0) onto a fresh HiLetgo NodeMCU ESP8266 CP2102 ESP-12E. Same old problem. So new and different SBC did not solve the problem.

dpgeorge, 2024-09-08:
@jonhp to make progress on this issue can you please provide full instructions on how to reproduce it, so someone else can try to hit the same bug you are seeing?

You provided some code in the original issue report, but running that on an ESP8266 it just turns the LED on and sits there waiting for incoming data.  It runs fine.

So, can you please state exactly how you get the crash to occur.  Eg, run this code on ESP8266, then run this other code on a PC, then see this output, etc.

jonhp, 2024-09-08:
@dpgeorge hi, I don't actually have a standalone test case. The driving network traffic is a live broadcast stream from a Weatherflow Tempest Weather Station. Here is a link to the protocol [https://apidocs.tempestwx.com/reference/tempest-udp-broadcast](url).
I am running a capture now with Wireshark of just the packets sent to port 50222. When it has captured a large block of packets, I will upload same. I looked for a program that could turn the trace into a test case driver, but I could not find one. Thanks, Jon

jonhp, 2024-09-08:
@robert-hh I did try a gc at the end of the endless loop earlier, but it did not help.

jonhp, 2024-09-08:
wireshark trace uploaded

dpgeorge, 2024-09-09:
Without a reproduction it will be hard to debug and fix this issue.

Even with the above wireshark capture, you need to provide details about how the test was set up, how long it ran, which packet it crashed on, etc.

As a pragmatic solution, I suggest using an ESP32 instead of an ESP8266.

jonhp, 2024-09-09:
I added a "small" amount of instrumentation code. This changed the timing and the fault problem occurs at a different frequency. None-the-less I have attached a matching wireshark trace and thonny shell. 
[matching wireshark and shell.zip](https://github.com/user-attachments/files/16928401/matching.wireshark.and.shell.zip)

jonhp, 2024-09-09:
@dpgeorge ESP32: there are literally dozens of these in the wild. I like this one for the physical size and extra flash memory [https://tinyurl.com/24fr4xld](url). however, it appears to lack an external antenna connector. I would like a board that is rock solid stable with a long life with less than 1A power requirement.

jonhp, 2024-09-11:
I ordered an ESP32-S3-DEVKITC-1-N8R2 from DigiKey. It has been running for quite a long time without any unplanned reboots. So, I guess, that for me, the problem is resolved/avoided. Doesn't really answer what is going on with the ESP8266. Thanks to everyone for your help and patience. Jon

jonnor, 2025-03-03:
Reporter says issue to be considered solved.

closed 2025-03-03 by jonnor