micropython/micropython #15751: machine.lightsleep makes time.sleep_ms takes more time on every call

staltux, 2024-08-30:
### Port, board and/or hardware

rp2040

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

### Reproduction

while True:
    # omitted code to turn led on and off    
    time.sleep_ms(500)
    machine.lightsleep(4000)

### Expected behaviour

every call to time.sleep_ms(500) should take the same time

### Observed behaviour

the led take more and more time to turn on and off after the same sleep_ms() 

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

peterhinch, 2024-08-31:
I can't replicate this with release build 1.23.0. I ran it for several minutes and the rate seemed constant.

However the later build RPI_PICO-20240606-v1.24.0-preview.31.g3c8089d1b failed, with the onboard LED flashing at about 1Hz. It seemed that `lightsleep` was doing nothing.
```py
from machine import Pin
import time
p = Pin(25, Pin.OUT)
while True:
    p.toggle()
    time.sleep_ms(500)
    machine.lightsleep(4000)
```

robert-hh, 2024-08-31:
Using a fresh built rp2 firmware I could not replicate either wrong behavior. The led pulse was always 500ms, and the lightsleep always 4000ms. Verified with a logic analyzer. 

MicroPython v1.24.0-preview.270.g35b6a66b0 on 2024-08-31; Raspberry Pi Pico with RP2040

Test code

```py
from machine import Pin, lightsleep
import time
led=Pin("LED", Pin.OUT)
p=Pin(0, Pin.OUT, value=0)
while True:
    led.toggle()
    p.toggle()
    time.sleep_ms(500)
    led.toggle()
    p.toggle()
    lightsleep(4000)
```

robert-hh, 2024-08-31:
I repeated the test with an oscilloscope, persistence set to infinite and statistics on. Screen shot below. There was no deviating event. The variation of the 500ms pulse was 12µs pk-pk.

[image]

peterhinch, 2024-09-01:
That is odd. I've now tried another Pico, with the same V1.24 preview build: you don't need a scope to see that the script isn't working. The LED behaviour is completely wrong. It is inconsistent between runs, but never correct, in my testing.

robert-hh, 2024-09-01:
Could it be related to the USB port? Here, it does not matter. Only that when using it at a USG hub the interface sometimes needs about 40s to register.

peterhinch, 2024-09-01:
I've re-tested with the latest build `RPI_PICO-20240829-v1.24.0-preview.270.g35b6a66b0` and it works.

I guess RPI_PICO-20240606-v1.24.0-preview.31.g3c8089d1b was buggy because it consistently fails.

@robert-hh We are seeing the same behaviour. Under V1.23 and the current preview build the script works OK.