micropython/micropython #15753: ESP32-camera support

cnadler86, 2024-08-31:
### Description

As far as I searched the discussions and issues, I couldn't find any attempt to implement an interface to the [esp32-camera](https://github.com/espressif/esp32-camera) driver in the oficial micropython repo. It should be not so difficult to implement these, since there are already some repos either having custom firmware with it, or also having an interface as source code (e.g. [this one](https://github.com/lemariva/micropython-camera-driver)). I would take these as reference.

My propose would be the following:

- Add the official esp32-camera driver from esp to the respective idf_component.yml as dependency.
- Add a Compiler Flag to include the c interface module or not in the build, depending if the board definition (mpconfigboard.h) includes this flag,
- In the board definition the respective default pins should be defined then,
- The basic look and feel of the module in micropython for the user could be something like the actual implementations:
`import camera
if camera.init(<keyword arguments>):
img = camera.capture()
else:
print('error')`

Another possibility would be to handle the camera as python object. In this case properties could be used to get/set camera parameters. But in this case I don't know exactly how to proceed in c (his is my first micropython contribution and I do most of my coding in high level languages). I guess in this case I would implement the camera as the first approach and build a frozen Micropython module on top.

What do you think?

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

dhalbert, 2024-08-31:
We made an API for `esp32-camera` in CircuitPython, if that would serve for inspiration:
https://github.com/adafruit/circuitpython/tree/main/ports/espressif/bindings/espcamera
https://github.com/adafruit/circuitpython/tree/main/ports/espressif/common-hal/espcamera
https://docs.circuitpython.org/en/latest/shared-bindings/espcamera/index.html
We ended up forking it to make some small changes:
https://github.com/adafruit/esp32-camera

One issue that has come up is that the current version of  `esp32-camera` uses the older, deprecated ESP-IDF I2C interface. We have been blocked moving to the newer I2C interface in general because of this (we are at ESP-IDF 5.3). However, someone _just_ submitted a PR about this to `esp32-camera`: https://github.com/espressif/esp32-camera/pull/677.

cnadler86, 2024-08-31:
> esp32-camera: espressif/esp32-camera#677

Thx Dan for the worthy information. I will take a look at your code then
and I am also happy on information about possible arising issues. At the
moment mpy is at 5.2.2, but who knows, perhaps we are at >5.3 when this
gets implemented.

cnadler86, 2024-09-01:
I looked at the code posted from @dhalbert, and it looks very good. Thanks Dan one more time.
I think this API as basis would be the right approach to build the API for micropython, since it handles the camera interface as mpy-object. But of course is more sophisticated than the actual implementations for micropython. 

I would not fork the esp32-camera driver (since the i2c interface will be updated soon) and use the one provided by espressif, because I and probably also the mpy-community would not have time or want to maintain this fork.

The question would be: do we want to have the same interface to the outside as circuitython, making camera code compatible between micropython and circuitpython, or do we want to define a own micropython interface (thinking forward: the camera API should have the same interface for another Ports). AFAIK there are no camera APIs in other micropython ports, or?

jonnor, 2024-09-01:
OpenMV also has an API for cameras, going back many years. https://docs.openmv.io/library/omv.sensor.html

cnadler86, 2024-09-02:
> OpenMV also has an API for cameras, going back many years. https://docs.openmv.io/library/omv.sensor.html

Didn't know this. Thanks for pointing it out.

What I see now is that there are at least 2 more or less big stakeholders (Adafruit and OpenMV), maybe others having their own camara interface based on micropython. So, IMHO the right way of implementing a camera interface (with a well defined HAL, Port-APIs, etc.) that could suit the demands of these stakeholders would be to start a sponsored project for that, so these stakeholder and everyone else using micropython as basis could use it. But it seams that this train has already left the station...

BTW: In these case I am for sure not the right guy to implement this. I (naively) thought I would start more or less from a green field with some examples and build an API for hobbyists, but it seams that this might not be the case.

dhalbert, 2024-09-02:
Maybe I am misinterpreting what you mean by "stakeholder". I would not call us, Adafruit, a stakeholder. We have no demands: MicroPython can have whatever camera API it wants. I was just suggesting that our implementation might be helpful to you as inspiration for the API and as an example of code that works that you could take or adapt. By the same measure, OpenMV is not a stakeholder, but can serve as similar inspiration.

cnadler86, 2024-09-02:
No, I think I formulated my thought the wrong way: Since micropython is the source of circuitpython, OpenMV and probably others, I think the biggest benefit for all developers using micropython together with a camera would be to have such an API that suits all their needs in the source (of course this is utopic, but that would be my vision). Since the source didn't have such an API, these groups developed/reused their own. Implementing now a Camera-API can have advantages for the groups that already have an API (of course only if they switch to it), such as maintaining the API by a wider community, finding bugs then probably faster, etc., but also having less flexibility when it comes to customization.
And of course implementing an API in MP is now much easier, because we can learn from the camera-API-journey of this groups =).
Perhaps what I am trying to say or put to discussion: Are these groups interested in migrating their APIs to MP and how active want they to get involved in it?

At the moment my plan would be more or less the following:
- implementing the esp32-cam driver with the respective additions in probably modmachine + bindings
- implementing the higher level Camera-API, based on these ESP-low level SW and the already available API-implementations
- all these for the basic functionality with a basic validation, trying to have an acceptable performance and memory usage, tested in 1-2 camera sensors (I cannot offer more than that).

cnadler86, 2024-09-11:
So, I have a first alpha, which I could compile and successfuly test the basics on my esp32s3. This is of course a work in progress and the information, comments, etc.  might not be good, but it is a starting point. 

[Here the link](https://github.com/cnadler86/mp_camera/)

jonnor, 2024-09-13:
Very cool @cnadler86 !

cnadler86, 2024-09-17:
Hello strangers,

Have finally a release candidate which I would call stable and OK-documented. Feel free to use it or comment on it.

I still think it would be good to incorporate this in micropython, but if this is not the strategy or the wish of the community, at least I would suggest to put it in [awesome-micropython](https://github.com/mcauser/awesome-micropython) and leave it as user-module.

jonnor, 2024-09-25:
Definitely get this into awesome-micropython. To do that, the best is to submit a pull request following the contributions guidelines: https://github.com/mcauser/awesome-micropython/blob/master/contributing.md#submitting-pull-requests
I have found the maintainer to be very responsive and friendly :)

I actually realize I have an ESP32+OV2640 device - maybe I will test it out on that device.

jonnor, 2024-09-25:
Having an issue when testing on ESP32 - iram overflow. I filed an issue in the repo, https://github.com/cnadler86/micropython-camera-API/issues/1

cnadler86, 2024-10-12:
Hi,

since here is not happening much I will close the Issue. The API is now in awesome-micropython. I will let the topic mature and depending on the resonance, open a PR to merge it in here in the future.

closed 2024-10-12 by cnadler86