micropython/micropython #15763: SoftI2C instance hangs reading TOF400(C) device until soft-reset.

kwinter745321, 2024-09-02:
### Port, board and/or hardware

STM32. STM32F411CEU6. aka the BlackPill.

### MicroPython version

MicroPython v1.20-0-326-gcffcce4b53 on 2023-07-26; WeAct Studio Core with STM32F411CE

Also, I rebuilt MP, and tried it again with same hang on this version:
MicroPython v1.24.0-preview.276.g1897fe622 on 2024-09-02

Driver file (placed on STM32 chip)
[vl53l1x.py.txt](https://github.com/user-attachments/files/16841087/vl53l1x.py.txt)

Ran program from Thonny.  Note: Thonny is 4.1.1
[TOF400Ftest.py.txt](https://github.com/user-attachments/files/16841089/TOF400Ftest.py.txt)

### Reproduction

Description
When running the program, the session in Thonny hangs until soft-reset.
In TeraTerm it just hangs.

Questions
1. Is there a way to fix the hang.  
2. If not, is there a way to interrupt the I2C session without needing a soft reset.

Hardware
A board titled MH-TOF400(C) from AliExpress is a Time of Flight device that uses I2C interface.
Board is connected to I2C pins of STM32F411CEU6 chip.

Alternate Hardware (that worked)
The device functions correctly on my Arduino ESP32-Wroom board.
Also, Three years ago using MicroPython on RPI Pico.

Preliminary Test
The I2C.scan() was performed and returned hex 0x29

Test
Driver and Thonny files are attached.
```
from machine import Pin, SoftI2C, lightsleep
from vl53l1x import VL53L1X
import time, sys

i2c=SoftI2C(scl=Pin("PB6"), sda=Pin('PB7'), freq=500000)
distance = VL53L1X(i2c)

dat = distance.read()
machine.lightsleep(100)
print("{}".format(dat))
machine.lightsleep(10)
distance.clear()

```
I monitored the session using a Saleae Logic 2 (Logic Analyzer), and TOF400 functions correctly.
A CSV file contains the session data.

Output does not appear until (red) Stop-icon is pressed.
```
799
done
```
Output is correct and can be calc from Logic Analyzer session (attached).

[run-1.24.csv](https://github.com/user-attachments/files/16841141/run-1.24.csv)
[run-1.24-console.txt](https://github.com/user-attachments/files/16841146/run-1.24-console.txt)

The code by Lee Halls was posted 3 years ago for RPI Pico MicroPython  at [Github.com/drakxtwo/vl53l1x_pico  ](https://github.com/drakxtwo/vl53l1x_pico)

### Expected behaviour

I expected the I2C data to be read and displayed in the Thonny Shell (without) needing to stop the program execution.

### Observed behaviour

The output hangs until the Thonny session is stopped.

### Additional Information

Initially, the Thonny session would crash when I did the Red Stop.

I compared the Arduino and MicroPython Driver activation data and walked through the Arduino code.  So, I added the function called clear() to the STM32 MicroPython driver. This instruction performs a Clear Interrupt.  In the Logic Analyzer session this command can be seen, and it causes the interrupt line to drop to zero briefly.  After this Could run the Thonny code without a crash.  But I still had to interrupt the session using the (red) stop button (which i assume does a soft reset).
 
The Arduino code performed the same Clear_interrupt() function in its loop.

### Code of Conduct

Yes, I agree

robert-hh, 2024-09-03:
As usual: Do you have pull-up resistors at SDA and SCL?

kwinter745321, 2024-09-04:
Hi. The board has them. I tried it also with a set of 10k ohm resistors on
sda and scl line.

It worked as is with esp32 running arduino.

Kevin