micropython/micropython #15769: Unable to disable default usb UART on ESP32S3

fduxiao, 2024-09-03:
### Port, board and/or hardware

esp32s3, self-made board

### MicroPython version

MicroPython v1.22.0-preview.975.g1897fe622.dirty on 2024-09-02; AntiGM (esp32s3 spiram octal) with ESP32S3

The board is self-made, which uses pin 19, 20 (USB D+/D-) to drive an ST7796 screen (i8080 parallel). I use an extra usb2serial bridge for downloading. See [here](https://github.com/fduxiao/AntiGM/blob/8170583b178ee4fd1f7caa2bc0c21bf533a2a7a4/anti_gm/defs.py#L20).

### Reproduction

To disable USB, I removed `boards/sdkconfig.usb` from `SDKCONFIG_DEFAULTS` and added the following.
```
CONFIG_USB_ENABLED=n
CONFIG_USB_CDC_ENABLED=n

CONFIG_TINYUSB_CDC_ENABLED=n
CONFIG_ESP_CONSOLE_USB_CDC=n
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=n
CONFIG_USB_OTG_SUPPORTED=n
```

But this results in a failure of `make BOARD=myboard`.

### Expected behaviour

You should be able to compile a firmware for esp32s3 with USB disabled. 

### Observed behaviour

`usb.c` complains that some symbols are missing, e.g., the following.
```
/Users/***/mcu/micropython/ports/esp32/usb.c:41:27: error: 'CONFIG_TINYUSB_CDC_RX_BUFSIZE' undeclared here (not in a function); did you mean 'CFG_TUD_CDC_RX_BUFSIZE'?
   41 | static uint8_t usb_rx_buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE];
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           CFG_TUD_CDC_RX_BUFSIZE

```

### Additional Information

I first tested this on [lv_micropython](https://github.com/lvgl/lv_micropython) and succeeded in making a usable firmware which can drive the screen. Since lv_micropython is a very old fork of micropython and is based on esp-idf v4.4, I decided to try micropython directly in case someday esp-idf v4 is no longer supported. Then came this issue, as is also encountered by others: #15290, #11315.

I noticed a change in `usb.c`. In [this commit](https://github.com/micropython/micropython/commit/e4650125b88a35f074097f16d84a8f49bd22ac06#diff-1e8ef61b5781a5e8c2048f1587152e6ede370a203a059185b77bb156357f46fa), you changed `CONFIG_USB_ENABLED` to `CONFIG_USB_OTG_SUPPORTED`. However, this is always set to yes as long as your `IDF_TARGET` is set to esp32s3. Try this simple CMakeLists.txt.
```cmake
cmake_minimum_required(VERSION 3.12)

# Define the output sdkconfig so it goes in the build directory.
set(SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig)

# Specific options for IDF v5.2 and later
set(SDKCONFIG_IDF_VERSION_SPECIFIC "")
if (IDF_VERSION VERSION_GREATER_EQUAL "5.2.0")
    set(SDKCONFIG_IDF_VERSION_SPECIFIC boards/sdkconfig.idf52)
endif()

set(IDF_TARGET esp32s3)

# Include main IDF cmake file.
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

# Define the project.
project(test)
```

Then `cat sdkconfig|grep USB_OTG` will show you:
```
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_OTG_PERIPH_NUM=1
CONFIG_ESP_ROM_USB_OTG_NUM=3
CONFIG_USB_OTG_SUPPORTED=y
```

Thus, you are unable to change it by `sdkconfig.board` because it only defines the default values. This explains #15290.

So my solution to this is to undef CONFIG_USB_OTG_SUPPORTED in `usb.h`
```C
// ports/esp32/usb.h

#ifndef MICROPY_INCLUDED_ESP32_USB_H
#define MICROPY_INCLUDED_ESP32_USB_H

#define MICROPY_HW_USB_CDC_TX_TIMEOUT_MS (500)

void usb_init(void);
void usb_tx_strn(const char *str, size_t len);

#endif // MICROPY_INCLUDED_ESP32_USB_H

#undef CONFIG_USB_OTG_SUPPORTED
#define CONFIG_USB_OTG_SUPPORTED 0

```

I guess you have to go back to a configurable variant name.

### Code of Conduct

Yes, I agree

andrewleech, 2024-09-03:
There has been a lot of related discussion to this recently with this change just merged today: https://github.com/micropython/micropython/pull/15727#pullrequestreview-2276408728

The USB support for S2 and S3 is also looking to get updated in https://github.com/micropython/micropython/pull/15108 though I'm still not sure if that will neccesarily allow completely disabling the USB port, that hasn't been explicitely tested or added to date.

projectgus, 2024-09-04:
Thanks for the detailed report, @fduxiao . There isn't functionality to disable the USB pins on S2 or S3 yet, but as discussed in https://github.com/micropython/micropython/issues/14217 it should be technically possible to do so both at compile time (with an easier build config option) and at runtime by simply assigning the pins to other functions. Assigning at runtime should work for most people.

I'm hoping to work on this soon and implement it, but none of the PRs currently open are quite enough to fix this.

In the meantime I'm going to close this as a duplicate of #14217, please follow that issue for updates.

closed 2024-09-04 by projectgus

jd3096-mpy, 2024-09-11:
> CONFIG_USB_ENABLED=n
> CONFIG_USB_CDC_ENABLED=n
> 
> CONFIG_TINYUSB_CDC_ENABLED=n
> CONFIG_ESP_CONSOLE_USB_CDC=n
> CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=n
> CONFIG_USB_OTG_SUPPORTED=n

I encountered the same situation as you, and now I want to compile a temporary firmware. What should I do?
After making changes here, the compilation still results in errors at the end.

#undef CONFIG_USB_OTG_SUPPORTED
#define CONFIG_USB_OTG_SUPPORTED 0

fduxiao, 2024-09-11:
@jd3096-mpy Please give some logs or description about the error.

straga, 2024-09-17:
@jd3096-mpy  Try #define MICROPY_HW_USB_CDC (0)