micropython/micropython #15776: RP2 build, doc strings, help()

thestumbler, 2024-09-03:
### Port, board and/or hardware

rp2 pico

### MicroPython version

`MicroPython v1.24.0-preview.278.g5e692d046.dirty on 2024-09-03; RP2040`

I have set `MICROPY_ENABLE_DOC_STRING (1)` in `mpconfigboard.h`

When building this way, I can clearly see, and print, `__doc__` for each class. But the `help()` function doesn't return the docstring like it does in Python on the desktop.Also, there is only a `__doc__` on a class -- on functions a doc string is not created regardless of the setting in `mpconfigboard.h`.  Is this the expected behavior? Is this maybe unique to the RP2 port? I seem to remember this working on a STM32F4 board.

### Reproduction

```
class Ocean:
  '''This is the Ocean Class.'''
  def __init__(self):
    pass
  def wave(self):
    '''Prints a wave.'''
    print('a wave')

```

### Expected behaviour

Expected help(Ocean) or help(o) to show the doc string. Instead it shows what looks like dir(Ocean)

Expected help(o.wave) to show the doc string for the wave() function. Instead there is no doc string.

### Observed behaviour

You can see the __doc__ object on the class, but not on the function.
help() doesn't show the usual help output

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

projectgus, 2024-09-04:
FWIW I see the same behaviour in the unix port if I enable `MICROPY_ENABLE_DOC_STRING` there.

I'm not the best person to answer this, but as far as I know this is all the support MicroPython's `help()` function has for doc strings.

1. The built-in help printer doesn't have any special handling for `__doc__`: https://github.com/micropython/micropython/blob/master/py/builtinhelp.c#L133
2. I don't see `MP_QSTR___doc__` anywhere in the C code except for the part of the parser that parses the docstring, and `__doc__` doesn't show up anywhere likely looking, either.

thestumbler, 2024-09-04:
Interesting. Without corresponding help() function support, why was the compile flag added to enable doc strings? Is there some utility of having only doc strings and not help() support that I'm missing?

thestumbler, 2024-09-04:
Finally dug up an STM32 Discovery board and tried this out. No joy. I guess I was remembering wrong. Here's the output (very similar, if not identical to the RP2 build)

```
MPY: sync filesystems
MPY: soft reboot
MicroPython v1.24.0-preview.282.g6be1dbc78 on 2024-09-04; F429I-DISCO with STM32F429
Type "help()" for more information.
>>> dir(Ocean)
['__class__', '__init__', '__module__', '__name__', '__qualname__', '__bases__', '__dict__', '__doc__', 'wave']
>>> help(Ocean)
object <class 'Ocean'> is of type type
  __qualname__ -- Ocean
  __doc__ -- This is the Ocean Class.
  __module__ -- __main__
  wave -- <function wave at 0xd002f460>
  __init__ -- <function __init__ at 0xd002f430>
>>> help(Ocean.wave)
object <function wave at 0xd002f460> is of type function
>>> dir(Ocean)
['__class__', '__init__', '__module__', '__name__', '__qualname__', '__bases__', '__dict__', '__doc__', 'wave']
>>> dir(Ocean.wave)
['__class__', '__name__', '__globals__']
>>> o = Ocean()
>>> help(o)
object <Ocean object at d002fb00> is of type Ocean
  __qualname__ -- Ocean
  __doc__ -- This is the Ocean Class.
  __module__ -- __main__
  wave -- <function wave at 0xd002f460>
  __init__ -- <function __init__ at 0xd002f430>
>>> help(o.wave)
object <bound_method> is of type bound_method
>>> dir(o)
['__class__', '__init__', '__module__', '__qualname__', '__dict__', '__doc__', 'wave']
>>> dir(o.wave)
['__class__', '__name__', '__globals__']
>>> o.wave.__doc__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'bound_method' object has no attribute '__doc__'
>>>
```

mhsmith, 2024-11-22:
Docstrings are handled  in `check_for_doc_string` in compile.py. But this function is only called when `scope->kind == SCOPE_MODULE` or `SCOPE_CLASS`, not `SCOPE_FUNCTION`. I don't know if this is deliberate.

markusand, 2026-08-20:
Still happening in STM32 builds

agatti, 2026-08-21:
> Docstrings are handled in `check_for_doc_string` in compile.py. But this function is only called when `scope->kind == SCOPE_MODULE` or `SCOPE_CLASS`, not `SCOPE_FUNCTION`. I don't know if this is deliberate.

Adding an explicit doc extraction in `SCOPE_FUNCTION` ([1]) acually extracts the docstring just fine and adds it to the compiler scope, but the scope is marked as *local*.  I guess local scopes and all their data get purged once they're not needed anymore, hence why the `__doc__` attribute is missing from the function.  Other special attributes like `__name__` are put in the *global* scope and thus are retained.

As of why, no idea, haven't looked into why the function scope is local either :)

```diff
diff --git i/py/compile.c w/py/compile.c
index 263f45676..d7817f65e 100644
--- i/py/compile.c
+++ w/py/compile.c
@@ -3090,6 +3090,9 @@ static bool compile_scope(compiler_t *comp, scope_t *scope, pass_kind_t pass) {
             #endif // MICROPY_EMIT_NATIVE
         }
 
+        if (!comp->is_repl) {
+            check_for_doc_string(comp, pns->nodes[3]);
+        }
         compile_node(comp, pns->nodes[3]); // 3 is function body
         EMIT_ARG(load_const_tok, MP_TOKEN_KW_NONE);
         EMIT(return_value);
```