micropython/micropython #15795: Build fails for STM32L452XX with USB enabled

higstar, 2024-09-05:
### Port, board and/or hardware

stm32 STM32L452xx

### MicroPython version

v1.23.0-287-g042693496

### Reproduction

Edit NUCLEO_L452RE's mpconfigboard.h adding the following:
```c
#define MICROPY_HW_ENABLE_USB       (1)
#define MICROPY_HW_USB_FS           (1)

```

### Expected behaviour

Expected the build to complete successfully.

### Observed behaviour

Build fails as follows:
```
...
stm32_it.c: In function 'OTG_CMD_WKUP_Handler':
stm32_it.c:396:9: error: implicit declaration of function '__HAL_PCD_UNGATE_PHYCLOCK' [-Werror=implicit-function-declaration]
  396 |         __HAL_PCD_UNGATE_PHYCLOCK(pcd_handle);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
```

### Additional Information

Patching as follows gets things underway:
```bash
sed -i 's_#elif defined(STM32L432xx)$_#elif defined(STM32L432xx) || defined(STM32L452xx)_g' ports/stm32/usbd_conf.c
sed -i 's_defined(STM32L1) || defined(STM32L432xx) || defined(STM32WB)_defined(STM32L1) || defined(STM32L432xx) || defined(STM32L452xx) || defined(STM32WB)_g' ports/stm32/mpconfigboard_common.h
sed -i 's_#elif defined(STM32L0) || defined(STM32L432xx)$_#elif defined(STM32L0) || defined(STM32L432xx) || defined(STM32L452xx)_g' ports/stm32/stm32_it.c
```

### Code of Conduct

Yes, I agree

Tico06, 2024-09-11:
Hi @higstar ,

I made a test by connecting an USB module to a NUCLEO-L452RE.

I patched the code with your sed proposal without any error.

And updated mpconfigboard.h, commenting REPL config, and adding USB config:
```c
// USART2 is connected to the ST-LINK USB VCP
// #define MICROPY_HW_UART_REPL        PYB_UART_2
// #define MICROPY_HW_UART_REPL_BAUD   115200

// USB config
#define MICROPY_HW_ENABLE_USB       (1)
#define MICROPY_HW_USB_FS           (1)
```

Then, first run a make clean:
```
$ make clean BOARD=NUCLEO_L452RE
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -rf build-NUCLEO_L452RE
```

Then the make submodules:
```
$ make submodules BOARD=NUCLEO_L452RE
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/libhydrogen lib/stm32lib lib/micropython-lib
Synchronisation de l'URL sous-module pour '../../lib/libhydrogen'
Synchronisation de l'URL sous-module pour '../../lib/micropython-lib'
Synchronisation de l'URL sous-module pour '../../lib/stm32lib'
```

Now, the funny things, the make:
```
$ make BOARD=NUCLEO_L452RE
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-NUCLEO_L452RE/genhdr
GEN build-NUCLEO_L452RE/genhdr/pins.h
...
CC build-NUCLEO_L452RE/pins_NUCLEO_L452RE.c
LINK build-NUCLEO_L452RE/firmware.elf
   text    data     bss     dec     hex filename
 307776      56   21140  328972   5050c build-NUCLEO_L452RE/firmware.elf
GEN build-NUCLEO_L452RE/firmware.bin
GEN build-NUCLEO_L452RE/firmware.dfu
GEN build-NUCLEO_L452RE/firmware.hex
```
Then, I strapped BOOT0 and deployed with dfu-util:
```
$ make BOARD=NUCLEO_L452RE USE_PYDFU=0 deploy
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Writing build-NUCLEO_L452RE/firmware.dfu to the board
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Multiple alternate interfaces for DfuSe file
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
File contains 1 DFU images
Parsing DFU image 1
Target name: ST...
Image for alternate setting 0, (1 elements, total size = 307840)
Setting Alternate Interface #0 ...
Parsing element 1, address = 0x08000000, size = 307832
Erase           [=========================] 100%       307832 bytes
Erase    done.
Download        [=========================] 100%       307832 bytes
Download done.
Done parsing DfuSe file
Submitting leave request...
Transitioning to dfuMANIFEST state
```
Unplugged the USB cable, unstrapped BOOT0, replugged USB cable and mpremote:
```
$ mpremote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
MicroPython v1.24.0-preview.76.gb4213c9c9.dirty on 2024-09-11; NUCLEO-L452RE with STM32L452RE
Type "help()" for more information.
>>> 
```

Please tell me if you need further tests.

Reg,

Eric.

dpgeorge, 2024-09-17:
Thanks @Tico06 for testing.

Fixed by 39ddfed9a0360af147193a6eb09228fd6afa0c41

closed 2024-09-17 by dpgeorge