micropython/micropython #15804: docs: Can't determine origin of MAC for W5500-PICO-EVB or W5500 chip, WIZNET5k module

BB486999, 2024-09-06:
### Documentation URL

https://docs.micropython.org/en/latest/library/network.WIZNET5K.html

### Description

In this case, I am working with the W5500-PICO-EVB, but also tried this with a external W5500 chip to confirm.

Each time that the nic is configured, whether its the onboard w5500-PICO-EVB or using a pico with an external w5500, with:

```
#SPI SETUP for onboard WizNET chip
MOSI = Pin(19)
MISO = Pin(16)
SCK = Pin(18)
CS = Pin(17)
RST = Pin(20)
INT = Pin(21)

#W5500 onboard chip initialization
def w5500_init_dhcp():
    spi=SPI(0,2_000_000, mosi=MOSI,miso=MISO,sck=SCK)
    print(spi)

    nic = network.WIZNET5K(spi,CS,RST, INT) # type: ignore #spi,cs,reset pin  DONT FORGET THE INT (interrupt), or ELSE WE INTRODUCE POLL DELAYS

    nic.active(True)

    #It looks like we have a mac address...
    mac = nic.config('mac')
    print("''''''''''''''''''''''''''")
    print("MAC ADDR: -> ", ':'.join('{:02x}'.format(mac[i]) for i in range(6)))
```

a mac address is generated for the pico.  It does not appear to be random, and will always come up with the same MAC each time it is powered up and run.  Also, between different w5500-PICO-EVB's, they all seem to generate a unique MAC address.  

I was playing around with manually resetting the w5500's registers directly from SPI, and I was able to write a 'new' mac of '00:00:00:00:00:00', and then was able to verify by reading that same mac from the chip.  However, after resetting the chip, starting the wiznet5k module again, it would appear with its unique generated MAC.

So basically, I'm trying to figure out where exactly this comes from:

       -I'm pretty sure this is not a true assigned MAC, given the original design for the pico
       -The MAC stayed the same if I used a w5500-PICO-EVB, or reconfigured and used the same w5500-PICO-EVb but with an external w5500 chip unit (not the onboard one, I also switched all pins to ensure I wasn't still using the onboard w5500).  This makes me think it is somehow tied in with the Pico itself

Where are these MACs coming from, and how are they generated?  It would be really nice if there was a little more documentation on the wiznet5k module.  Note, I recently had opened an issue, and found that there was an INT parameter in the setup for the interrupt pin which wasn't documented, and was causing many bugs in my programs.

If this is a valid method of generating MACs, then I'll probably just let it do its thing.  If not, then I'll need to find an alternative and assign my own.  

### Code of Conduct

Yes, I agree

jonnor, 2024-09-08:
The code for this module is found in extmod/network_wiznet5k.c
https://github.com/micropython/micropython/blob/3ca01eccae528a74fec8b1731f472d41018e5225/extmod/network_wiznet5k.c#L171
It uses getSHAR and setSHAR functions from this driver https://github.com/andrewleech/wiznet_ioLibrary_Driver
The datasheets of the Wiznet W5500 can also be consulted.

Combined, they should have the answers you seek.

BB486999, 2024-09-08:
Hey Jonnor, thanks for the response.  I actually forgot to mention that I did take a look through the source code in that folder you shared (forgot to mention in the above post), and did come across the same function (below from the w5500.h):

```
/**
 * @ingroup Common_register_access_function
 * @brief Set local MAC address
 * @param (uint8_t*)shar Pointer variable to set local MAC address. It should be allocated 6 bytes.
 * @sa getSHAR()
 */
#define setSHAR(shar) \
		WIZCHIP_WRITE_BUF(SHAR, shar, 6)

/**
 * @ingroup Common_register_access_function
 * @brief Get local MAC address
 * @param (uint8_t*)shar Pointer variable to get local MAC address. It should be allocated 6 bytes.
 * @sa setSHAR()
 */
#define getSHAR(shar) \
		WIZCHIP_READ_BUF(SHAR, shar, 6)

/**
 * @ingroup Common_register_access_function
 * @brief Set local IP address
 * @param (uint8_t*)sipr Pointer variable to set local IP address. It should be allocated 4 bytes.
 * @sa getSIPR()
 */
 ```

 It appears all this function does it take in a predefined MAC and write it via SPI to the w5500.  It should be noted that I also worked from the w5500 data sheet, and was able to replicate the same using raw SPI in micropython (I wrote 00:00:00:00:00:00).  ALSO, i did receive confirmation from Wiznet support that these chips (both the w5500, nor the w5500-Pico-evb ship with MAC addresses).

So with that said, I believe the getSHAR and setSHAR functions make sense.

 However, my question is where does the MAC originate in the code?  (I’ve had a hard time tracing through all the functions).  As stated above, each time I configure a new w5500-PICO-EVB, both using its own onboard w5500 or even hooking it up to an external w5500, it seems to always generate a unique MAC address in both cases, and it seems to be tied not to the w5500 but to the RP2040 itself.  This is what I have not yet been able to figure out.  I’ll continue digging through the source code, but if anyone has some pointers it would be much appreciated.  Thanks

andrewleech, 2024-09-16:
By memory it ultimately comes from here: https://github.com/micropython/micropython/blob/master/ports%2Frp2%2Fmphalport.c#L187

However I thought that should be consistently the same for any single unit, generated from the fixed unique id register.

The random in the title is referring to the official block of ip addresses that's set aside for use by devices that generate random addresses because they haven't bought their own block, this function ensures the generated address is in this block.

BB486999, 2024-09-17:
Ah that it was I was looking for but couldn’t find.  Thank you!

So would it be safe to say that I should not have MAC conflicts between picos, and also not if I use them in a network with other devices which have licensed MAC addresses (shouldn’t conflict since they are generated from the unlicensed block)?

andrewleech, 2024-09-18:
That's the intention yes!

razvanphp, 2024-09-23:
I also get a "random" mac address like `02:37:47:12:32:bb` when using STM32 with PHY.

Can we maybe propose an API change for setting the MAC address like this (or in the constructor)?
```
mac_bytes = b'\xA8\x61\x0A\xAE\x44\xE0'
network.LAN.config(trace=8, low_power=False, mac=mac_bytes)
```
Current documentation: https://docs.micropython.org/en/latest/library/network.LAN.html#network.LAN.config

TheGuyDanish, 2025-02-17:
I would also love to see the ability to set the MAC either in a config call or in the constructor!