micropython/micropython #15806: TypeError: can't convert float to int

aleferna12, 2024-09-07:
### Port, board and/or hardware

rp2, Pico W

### MicroPython version

Version: MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

This works:

```Timer(-1, mode=Timer.PERIODIC, period=1000, callback=lambda _: led.toggle())```

But this doesn't:

```Timer(-1, mode=Timer.PERIODIC, period=1000.0, callback=lambda _: led.toggle())```

Error:

```TypeError: can't convert float to int```

Surprised me a lot, is this intended behaviour?

Cheers

### Reproduction

```
from machine import *

led = Pin("LED", Pin.OUT)
# Some arguments are only accepted as integers! That's surprising
timer = Timer(-1, mode=Timer.PERIODIC, period=1000.0, callback=lambda _: led.toggle())
```

### Observed behaviour

Fails with: `TypeError: can't convert float to int`

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

andrewleech, 2024-09-07:
I believe this is indeed intended behaviour.

Most functions that accept an integer argument will not automatically convert non-integers as that could create unexpected behaviour. In the same way it won't automatically convert a string like "1" to int.

sosi-deadeye, 2024-09-07:
If it silently converts the `float` to an `int`, the outcome would be `0` if the float was `0.9`.

aleferna12, 2024-09-07:
> If it silently converts the `float` to an `int`, the outcome would be `0` if the float was `0.9`.

That is what i think would expected behaviour for the function, but it currently doesnt do that. Instead it fails with a TypeError

robert-hh, 2024-09-07:
The Timer has ms granularity. It seems to me consistent to accept only integers as argument. If if would accept floats, then probably soon someone would come along and complain, that the timer would not run at e.g. a period of 2.5 ms, when told so. If the period in an application is by chance derived as a float value, it can convert it easily into an int when setting up the Timer.

aleferna12, 2024-09-07:
> The Timer has ms granularity. It seems to me consistent to accept only integers as argument. If if would accept floats, then probably soon someone would come along and complain, that the timer would [...]

Yeah that is fair.

closed 2024-09-07 by aleferna12