micropython/micropython #15814: Raspberry Pi Pico ADC changes the sampling rate in Free-running Sampling mode

leezisheng, 2024-09-09:
### Port, board and/or hardware

Raspberry Pi Pico

### MicroPython version

MicroPython v1.23.0 ,RPI_PICO-20240602-v1.23.0

### Reproduction

    def configure_adc_sample_rate(self, sample_rate):
        """
        Configure the ADC sampling rate. Due to unknown reasons, the DIV register cannot be configured here. The actual sampling period is fixed at 65
        :param sample_rate: Desired sample rate (Hz)
        :return: None
        """
        # Ensure that the sampling rate is within a reasonable range
        if sample_rate <= 0 or sample_rate > 48_000_000:
            raise ValueError("sample rate out of range")

        # Calculate divider value
        total_period = 48_000_000 / sample_rate
        int_part = int(total_period) - 1
        frac_part = int((total_period - int_part - 1) * 256)
        div_value = mem32[DMA_ADC_Transfer.DIV_REG]

        
        # Write the sampling period to the DIV register
        mem32[DMA_ADC_Transfer.DIV_REG] = div_value | (int_part << 8) | frac_part

### Expected behaviour

I am configuring the ADC sampling rate here. Due to unknown reasons, the DIV register cannot be configured here. The actual sampling period is fixed at 65.
Can anyone help me what is going wrong here？

### Observed behaviour

(base) PS D:\lee\windows terminal\terminal-1.17.11461.0> mpremote connect COM8
Connected to MicroPython at COM8
Use Ctrl-] or Ctrl-x to exit this shell

>>> div_value = mem32[DMA_ADC_Transfer.DIV_REG]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'mem32' isn't defined
>>> from machine import mem32
>>> div_value = mem32[DMA_ADC_Transfer.DIV_REG]
>>> div_value
12582656
>>> total_period = 48_000_000 / 48000
>>> total_period = 48_000_000 / 480000
>>> total_period
100.0
>>> int_part = int(total_period) - 1
>>> frac_part = int((total_period - int_part - 1) * 256)
>>> mem32[DMA_ADC_Transfer.DIV_REG] = div_value | (int_part << 8) | frac_part
>>> mem32[DMA_ADC_Transfer.DIV_REG]
12582656
>>> total_period = 48_000_000 / 480000
>>> int_part = int(total_period) - 1
>>> frac_part = int((total_period - int_part - 1) * 256)
>>> mem32[DMA_ADC_Transfer.DIV_REG] = div_value | (int_part << 8) | frac_part
>>> mem32[DMA_ADC_Transfer.DIV_REG]
12582656
>>>

### Additional Information

[image]

### Code of Conduct

Yes, I agree

leezisheng, 2024-09-09:
[image]
[image]
[image]

leezisheng, 2024-09-09:
[image]

ricksorensen, 2024-09-09:
What is the value of `DMA_ADC_Transfer.DIV_REG`?  The value of the DIV register should be 0 after boot while you show `hex(12582656)=0xbfff00` 

Here is my quick run with current development version using your example.
```
$ mpremote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
MicroPython 0a11832cd on 2024-08-22; XIAO RP2040 Local with RP2040
Type "help()" for more information.
>>> radcbase=0x4004c000
>>> radcdiv=radcbase+0x10
>>> import machine
>>> machine.mem32[radcdiv]
0
>>> tper=100
>>> itper=int(tper)-1
>>> fper=int((tper-itper-1)*256)
>>> print(f"tp: {tper}    ip: {itper}   fp: {fper}")
tp: 100    ip: 99   fp: 0
>>> divval = (itper<<8) | fper
>>> hex(divval)
'0x6300'
>>> machine.mem32[radcdiv]=divval
>>> machine.mem32[radcdiv]
25344
>>> hex(25344)
'0x6300'
>>>
```

leezisheng, 2024-09-09:
done,expression `div_value | (int_part << 8) | frac_part` change to  `(int_part << 8) | frac_part`.

scruss, 2024-09-11:
Is this teh same as your discussion here: [Raspberry Pi Pico ADC changes the sampling rate in Free-running Sampling mode · micropython · Discussion #15815](https://github.com/orgs/micropython/discussions/15815)?

If so, please close this issue.