micropython/micropython #15833: Accept not responding on Pico W after 5 mins

countrypaul, 2024-09-11:
### Port, board and/or hardware

Paspberry Pi Pico W

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

### Reproduction

`import network
import socket
import time

ssid = 'SSID'
password = 'PASSWORD'

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(ssid, password)
while not wlan.isconnected():
    time.sleep(1)
print("Connected", wlan.ifconfig()[0])

addr = socket.getaddrinfo('0.0.0.0', 80)[0][-1]
s = socket.socket()
s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
s.bind(addr)
s.listen(1)

while True:
   cl, addr = s.accept()
   request = cl.recv(1024)
   cl.send("HTTP/1.0 200 OK\r\nContent-type: text/html\r\n\r\n")
   cl.send("<html><body>{}</body></html>".format(time.time()))
   cl.close()
`

### Expected behaviour

I expected the Pico W to accept requests and respond with the time (single integer) indefinitely.

### Observed behaviour

Running the example code (which is a simplified version of the webserver example on the RPI site) I can send requests from FIrefox to thePico W ip address and get an imediate response. If the Pico W is left about 5 minutes or more with receiving a request then it no longer appears to accept on he socket. 

### Additional Information

If I put a timeout on the socket of 30 seconds the Pico W works for longer than 5 minutes but does eventually stop responding.
Putting a line into the html so that it refeshes every minute will keep the Pico responding for hours.

### Code of Conduct

Yes, I agree

sufyan-mukadam, 2024-09-11:
Have you tried changing the power mode, try `wlan.config(pm=network.WLAN.PM_NONE)` to disable the power saving.

sufyanaslam198, 2024-09-11:
Have you tried changing the power mode, try `wlan.config(pm=network.WLAN.PM_NONE)` to disable the power saving.

countrypaul, 2024-09-11:
I did try this which I believe is equivalent to the above suggestion: `wlan.config(pm = 0xa11140) # Disable power-saving for WiFi`

It made no difference. For completeness, I have been using a dedicated AP which is about 3m from where the Pico W is, so I believe it highly unlikely if not impossible for the cause to be too much network traffic. The results are quite reproducible which also would be more than a little surprising if it was network load.

countrypaul, 2024-09-12:
I forgot to mention I measured memory usage (thanks to scruss on the RPi forum for pointng it out). I put checks in before and after the call to accept but when the accept does not simply return obviously I don't get a reading. There was about 175K free beforehand. In the case where I have a timeout, then I closed the socket in the error handling and went through the opening process again but got identical results everytime round suggesting it is not a memory leak or shortage problem. I don't know whether memory is becomimg too fragmented, but if so that must be happening in the accept tree which would seem very unlikely to me

dpgeorge, 2024-09-12:
I tested the above code and tried to reproduce the scenario, but did not see any issues.

My test set up is:
- `MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040`
- The Pico W runs the test code posted in the "Reproduction" part at the top comment here, edited only to change the AP SSID and password.
- The host PC runs the following Python script:

```py
import time
import requests

t = 2
while True:
    print('send request')
    response = requests.get("http://<pico-w-ip>/")
    print(response.text)
    response.close()
    print('sleep', t)
    time.sleep(t)
    t = int(t * 1.5)
```

That host script aims to replicate the case of waiting longer and longer times between requests.

I ran it for about 1.5 hours and got the following output:
```
$ python host.py
send request
<html><body>1609459557</body></html>
sleep 2
send request
<html><body>1609459559</body></html>
sleep 3
send request
<html><body>1609459563</body></html>
sleep 4
send request
<html><body>1609459567</body></html>
sleep 6
send request
<html><body>1609459573</body></html>
sleep 9
send request
<html><body>1609459583</body></html>
sleep 13
send request
<html><body>1609459596</body></html>
sleep 19
send request
<html><body>1609459615</body></html>
sleep 28
send request
<html><body>1609459644</body></html>
sleep 42
send request
<html><body>1609459686</body></html>
sleep 63
send request
<html><body>1609459749</body></html>
sleep 94
send request
<html><body>1609459843</body></html>
sleep 141
send request
<html><body>1609459985</body></html>
sleep 211
send request
<html><body>1609460196</body></html>
sleep 316
send request
<html><body>1609460513</body></html>
sleep 474
send request
<html><body>1609460987</body></html>
sleep 711
send request
<html><body>1609461699</body></html>
sleep 1066
send request
<html><body>1609462765</body></html>
sleep 1599
send request
<html><body>1609464364</body></html>
sleep 2398
```
You can see that the requests to the Pico W all succeeded, even with large intervals between requests.

@countrypaul can you please try the above `host.py` script to see if you can still reproduce your issue?  Also, maybe try on a more noisy/busy AP, because it could be that the additional traffic is keeping the Pico W awake, and in your tests with a dedicated AP there is no additional traffic and that's why you see the issue.

countrypaul, 2024-09-12:
@dpgeorge Have run the tests as requested with, as you can see, a very different result!
Not really sure what this means. It implies to me the client timedout - it was around the 300s mark as I was looking for the next iteration to start.
I am repeating the test just in case it was an anomally. These are on the dedicated AP. 

```
<html><body>1726148729</body></html>
sleep 141
send request
<html><body>1726148874</body></html>
sleep 211
send request
<html><body>1726149090</body></html>
sleep 316
send request
Traceback (most recent call last):
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\connection.py", line 199, in _new_conn
    sock = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\util\connection.py", line 85, in create_connection
    raise err
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\util\connection.py", line 73, in create_connection
    sock.connect(sa)
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\connectionpool.py", line 789, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\connectionpool.py", line 495, in _make_request
    conn.request(
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\connection.py", line 441, in request
    self.endheaders()
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\http\client.py", line 1331, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\http\client.py", line 1091, in _send_output
    self.send(msg)
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\http\client.py", line 1035, in send
    self.connect()
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\connection.py", line 279, in connect
    self.sock = self._new_conn()
                ^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\connection.py", line 208, in _new_conn
    raise ConnectTimeoutError(
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.HTTPConnection object at 0x000001BB32691CD0>, 'Connection to picowtest.local timed out. (connect timeout=None)')

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\requests\adapters.py", line 667, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\connectionpool.py", line 843, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\urllib3\util\retry.py", line 519, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='picowtest.local', port=80): Max retries exceeded with url: / (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x000001BB32691CD0>, 'Connection to picowtest.local timed out. (connect timeout=None)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\notwa\Documents\host.py", line 8, in <module>
    response = requests.get("http://picowtest.local/")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\requests\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\requests\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\requests\sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\requests\sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\notwa\AppData\Local\Programs\Python\Python312\Lib\site-packages\requests\adapters.py", line 688, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: HTTPConnectionPool(host='picowtest.local', port=80): Max retries exceeded with url: / (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x000001BB32691CD0>, 'Connection to picowtest.local timed out. (connect timeout=None)'))
```

countrypaul, 2024-09-12:
I have repeated the test twice more, each time I start with a hard reset of the Pico W having left it unplugged for 30 seconds, and all three tmes it has timed out when making the request after a sleep of 316 seconds (just over 5 mins) which suggeststo me this Pico W has a problem after about 5 mins of inactivity whilst waiting to accept a connection. I have no idea why, whether hardware or software?

countrypaul, 2024-09-13:
Found another Pico W that was purchsed at same time as the first but still in its sealed anti-static bag. After installing the correct uf2 file ( I installed the uf2 for just a Pico first for some unknown reason firstband wondered why it would not import network lib) I have run the test 3 times with almost identical results other than it passes the 316s and times out on the 474s sometimes. Both Picos were purchased in Feb 2023 from "The Pi Hut" in UK and are marked (c) 2022.

scruss, 2024-09-17:
> `network.hostname('PicoWTest.Local')`

while it's unlikely to be the root cause, don't add *.local* to the hostname. mDNS isn't quite working yet on the Pico W port

dpgeorge, 2024-09-18:
@countrypaul can you please try this PR: #15870, and then add the following line to your Pico W code, before the `wlan.active(1)`:
```py
wlan.config(trace=1)
```
That will allow you to see all async WLAN events, including association and deassociation with access points.  It may show something interesting around the 5 minute mark.

countrypaul, 2024-09-18:
@dpgeorge I have tried putting the wlan.comfig(trace=1) into the server code but it seems to make no difference, I am seeing no other output, just the same output as previously shown and certainly not the sort you illustate in PR:#15870

Not sure how much of the following is relevant/important but just in case  have included various details below.

For reference, I am using Thonny 4.1.1 on Windows 10 (Home) with Python 3.12.5 (tags/v3.12.5:ff3bc82, Aug  6 2024, 20:45:27) [MSC v.1940 64 bit (AMD64)]

After the client has timed out because the Pico W server has "hung" for want of a better expression, I can no longer interupt the PicoW and attemptsto do so in Thonny to do so result in the message:```

Unable to connect to COM4: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)

If you have serial connection to the device from another program, then disconnect it there first.

Process ended with exit code 1.
```  
I have to physically disconnect the Pico W from the PC and then reconnect to obtain any response fom the PicoW.

dpgeorge, 2024-09-18:
> Just realised I have overlooked what is probably an obvious oversight by myself. I took your instructions exactly as given without thinking, but now realise you have made a change to the source code so I presume I should be using a preview build of the firmware that includes your change.

You actually need to rebuild the firmware with the patch over at that PR.

To save you doing that I've attached the relevant firmware here:
[RPI_PICO_W-20240918-v1.24.0-preview.316.gc609177db.zip](https://github.com/user-attachments/files/17044735/RPI_PICO_W-20240918-v1.24.0-preview.316.gc609177db.zip)

So, please try that firmware with the `wlan.config(trace=1)` line.

> For reference, I am using Thonny 4.1.1

Hmm, it could be Thonny is interfering with the Pico W and causing the issue.

Can you also try using a different program to run the server code?  Eg mpremote: run `pip install mpremote` then `mpremote connect COM4 run server.py`.

countrypaul, 2024-09-18:
I tired with firmware : MicroPython v1.24.0-preview.315.g8feb714b4 on 2024-09-18; Raspberry Pi Pico W with RP2040

As I suspected no additional output, but two changes in behaviour:
1) The timeout period seems to have changed to 474 with this firmware and the call to wlan.comfig(trace=1) present.
2) Interupting from wihin Thonny does not always result in needing to unplug the PicoW

countrypaul, 2024-09-18:
I am working on the principle of making one change at a time and retesting several times before making another change.
I tested after changing from Thonny to mpremote, still on the preview firmware and still with the call to wlan.config(trace=1) present.
The client timeout varies more but is generally longer, 474, and 711 being common though 316 is probably most frequent wih the occasional one below 20s.  More interesting perhaps is that I have run most of the tests without having to restart the server.

I will update the firmware tomorrow to the patched version that should log some additional information.

countrypaul, 2024-09-19:
I updated to the patched firmware and tried the test gain, using mpremote to run the server -other than seeing some exra ouput at the start there was nothing else output to the console.:
```

C:\Users\notwa>
C:\Users\notwa>mpremote connect com4 run simpletest.py
Connecting...
[   18230] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   18231] ASYNC(0000,AUTH,0,0,0)
[   18233] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   18234] ASYNC(0000,ASSOC,0,0,0)
[   18234] ASYNC(0001,LINK,0,0,0)
[   18319] ASYNC(0000,JOIN,0,0,0)
[   18320] ASYNC(0000,SET_SSID,0,0,0)
[   18368] ASYNC(0000,PSK_SUP,6,0,0)
Connected 192.168.1.133
1609459229
```
The client stil timed out once it got to a 5minute wait:  
```

C:\Users\notwa>python host.py
send request
<html><body>1609459243</body></html>
sleep 2
send request
<html><body>1609459249</body></html>
sleep 3
send request
<html><body>1609459256</body></html>
sleep 4
send request
<html><body>1609459264</body></html>
sleep 6
send request
<html><body>1609459275</body></html>
sleep 9
send request
<html><body>1609459288</body></html>
sleep 13
send request
<html><body>1609459305</body></html>
sleep 19
send request
<html><body>1609459328</body></html>
sleep 28
send request
<html><body>1609459361</body></html>
sleep 42
send request
<html><body>1609459407</body></html>
sleep 63
send request
<html><body>1609459474</body></html>
sleep 94
send request
<html><body>1609459572</body></html>
sleep 141
send request
<html><body>1609459718</body></html>
sleep 211
send request
<html><body>1609459933</body></html>
sleep 316
send request
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\urllib3\connection.py", line 199, in _new_conn
    sock = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\urllib3\util\connection.py", line 85, in create_connection
    raise err
[... 42 lines cut ...]

Traceback (most recent call last):
  File "C:\Users\notwa\host.py", line 8, in <module>
    response = requests.get("http://PicoWTest.local/")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\requests\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\requests\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\requests\sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\requests\sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\requests\adapters.py", line 688, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: HTTPConnectionPool(host='picowtest.local', port=80): Max retries exceeded with url: / (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x000001D195529E80>, 'Connection to picowtest.local timed out. (connect timeout=None)'))

C:\Users\notwa>
```

I will repeat the tests to see if anything else shows up.

countrypaul, 2024-09-19:
I tried to repeat the test without restarting the server but the client always timed out without establihig a connection. I therefore stopped the server using ^C and simply attempted to restart but got the following:

```

C:\Users\notwa>mpremote connect com4 run simpletest.py
b''
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\Python312\Scripts\mpremote.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python312\Lib\site-packages\mpremote\main.py", line 528, in main
    handler_func(state, args)
  File "C:\Program Files\Python312\Lib\site-packages\mpremote\commands.py", line 220, in do_run
    _do_execbuffer(state, buf, args.follow)
  File "C:\Program Files\Python312\Lib\site-packages\mpremote\commands.py", line 187, in _do_execbuffer
    state.ensure_raw_repl()
  File "C:\Program Files\Python312\Lib\site-packages\mpremote\main.py", line 470, in ensure_raw_repl
    self.transport.enter_raw_repl(soft_reset=soft_reset)
  File "C:\Program Files\Python312\Lib\site-packages\mpremote\transport_serial.py", line 155, in enter_raw_repl
    raise TransportError("could not enter raw repl")
mpremote.transport.TransportError: could not enter raw repl
```
Power cycling the PicoW has allowed the server to restart and further testing.

countrypaul, 2024-09-20:
I think I have found part of the reason for the problem - the AP I have been using is a Siemens Gigaset SE587 WLAN dsl hub just being used as an AP & ethernet switch. It seemed to be the only thing that I had not changed and its use stemmed from the fact that the PicoW would not connect to my BT Hub probably because the BTHubt uses the same SSID for both 2.4 & 5GHz (plenty of instances of probems with this and workarounds). I found a very old (might be 20yrs old) 3Com OfficeConnect Wireless 11g Access Point 3CRWE454G72 stashed away and have tested with that. I am using the same software as previously, so again ony one change (plus ssid/password). There have been no timeouts, with sleep times increasing to ~1hr elapsed (I did not wait for the next one of 1.5hrs). Repeated the test including the hour wait with no timeouts. Tested by leaving the server without any request for about 10mins - the result no problem compared to failure when using the Siemens unit.

I suspect that there is a timeout of 5 minutes in the Siemens unit that is causing the PicoW problems in handling resulting in the PicoW 'locking up'. Nothing else connected to the Siemens unit has shown any problems so whether there is a bug in there that gives the PicoW problems is unclear to me. Regardless of whether the Siemens unit has any problems, the Pico should deal with the issue even if by reporting an error and stopping rather than "hangng" with no indication there is an issue.

countrypaul, 2024-09-21:
After more reading about how Wifi works, it appears that a client is timedout be the AP for inactivity after a preset time (often 5 minutes) which it appears more sophisticated APs allow to be set.  The is client timeout on the AP is triggered by there being no WiFi traffic from the client for the defined time, in which case the AP should send a disassociate message to the client. I am not in a position to determine if this is what is happening in the above case or not, not can I determine if the disassociate message is sent - but it would seem logical to me that the handling of this situation is causing the problem. My guess (as I have no real evidence and cannot check) is that perhaps the disassociate message is not sent so the PicoW does not realise what has happened and does not take appropriate action - but as said it is my guess.

scruss, 2024-09-21:
> ... perhaps the disassociate message is not sent so the PicoW does not realise what has happened and does not take appropriate action - but as said it is my guess.

A simpler conclusion is that your first router is unreliable. Where are you getting your reference that wifi has a 5 minute (or so) timeout?

I have repeated your code + Damien's client with sleep times of up to an hour, and all is working happily with my router.

countrypaul, 2024-09-22:
Rather than unreliable which to me implies some inconsistency I would say it may have faulty firmware as it is totally consistent - either way I would agree that it appears to have a problem.

I have read several articles that appear to have similar problems and refer to the Wifi timeout however they are not related to the PicoW. I wil admt I am jumping to a conclusion and as I said I cannot test this theory.  I am of the view that the router/AP "fouling up" should not cause a WiFi client to simply stop responding and requie a hard reset.

A couple of articles that are along these lines (the first is one of a number of lines that suggest a default 300s timeout):
https://www.cisco.com/c/en/us/td/docs/wireless/controller/8-5/config-guide/b_cg85/wlan_timeouts.pdf
https://github.com/arendst/Tasmota/discussions/12512
https://community.ui.com/questions/TimeOut-problems-with-AP-Wireless-/ec45c273-8ee5-4582-ad0d-2dc1a7fac0a1
https://community.cambiumnetworks.com/t/question-about-ap-wlan-profile-session-and-inactivity-timeout-settings/99638
https://forum.gl-inet.com/t/wifi-inactivity-timeout/30087

dpgeorge, 2024-09-22:
Good that you seemed to have found the cause of the 5 minute non-responsiveness, due to the AP inactivity setting.

> I am of the view that the router/AP "fouling up" should not cause a WiFi client to simply stop responding and requie a unit hard reset.

Yes I agree, the Pico W should not lock up.

Can you please give some more details on exactly how the lock up presents itself?  Please just use mpremote for the testing (Thonny may do other things that make it harder to debug).  You should be able to stop mpremote with Ctrl-C at any time, and rerun it, and it shouldn't lock up.  You should also always be able to do `mpremote connect <port> repl` and get a REPL, at which point you can use Ctrl-C to stop any running program on the Pico W, then Ctrl-B to get back to the normal REPL, if it's not already in that state.

countrypaul, 2024-09-22:
The problem shows up very easily and reproducibly with either mpremote or Thonny. If I start the server simpletest.py (using the patched firmware from yourself) I get the following output:

```
C:\Users\notwa>mpremote connect com4 run simpletest.py
Connecting...
[   27150] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   27151] ASYNC(0000,AUTH,0,0,0)
[   27154] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   27155] ASYNC(0000,ASSOC,0,0,0)
[   27155] ASYNC(0001,LINK,0,0,0)
[   27216] ASYNC(0000,JOIN,0,0,0)
[   27217] ASYNC(0000,SET_SSID,0,0,0)
[   27297] ASYNC(0000,PSK_SUP,6,0,0)
Connected 192.168.1.133
1609459237
```
If I leave the server for more than 5 minutes without anything connecting then it gets to the point where it is "hung up" - a client trying to connect to it simply times out.  No output seen on the mpremote process.  If a client is used (e.g. host.py) then once the sleep times exceeed 5 minutes (although in some cases it still seems to work with 316 seconds but not 474) the client times out.

Using ^C in mpremote works as you indicate it should (in Thonny it could be interupted but would result in the position that the only way to get the PicoW behaving was to power cycle the unit). I have seen similar effect in mpremote whereby it produces an error message suggesting another process has control of the usb port with the result disconnecting/reconnecting the PicoW was the ony solution (that I could find).

Whilst trying to work out what was the problem earlier, I added a socket.settimeout(30) and used a try except block to catch the timeout hen close then reopen the socket. The only function that appeared to use te timeout was the socket.accept function which is the only on that the process had to wait for remote input. This seemed to impact the process but if left for a longer period the client would need to try several times to effect a response and eventually even then it would just timeout.