micropython/micropython #15837: Can't allocat enough memory for deflate when stop asyncio

ARIYAWOW, 2024-09-12:
### Port, board and/or hardware

port-stm32

### MicroPython version

MicroPython v1.24.0-preview.224.g6c3dc0c0b.dirty on 2024-09-06; PYBv1.0 with STM32F405RG

### Reproduction

``` python
....
loop.stop()

import gc
gc.mem_free()
>>> 59264

import deflate   
f = open("firmware12.dfu.gz","rb")
f = deflate.DeflateIO(f, deflate.GZIP)
f.read(6)
>>>Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 32768 bytes
```

### Observed behaviour

actually，I want to flash firmware via fwupdate.py, but it report error like the code

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-09-12:
What does `gc.mem_free()` say after the line `f = deflate.DeflateIO(f, deflate.GZIP)`?

The `32768` byte allocation only happens for `f.read(6)` but at that point:

1. There might not be 59264 bytes free
2. That reported free memory may not constitute a single, contiguous region. See: https://docs.micropython.org/en/latest/reference/constrained.html#fragmentation

A `gc.collect()` might help...

ARIYAWOW, 2024-09-13:
> What does `gc.mem_free()` say after the line `f = deflate.DeflateIO(f, deflate.GZIP)`?
>
> The `32768` byte allocation only happens for `f.read(6)` but at that point:
>
> 1. There might not be 59264 bytes [...]

I agree with the second point，I tired `gc.collect()` , it worked well, but also report error
```python
>>> import gc
>>> gc.mem_free()
38960
>>> gc.collect()
>>> gc.mem_free()
61040
>>> import deflate
>>> f = open("firmware12.dfu.gz","rb")
>>> f = deflate.DeflateIO(f, deflate.GZIP)
>>> gc.mem_free()
60128
>>> gc.collect()
>>> gc.mem_free()
60784
>>> f.read(6)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 32768 bytes
```

ARIYAWOW, 2024-09-13:
I froze the code，my function worked fine

closed 2024-09-14 by ARIYAWOW