micropython/micropython #15841: Starting webrepl on ESP32S3 crashes with Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

gampam2000, 2024-09-12:
### Port, board and/or hardware

esp32 esp32s3 

### MicroPython version

MicroPython v1.24.0-preview.309.gf1bdac375.dirty on 2024-09-12; ESP32-S3
ESP-IDF release/v5.2 commit c8c6fd1

### Reproduction

```
import webrepl
webrepl.start(password='test')
```

### Expected behaviour

webrepl is started
return to the micropython >>> promt

### Observed behaviour

it crashes instantly after  `webrepl.start(password='test')` is called.
It seem it is depended of the ESP-IDF v5.2 version, with an older version of ESP-IDF release/v5.2 I did not observe this for example with commit id: e92f89ca33624b79784f01dc3c1312cbf649210b lib/esp-idf (v5.2.2-384-ge92f89ca33)

```
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382ed6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1108
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbdc
load:0x403cc700,len:0x2f4c
entry 0x403c989c
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4005544b  PS      : 0x00060030  A0      : 0x82036505  A1      : 0x3fcbcb20  
A2      : 0x00000000  A3      : 0x3c148b5c  A4      : 0x3fcbcb70  A5      : 0x3fcbcbc0  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x82037ba4  A9      : 0x3fcbcad0  
A10     : 0x00000000  A11     : 0x3fca56a0  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x3fca56c8  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x40055448:0x3fcbcb20 0x42036502:0x3fcbcb30 0x4203659a:0x3fcbcb70 0x42018b43:0x3fcbcbf0 0x4201f765:0x3fcbcc20 0x4201f82d:0x3fcbcc40 0x40378f42:0x3fcbcc60 0x42018adb:0x3fcbcd00 0x4201f765:0x3fcbcd30 0x40378e9f:0x3fcbcd50 0x42018adb:0x3fcbcdf0 0x4201f765:0x3fcbce20 0x4201f82d:0x3fcbce40 0x40378f42:0x3fcbce60 0x42018adb:0x3fcbcf00 0x4201f765:0x3fcbcf50 0x4201f77a:0x3fcbcf70 0x420535e3:0x3fcbcf90 0x4205393b:0x3fcbd020 0x42032e56:0x3fcbd050

ELF file SHA256: 85bda6bc4

Rebooting...
```

### Additional Information

```
xtensa-esp32s3-elf-addr2line -pfiaC -e ./build/micropython.elf 0x40055448:0x3fcbc7e0 0x42036502:0x3fcbc7f0 0x4203659a:0x3fcbc830 0x42018b43:0x3fcbc8b0 0x4201f765:0x3fcbc8e0 0x4201f82d:0x3fcbc900 0x40378f42:0x3fcbc920 0x42018adb:0x3fcbc9c0 0x4201f765:0x3fcbc9f0 0x40378e9f:0x3fcbca10 0x42018adb:0x3fcbcab0 0x4201f765:0x3fcbcae0 0x4201f82d:0x3fcbcb00 0x40378f42:0x3fcbcb20 0x42018adb:0x3fcbcbc0 0x4201f765:0x3fcbcbf0 0x4201f7d2:0x3fcbcc10 0x4203e255:0x3fcbcc50 0x4203e3e2:0x3fcbcc80 0x4201f8ac:0x3fcbcd70 0x403791cf:0x3fcbcdb0 0x42018adb:0x3fcbce50 0x4201f765:0x3fcbce80 0x4201f77a:0x3fcbcea0 0x420535e3:0x3fcbcec0 0x4205393b:0x3fcbcf50 0x42032e86:0x3fcbcf80

0x40055448: ?? ??:0
0x42036502: _getaddrinfo_inner at /smw-build/lib/micropython/ports/esp32/modsocket.c:256
0x4203659a: esp_socket_getaddrinfo at /smw-build/lib/micropython/ports/esp32/modsocket.c:946
0x42018b43: fun_builtin_var_call at /smw-build/lib/micropython/py/objfun.c:119
0x4201f765: mp_call_function_n_kw at /smw-build/lib/micropython/py/runtime.c:717
0x4201f82d: mp_call_method_n_kw at /smw-build/lib/micropython/py/runtime.c:733
0x40378f42: mp_execute_bytecode at /smw-build/lib/micropython/py/vm.c:1042
0x42018adb: fun_bc_call at /smw-build/lib/micropython/py/objfun.c:267
0x4201f765: mp_call_function_n_kw at /smw-build/lib/micropython/py/runtime.c:717
0x40378e9f: mp_execute_bytecode at /smw-build/lib/micropython/py/vm.c:957
0x42018adb: fun_bc_call at /smw-build/lib/micropython/py/objfun.c:267
0x4201f765: mp_call_function_n_kw at /smw-build/lib/micropython/py/runtime.c:717
0x4201f82d: mp_call_method_n_kw at /smw-build/lib/micropython/py/runtime.c:733
0x40378f42: mp_execute_bytecode at /smw-build/lib/micropython/py/vm.c:1042
0x42018adb: fun_bc_call at /smw-build/lib/micropython/py/objfun.c:267
0x4201f765: mp_call_function_n_kw at /smw-build/lib/micropython/py/runtime.c:717
0x4201f7d2: mp_call_function_0 at /smw-build/lib/micropython/py/runtime.c:691
 (inlined by) mp_parse_compile_execute at /smw-build/lib/micropython/py/runtime.c:1628
0x4203e255: do_load_from_lexer at /smw-build/lib/micropython/py/builtinimport.c:163
 (inlined by) do_load at /smw-build/lib/micropython/py/builtinimport.c:259
0x4203e3e2: process_import_at_level at /smw-build/lib/micropython/py/builtinimport.c:513
 (inlined by) mp_builtin___import___default at /smw-build/lib/micropython/py/builtinimport.c:608
0x4201f8ac: mp_import_name at /smw-build/lib/micropython/py/runtime.c:1533
0x403791cf: mp_execute_bytecode at /smw-build/lib/micropython/py/vm.c:1235
0x42018adb: fun_bc_call at /smw-build/lib/micropython/py/objfun.c:267
0x4201f765: mp_call_function_n_kw at /smw-build/lib/micropython/py/runtime.c:717
0x4201f77a: mp_call_function_0 at /smw-build/lib/micropython/py/runtime.c:691
0x420535e3: parse_compile_execute at /smw-build/lib/micropython/shared/runtime/pyexec.c:119
0x4205393b: pyexec_file at /smw-build/lib/micropython/shared/runtime/pyexec.c:682
 (inlined by) pyexec_file_if_exists at /smw-build/lib/micropython/shared/runtime/pyexec.c:694
0x42032e86: mp_task at /smw-build/lib/micropython/ports/esp32/main.c:147
```

### Code of Conduct

Yes, I agree

gampam2000, 2024-09-13:
further tests showed that it crashes when binding to a socket:

```
import usocket
socket = usocket.socket(usocket.AF_INET, usocket.SOCK_DGRAM)
socket.setblocking(False)
socket.setsockopt(usocket.SOL_SOCKET, usocket.SO_REUSEADDR, 1)
socket.bind(usocket.getaddrinfo("192.168.4.1", 53, 0, usocket.SOCK_DGRAM)[0][-1])
```

```
>>> import usocket
I (216016) modsocket: Initializing
>>> socket = usocket.socket(usocket.AF_INET, usocket.SOCK_DGRAM)
>>> socket.setblocking(False)
>>> socket.setsockopt(usocket.SOL_SOCKET, usocket.SO_REUSEADDR, 1)
>>> socket.bind(usocket.getaddrinfo("192.168.4.1", 53, 0, usocket.SOCK_DGRAM)[0][-1])
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4005544b  PS      : 0x00060030  A0      : 0x82036505  A1      : 0x3fcbccf0  
A2      : 0x00000000  A3      : 0x3c148b4c  A4      : 0x3fcbcd40  A5      : 0x3fcbcd90  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x82037ba4  A9      : 0x3fcbcca0  
A10     : 0x00000000  A11     : 0x3fca56a0  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x3fca56c8  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x40055448:0x3fcbccf0 0x42036502:0x3fcbcd00 0x4203659a:0x3fcbcd40 0x42018b43:0x3fcbcdc0 0x4201f765:0x3fcbcdf0 0x4201f82d:0x3fcbce10 0x40378f42:0x3fcbce30 0x42018adb:0x3fcbced0 0x4201f765:0x3fcbcf40 0x4201f77a:0x3fcbcf60 0x420535e3:0x3fcbcf80 0x420538d8:0x3fcbd010 0x42032eb0:0x3fcbd050

ELF file SHA256: 91619fbd9

Rebooting...
```
xtensa-esp32s3-elf-addr2line shows that it occures here:
`0x42036502: _getaddrinfo_inner at /smw-build/lib/micropython/ports/esp32/modsocket.c:256`

robert-hh, 2024-09-13:
Did you build the firmware yourself? 
I cannot find a esp-idf commit with the commit hash containing c8c6fd1. Did you check out the esp-idf v5.2.2 and updated the submodules after that?
Are you sure that your copy of the esp-idf follows https://github.com/espressif/esp-idf?

gampam2000, 2024-09-13:
Yes I built the firmware myself, it worked for many months and many builds.
I'm sure that I updated the submodules. But to be sure I updated it again, with the same error after the build.

I linked here the versions I use:

c8c6fd141377b9a48b29c8ebb51fb7efadde0566 lib/esp-idf (v5.2.2-707-gc8c6fd1413) 
https://github.com/espressif/esp-idf/commit/c8c6fd141377b9a48b29c8ebb51fb7efadde0566

f1bdac375240941edfc0aa04b646dc3c53d6b371 lib/micropython (v1.23.0-310-gf1bdac375)

If I switch to this older commit from July of the ESP-IDF I do not get this error:
e92f89ca33624b79784f01dc3c1312cbf649210b lib/esp-idf (v5.2.2-384-ge92f89ca33)
https://github.com/espressif/esp-idf/commit/e92f89ca33624b79784f01dc3c1312cbf649210b

agatti, 2024-09-13:
I thought the only supported versions of ESP-IDF were the ones coming off the git tag from Espressif's repository?

With `git checkout v5.2.2` the commit I get is https://github.com/espressif/esp-idf/commit/3b8741b172dc951e18509698dee938304bcf1523, and it doesn't seem to crash here (my network topology is probably different, though).

gampam2000, 2024-09-13:
I thought that all version of v5.2.2 where supported, but that's where I could be wrong.
So just to clarify this, the newest supported esp-idf version is this one: https://github.com/espressif/esp-idf/commit/3b8741b172dc951e18509698dee938304bcf1523 from May 31, 2024?

Is this also the version used for the automated preview builds?

robert-hh, 2024-09-13:
That is the version with the v5.2.2 tag.

agatti, 2024-09-13:
> So just to clarify this, the newest supported esp-idf version is this one: [espressif/esp-idf@3b8741b](https://github.com/espressif/esp-idf/commit/3b8741b172dc951e18509698dee938304bcf1523) from May 31, 2024?

In this case, yes.  In `ports/esp32/README.md` there are the git commands used to make sure you get a supported version.

> Is this also the version used for the automated preview builds?

I assume so, but I've always built my own :)

gampam2000, 2024-09-13:
ok than this issue is solved. With the tag v5.2.2 it works.

closed 2024-09-13 by gampam2000

dpgeorge, 2024-09-16:
Just to be clear, there is only one v5.2.2 version of the IDF.

A version with a tag like `v5.2.2-707-gc8c6fd1413` is not a release and is not the same as `v5.2.2`.

straga, 2024-11-07:
5.2.3 crash again

dpgeorge, 2024-11-11:
> 5.2.3 crash again
> fix - #15999

I can confirm that 5.2.3 crashes when calling `getaddrinfo`, and that setting `AI_CANONNAME` fixes it.