micropython/micropython #15844: Webserver Socket stops responding after a minute.

mispacek, 2024-09-13:
### Port, board and/or hardware

esp32 port, ESP32 and ESP32C3

### MicroPython version

### Port, board and/or hardware
ESP32 (lolin32-Lite, esp32 devkit v1)  and ESP32C3 (esp32c3 supermini)
both board have same issues

### MicroPython version
MicroPython v1.23.0 on 2024-06-02; ESP32C3 module with ESP32C3
in older versions than MicroPython v1.20.0 code works fine withou issues

MicroPython v1.19.1 works OK
MicroPython v1.20.0 works OK
MicroPython v1.21.0 NOT working
MicroPython v1.22.0 NOT working
MicroPython v1.23.0 NOT working

### Reproduction

Copy content of zip file to your ESP device. Set correct SSID and Password and reset your board.
[issue.zip](https://github.com/user-attachments/files/16995089/issue.zip)

The web server will offer you index.html, which cyclically send requests to server and server answers back.
in terminal you can see positions of virtual joysticks on webpage.
[image]

After about a minute, the server stops responding. There is enough free RAM, I don't know where the problem is, everything works correctly on older versions.

There is another version of the web server available in the Zip archive, without the use of threads, but the problem is exactly the same.

### Expected behaviour

I expect that the server will behave the same in all versions. It should respond indefinitely to web browser requests.

### Observed behaviour

In Chrome, using developer tools, I check the responses from the server. After 1-5 minutes, the server stops responding and sometimes restarts.
[image]

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

TRadigk, 2024-09-14:
Hi,

I've had the same issue and no solution for that. But since I already was going for seed xiao ESP32C3, I was able to replace it with an ESP32S3 (also seed xiao), which is a little bit more expensive but has the same footprint and mostly same pin configuration.

In my investigations I also found out:
long requests (256 Bytes) lead to immediate "hang", the device stops responding to "ping" requests. With wireshark I found out that the device always answers the HTTP ACK, but might not send data. If helpful I could provide this additional info in more detail.

xuancong84, 2024-09-15:
I also encounter the same problem on esp32c3, except that instead of one minute, the web server stops responding after a random period of time between a few minutes and one day or so. However, `ping` continues to work.

The randomness of this bug makes it very difficult to solve.

dpgeorge, 2024-09-16:
Thanks for the clear report and the reproduction.  Using the instructions in the original report, with ESP32_GENERIC firmware, IDF 5.2.2 and a recent master MicroPython `v1.24.0-preview-309-gf1bdac375` I was able to reproduce the issue: after around 1 minute of having the webpage open and accessing the ESP32 server, TCP connections would stop.  They would eventually resume after a few minutes.

This looks like the same root cause as with the issue #12819.  It's because TCP sockets are remaining open on the ESP32 server for up to 2 minutes.

I tested the fix suggested in #12819 which is to add `CONFIG_LWIP_TCP_MSL=6000` to `ports/esp32/boards/sdkconfig.base`  That fixes the issue for me, both the threading and non-threading version of the webserver (v1 and v2) work fine with that adjusted MSL (at least, I can run for much longer and don't see any issues).

I also made a simpler client program in Python to show the same issue, so you don't need to test with a web browser.  Just run this on the host PC:
```py
import time
import requests

IP = '<ip of esp32>'

for i in range(1000):
    r = requests.get(f'http://{IP}/*JOY;{i};0;0;0;0;0')
    r.close()
    time.sleep(0.15)
```
Eventually that will stop sending requests.

dpgeorge, 2024-09-16:
The problem does not happen on a PYBD-SF6 (and presumably all other bare-metal lwIP MicroPython boards) because those boards have a very limited amount of static memory allocated to lwIP, and lwIP is then forced to close and reuse sockets that are sitting in the TIME-WAIT state.

In particular, in the `tcp_alloc()`, the "tcp_alloc: killing off oldest TIME-WAIT connection" debug message is printed (and hence `tcp_kill_timewait()` run) on PYBD-SF6 but not on ESP32.

Ideally we need a way on ESP32 to limit the maximum number of opened PCBs, and force `tcp_alloc()` to reclaim PCBs when that limit is reached.  But I'm not sure how to do that.  Somehow we need to hook into the `memp_malloc(MEMP_TCP_PCB)` call and return NULL when the maximum number of PCBs is reached.

TRadigk, 2024-09-16:
Hi Damien,
I couldn't encounter the issue on ESP32 S3 (with 8 MB) of Ram in **1000** subsequent requests, where the ESP32 C3 hang up after like 20. Since you're changing something in the ESP32 port, should the issue also occur on the ESP32S3 but due to much more available RAM way later?

dpgeorge, 2024-09-19:
> should the issue also occur on the ESP32S3 but due to much more available RAM way later?

Probably not, because the TCP socket waits in TIME-WAIT state for 2 minutes, after which it is reclaimed.  So there's probably enough RAM to allocate a few thousand sockets before they need to be recycled, at which point the 2 minutes from the first sockets has passed and their memory is freed.

projectgus, 2024-09-26:
This looks like the same port reuse problem previously described here: https://github.com/micropython/micropython/issues/12819#issuecomment-1797336596 - the connection stalls once the same port is reused, and only unstalls later 

[image]

Going to try and reproduce this without MicroPython, as it seems like an ESP-IDF/LWIP TCP/IP stack bug...

projectgus, 2024-09-26:
(Sorry for duplicate post, previous screenshot was wrong.)

I can reproduce using Damien's client code and the provided files, and I see it getting stuck when the client reuses the same port number. Same as described here: https://github.com/micropython/micropython/issues/12819#issuecomment-1797336596

[image]

Eventually the existing socket with this port ends TIME-WAIT and everything starts working again.

dpgeorge, 2024-09-26:
Testing again with Wireshark enabled, I can confirm that a TCP port number is being reused, and that's exactly the point at which the esp32 server stops responding.  And the esp32 is ACK'ing the RST.

I actually tested by using `examples/network/http_server.py` as the server code on the esp32.  So it looks like the issue is very easy to reproduce with any server code.

projectgus, 2024-09-26:
I think the problem is that the socket isn't being properly closed on the MicroPython side.

The capture in the screenshot above is incomplete, but even on a complete capture I only see the client sending a FIN but the server (MicroPython) never sends a FIN packet back. This is despite the server calling close on the socket.

On the ESP-IDF http server the client initiates the close with a FIN and then the server sends a FIN back, and the problem doesn't reoccur.

There is a specific sequence to closing HTTP sockets that should be followed, although I think even a server which doesn't follow that sequence shouldn't be seeing this behaviour.

dpgeorge, 2024-09-26:
What's the correct way to close?  We call `lwip_close(fd)`.  Maybe we need to call the higher level function `close()`?

projectgus, 2024-09-26:
I think MicroPython is doing the right thing, it's just somehow LWIP is transtioning the socket to TIME-WAIT without ever sending a FIN packet... It's quite odd, it might be a race condition with the timing of the client sending its FIN and the server calling `close()` on the socket...

projectgus, 2024-10-01:
I've hidden the comments above because I was reading the WireShark output wrong, my mistake!

The underlying problem is the [same as described here](https://github.com/micropython/micropython/issues/12819#issuecomment-1797336596). The conditions are:

1. The server (MicroPython) closes the connection first. If the client closes the connection first, it doesn't enter TIME-WAIT on the server so the problem doesn't trigger. However if the server closes the connection first then it *does* enter TIME-WAIT.
2. The client opens a new connection with the same source port, before the TIME-WAIT has expired (two minutes I think, by default).

What happens then is that the client sends TCP SYN, the server sends TCP ACK, the client recognises the ACK is invalid because it belongs to the old connection and sends TCP RST, but the server ignores this ([required for RFC1337](https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/tcp_in.c#L745)). It will stay stuck like this until after the 2 minutes expires, although the client usually times out first.

## Reproduce without MicroPython

I can reproduce this in C if I modify the [tcp_server example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/tcp_server) to close the socket after it sends the first response back to the client.

ESP-IDF server patch:

```patch
diff --git i/examples/protocols/sockets/tcp_server/main/tcp_server.c w/examples/protocols/sockets/tcp_server/main/tcp_server.c
index 5f1c284960..b4ac5c97b3 100644
--- i/examples/protocols/sockets/tcp_server/main/tcp_server.c
+++ w/examples/protocols/sockets/tcp_server/main/tcp_server.c
@@ -59,6 +59,7 @@ static void do_retransmit(const int sock)
                 to_write -= written;
             }
         }
+        break;
     } while (len > 0);
 }

```

Client program to repro with tcp_server:

```python
#!/usr/bin/env python
import socket
import time

HOST="10.42.0.150"  # <-- Server IP goes here
PORT=3333

while True:
    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    s.settimeout(5)  # so fails when bug triggers
    s.connect((HOST, PORT))

    s.send(b"ABCD")
    print(time.time(), s.recv(4))
    s.close()
```

## Workarounds

I can't find a perfect workaround, but I think any of these would work:

* Set a timeout on the client side and retry when the connect times out. For example, [if doing this from JavaScript](https://developer.mozilla.org/en-US/docs/Web/API/XMLHttpRequest/timeout).
* Slow down the rate of requests so that ports aren't reused within a two minute window. It'd be a big change, but a long lived WebSocket could be a good fit for this kind of frequent data request.
* Implement HTTP/1.1 connection reuse for the server. (Another big change.)
* Have the client close the connection, not the server. Unfortunately I think this is only possible by implementing proper HTTP/1.1. The kind of server attached to this issue is an HTTP/1.0 style ["close-delimited response message"](https://www.rfc-editor.org/rfc/rfc9112.html#name-message-body-length), i.e. the server needs to Close the connection because this is how the client knows there is no more data. Doing it in the recommended HTTP/1.1 way means setting `Connection: close` and a `Content-Length:` header, so it's a lot more complex...

## Fixes?

Linux and OpenBSD servers don't have this problem because if they see an out-of-sequence SYN for a TIME-WAIT socket then they allow it as a new connection. [OpenBSD code](https://github.com/openbsd/src/blob/477ac4f17e6a6b377943ecf8e34bc324defc64a6/sys/netinet/tcp_input.c#L1226), [Linux code](https://github.com/torvalds/linux/blob/master/net/ipv4/tcp_minisocks.c#L173). 

We could request LWIP support be added to do the same thing. The issues I can see are:

* OpenBSD and Linux also check timestamps by default, which is recommended approach to avoid "TIME-WAIT assassination" (see RFC1337). LWIP doesn't implement the embedded timestamp option.
* The way LWIP is structured in the code then it's kind of fiddly to recycle a TIME-WAIT socket.
* Due to the above, I suspect Espressif would only add it to their esp-lwip fork if upstream LWIP agreed it was a good idea. And the upstream LWIP is not particularly active.

A simpler/quicker "fix" would to be to immediately clean up the socket if an RST is received in TIME-WAIT (i.e. [change this bit](https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/tcp_in.c#L751)). However this does open the possibility of "TIME-WAIT assassination", trading one bug for a worse one in some cases (old data arrives on new socket :scream_cat:).

projectgus, 2024-10-01:
@mispacek can you please try flashing this ESP32-C3 build and see if you can still reproduce the problem? 
[ESP32_GENERIC_C3_LWIP_RST_patch.zip](https://github.com/user-attachments/files/17201313/ESP32_GENERIC_C3_LWIP_RST_patch.zip).

(If anyone else wants to test on a different board, let me know.)

This *isn't a fix*, it's the hacky fix-with-othert-issues mentioned at the very end of the last comment. Patch is https://github.com/projectgus/esp-lwip/tree/hack-cleanup-time-wait-on-rst - but if this build fixes the problem then it at least confirms this is the issue.

ma261065, 2024-10-01:
Isn't rfc1337 saying that immediately cleaning up the socket if an RST is received in TIME-WAIT is the fix?

```
Fixes for TWA Hazards

   We discuss three possible fixes to TCP to avoid these hazards.

   (F1) Ignore RST segments in TIME-WAIT state.
```

projectgus, 2024-10-01:
> Isn't rfc1337 saying that immediately cleaning up the socket if an RST is received in TIME-WAIT is a fix for TWA, rather than 'opening up the possibility of TWA' as you said?

My understanding is that TIME-WAIT assassination is when the TCP stack cleans up the socket too early (the bad effect being that after this, additional out-of-band packets might be delivered to the application on a new socket). The fix of ignoring RST means ignoring the RST packet and doing nothing, rather than cleaning up the socket early.

projectgus, 2024-10-01:
One more contributing factor to this is that ESP-IDF doesn't limit the number of sockets in TIME-WAIT. The config option that should limit this is not effective (see https://github.com/espressif/esp-idf/issues/9670).

If that bug was fixed then this problem would happen much less often because TIME-WAIT sockets would be cleaned up more rapidly when there are a lot of short lived connections. However, I think the problem might still happen sometimes - in my packet captures sometimes Linux reuses the same source port very quickly, for whatever reason.

projectgus, 2024-10-02:
Candidate fix in the linked PR, and I've posted some builds there if anyone is able to test: https://github.com/micropython/micropython/pull/15952#issuecomment-2387676282

PR #15952 (2024-10-10): esp32: Apply the LWIP active TCP socket limit.
### Summary

This is a workaround for a bug in ESP-IDF where the configuration setting for LWIP maximum active TCP sockets (PCBs) is not applied. See https://github.com/espressif/esp-idf/issues/9670

Fixes cases where a lot of short-lived TCP connections can cause:

- Excessive memory usage (unbounded number of sockets in TIME-WAIT).
- Much higher risk of stalled connections due to repeated port numbers. The maximum number of active TCP PCBs is reduced from 16 to 12 to further reduce this risk.

This is not a 100% fix for the second point: a peer  [... cut]
closed 2024-10-10 by dpgeorge