micropython/micropython #15846: esp32 machine.enable_irq() not working

xuancong84, 2024-09-14:
### Port, board and/or hardware

esp32

### MicroPython version

up to commit f1bdac375240941edfc0aa04b646dc3c53d6b371

### Reproduction

machine.disable_irq();machine.enable_irq()

### Expected behaviour

The program can continue.

### Observed behaviour

The program cannot continue. After WDT timeout, system restarts.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2024-09-14:
I cannot replicate it.
`state=machine.disable_irq();machine.enable_irq(state)`
works.

xuancong84, 2024-09-14:
Sorry, I mean the following:
```
def g_dbg1():
	print('start')
	st_irq = machine.disable_irq()
	tm_til = ticks_us()+1000
	while ticks_us()<tm_til: pass
	machine.enable_irq(st_irq)
	print('stop')

def g_dbg2():
	print('start')
	tm_til = ticks_us()+1000
	while ticks_us()<tm_til: pass
	print('stop')
```

`g_dbg2()` will always run(). However, in the master commit firmware, `g_dbg1()` will always crash with WDT timeout. By default, `CONFIG_ESP_INT_WDT_TIMEOUT_MS` is set to 300ms, however, even waiting for 1ms (1000us) will cause it to crash.

robert-hh, 2024-09-14:
Confirmed. It works on a genuine ESP32, but fails on a ESPC3. Using the script below it works up to a value of >290000 (= 290ms), and reboots at 300000. But is fails when sleep_us() is replaced by sleep_ms() even with a value of 1.
```
from time import ticks_us, sleep_ms, sleep_us
import machine

def g_dbg1(t=100000):
    print('start')
    st_irq = machine.disable_irq()
    sleep_us(t)
    machine.enable_irq(st_irq)
    print('stop')

def g_dbg2():
    print('start')
    tm_til = ticks_us()+1000
    while ticks_us()<tm_til: pass
    print('stop')
```

xuancong84, 2024-09-14:
Thanks @robert-hh for testing and confirming!
It reboots at 300000us because by default `CONFIG_ESP_INT_WDT_TIMEOUT_MS` is set to 300, i.e., 300ms or 300000us or 0.3s. I can easily change `CONFIG_ESP_INT_WDT_TIMEOUT_MS` and rebuild. This is of no problem. The problem is that In my context, I cannot use `sleep_us(t)` or `sleep_ms(t)` as the time lapse due to code execution can be significant. Strangely, `ticks_us()` stops advancing after a while when IRQ is disabled.

robert-hh, 2024-09-14:
I tried a variant using a busy loop instead of calling ticks_us() or sleep_us() and toggled a Pin to tell, how fast the WDT fires. It was after like 85µs with that code and exactly for n=30.
```
def g_dbg1(n=10):
    p=machine.Pin(3, machine.Pin.OUT, value=0)
    print('start')
    st_irq = machine.disable_irq()
    p(1)
    i = n
    while i > 0:
        i -= 1
    p(0)
    machine.enable_irq(st_irq)
    print('stop')
```
Interestingly, if I change that code to add a `sleep_us()` into the while loop, the time is multiplied by the delay time of that sleep call, and the WDT fires still at 30 iterations of that loop. So if I call `sleep_us(1000)` in the loop, it fires after about 30ms.

Beside that, the WDT event that happens is `ESP_RST_INT_WDT`, the interrupt watchdog.

xuancong84, 2024-09-15:
I tried one step further and found one closer cause:
```
def g_dbg3(n:int):
	print('start')
	st_irq = machine.disable_irq()
	a=[ticks_us() for i in range(n)]
	machine.enable_irq(st_irq)
	print('stop')
	print('a=',a)
```
It seems that **while IRQ is disabled, calling `ticks_us()` for >=32 times will cause hang and trigger WDT reset** after `CONFIG_ESP_INT_WDT_TIMEOUT_MS=300ms`. I also tried to set `CONFIG_ESP_INT_WDT=n`, and it will hang forever.

agatti, 2024-09-17:
The 32 times bit is probably related to this: https://github.com/micropython/micropython/blob/8feb714b4d6bc29af3221c0abd47c4adb754f904/ports/esp32/mpconfigport.h#L87 as when the watchdog is disabled it gets stuck here:

```
#0  0x4038f658 in vPortYield () at /home/agatti/.local/esp-idf-504/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:327
#1  vPortYield () at /home/agatti/.local/esp-idf-504/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:310
#2  0x42009c50 in mp_thread_mutex_unlock (mutex=<optimized out>) at /home/agatti/src/micropython/ports/esp32/mpthreadport.c:225
#3  0x403834a6 in mp_execute_bytecode (code_state=code_state@entry=0x3fca7c50, inject_exc=inject_exc@entry=0x00000000 <mp_obj_base_t *>)
    at /home/agatti/src/micropython/py/vm.c:1361
...
```

So, interrupts are disabled, the scheduler is idle, and the interpreter got out of the GIL, no wonder it's waiting for something that won't happen :)

robert-hh, 2024-09-17:
Just strange that it works with a genuine ESP32.

agatti, 2024-09-17:
Ruling out an ESP-IDF issue, I'm wondering if it can be due to a difference on how things should be handled between Xtensa and RISC-V, or maybe because the C3 is single core and almost every other Sx MCU is dual core.

I wonder if this happens on an ESP32C6 as well, since that's both RISC-V and dual core.

xuancong84, 2024-09-17:
> Just strange that it works with a genuine ESP32.

Not just that, even on ESP32C3, this does not happen earlier on, i.e., before around 529dfd92a328ab28825c1d3ab8178dd41189fd02

agatti, 2024-09-17:
@xuancong84 is this commit the one that breaks your code 337742f6c70a7b9d407df687774bb9c9cc6a1656 ?

xuancong84, 2024-09-17:
> @xuancong84 is this commit the one that breaks your code: [337742f](https://github.com/micropython/micropython/commit/337742f6c70a7b9d407df687774bb9c9cc6a1656) ?

Yes, you are right. I have tested the commit before and after this one: before this commit it does not crash, after this one it crashes.

agatti, 2024-09-17:
That's where it gets stuck indeed.  Now, since it works on an ESP32-D0WDQ6 when built with BOARD_VARIANT=UNICORE, so it must be a RISC-V only thing.  I don't have a C2, C5, or a C6 to see if it is a C3-specific issue, but at least we ruled out the amount of cores being an issue.

Maybe @projectgus can look into this a bit more? :)

xuancong84, 2024-09-17:
Based on my limited understanding of this deadlock, I have tried a temporary fix (https://github.com/xuancong84/micropython/commit/f231bf83e6373c5015b128abd32eef7389d8e5f6) and it works out correctly. You can copy my patch if you think it is okay and cannot find a better fix.

Basically, I check whether IRQ is disabled or not before doing the forced context switch.

agatti, 2024-09-17:
@xuancong84 maybe just handle g_irq_state yourself in machine_enable_irq, as you've got no guarantees the interrupt state mask coming in will always be zero on all processors (xtensa _and_ rv32) :)

agatti, 2024-09-17:
@robert-hh thanks. I assumed ESP-IDF would guarantee the same behaviour across models for core things like these.

xuancong84, 2024-09-17:
> @xuancong84 maybe just handle g_irq_state yourself in machine_enable_irq, as you've got no guarantees the interrupt state mask coming in will always be zero on all processors (xtensa _and_ rv32) :)

Is there any predefined macros indicating whether the device is single-core or not? I think only on single-core chips, when IRQ is disabled, context switch should be forbidden. A user function disables IRQ typically because it wants to do something time critical, e.g., sending/receiving infra-red/RF433 pulses according to time intervals, so it makes sense here not to switch context to some other thread.

agatti, 2024-09-17:
There's `CONFIG_FREERTOS_UNICORE`, but in this case the problem only occurs on RV32 processors, as I tried on an Xtensa MCU with the second core disabled.

You're better off using `CONFIG_IDF_TARGET_ESP32C3` (and `CONFIG_IDF_TARGET_ESP32C6`) to limit the extent of your changes.

robert-hh, 2024-09-17:
Wouldn't that change just require a single flag telling whether IRQ was enabled or disabled? Besides that, it does not look like a good solution (sorry to say that). It does not address the root cause of the problem.
Given that commit 337742f fixed something that was more a rough edge than a bug it might be better to just roll back that commit.

xuancong84, 2024-09-18:
> There's `CONFIG_FREERTOS_UNICORE`, but in this case the problem only occurs on RV32 processors, as I tried that on an Xtensa MCU with the second core disabled.
> 
> You're better off using `CONFIG_IDF_TARGET_ESP32C3` (and `CONFIG_IDF_TARGET_ESP32C6`) to limit the extent of your changes.

1. Thanks @agatti and @robert-hh , I have a question. On a N-core chips, does it run N interpreters threads, one for each core?

2. In #15423, the author reported that the irregular time interval occurs only on MicroPython above v1.20, so what changes after v1.20 leads to this bad behaviour? I think the root cause is not addressed. Instead, it is worked around by adding a forced context switch ([337742f](https://github.com/projectgus/micropython/commit/337742f6c70a7b9d407df687774bb9c9cc6a1656)) which leads to the problem I am reporting here. So in addition to rolling back [337742f], we should find out the root cause of that issue right?

3. I still believe that if a user function disables IRQ, typically it is because it wants to do something time critical, e.g., sending/receiving infra-red/RF433 pulses according to **precise** time intervals, so it does make sense here not to switch context to another thread that might introduce **non-deterministic time elapses** that disrupts the resulting time precision.

agatti, 2024-09-18:
> I have a question. On a N-core chips, does it run N interpreters threads, one for each core?

I believe there's only one interpreter, pinned to a single core.  See `app_main` in `ports/esp32/main.c`.  Threads are also spawned on the same core as the interpreter (check out `mp_thread_create_ex` in `ports/esp32/mpthreadport.c`).

agatti, 2024-09-18:
Also, regarding whether this issue in its current form is unicore-specific or not, I start wondering whether I'm right in saying it's RV32-only.

Running a firmware built for `ESP32_GENERIC` with the `UNICORE` board variant maybe is not exactly the same thing as running the default firmware on a ESP32-S2.  That is a true single-core Xtensa MCU, and is not emulating a single-core execution environment as in the case I tested.  Sadly, I don't have such an MCU to test thing against, but maybe someone else does.

robert-hh, 2024-09-18:
> maybe is not exactly the same thing as running the default firmware on a ESP32-S2

No problem with a ESP32S2. Tested with a UM Feather S2.

agatti, 2024-09-18:
@robert-hh thanks for testing this on an S2!

With this bit of information then it may even be an ESP-IDF issue, as they've got a minor incompatibility across their two platforms for the same scenario.  I'll try this on my C3 with a few different ESP-IDF versions to see if anything changes.

xuancong84, 2024-09-18:
> With this bit of information then it may even be an ESP-IDF issue, as they've got a minor incompatibility across their two platforms for the same scenario. I'll try this on my C3 with a few different ESP-IDF versions to see if anything changes.

I have tested this on different versions of ESP-IDF already (v5.2.2, v5.0.4, and v5.0.7). It should not be due to ESP-IDF. However, ESP-IDF v5.2.2 results in much lesser free heap than v5.0.4 (30k+ vs 50k+) on exactly the same MicroPython webserver (I reflashed the MicroPython firmware only, all .mpy and config files are left unchanged).

agatti, 2024-09-18:
I'd argue it could still be, though.

One would expect that code that doesn't use model-specific peripherals or CPU-specific registers should behave the same regardless whether it is running on an Xtensa or on a RISC-V CPU.  If nothing else changes on your end except the MCU model you're building for and you see different behaviours, maybe the issue it's not exactly on your end (you still have to deal with it though).

Maybe reproducing the issue outside MicroPython could help to see if there's something MicroPython does that triggers the issue or not.  In the former case you've got more data to eventually figure out a cleaner fix, in the latter case you've got a minimal test case to create an issue on ESP-IDF with.

xuancong84, 2024-09-20:
The other thing I would like to report is that: similar to #15844 , after [337742f](https://github.com/projectgus/micropython/commit/337742f6c70a7b9d407df687774bb9c9cc6a1656), even without constantly pinging/accessing, webserver socket will stop responding in a few minutes or up to a day (random period of time) for my webserver. My temporal fix (https://github.com/xuancong84/micropython/commit/f231bf83e6373c5015b128abd32eef7389d8e5f6) also fixes this issue as well.

projectgus, 2024-09-25:
Thanks for the detailed report and investigations, @xuancong84! I've submitted a candidate fix in the linked PR, please take a look.

> the author reported that the irregular time interval occurs only on MicroPython above v1.20, so what changes after v1.20 leads to this bad behaviour? I think the root cause is not addressed.

The change is because the FreeRTOS kernel doesn't guarantee to context switch if task A releases a lock that task B is blocked on, where A and B have equal priority - and the GIL is a lock like this. Runtime may have been more equal in earlier FreeRTOS versions, but this is just was an implementation artifact not by design. The PR #15476 has some more explanation.

> One would expect that code that doesn't use model-specific peripherals or CPU-specific registers should behave the same regardless whether it is running on an Xtensa or on a RISC-V CPU.

The catch is that MicroPython was calling a "from ISR" port layer API to disable interrupts. My view is this breaks the "contract" between the FreeRTOS application and the kernel, because the kernel no longer knows it's in a critical section even though it is. On Xtensa this changes nothing, but on RISC-V it lead to the hang if you do this and then call `taskYIELD()`.

> However, ESP-IDF v5.2.2 results in much lesser free heap than v5.0.4 (30k+ vs 50k+) on exactly the same MicroPython webserver (I reflashed the MicroPython firmware only, all .mpy and config files are left unchanged).

That's a big difference! I think this is less bad than it seems, though - the difference is fragmentation not free heap.

ESP-IDF V5.0.4 on C3:

```
>>> micropython.mem_info()
stack: 516 out of 15360
GC: total: 64000, used: 1584, free: 62416, max new split: 147456
 No. of 1-blocks: 29, 2-blocks: 7, max blk sz: 18, max free sz: 3888
```

ESP-IDF V5.2.2:

```
>>> micropython.mem_info()
stack: 484 out of 15360
GC: total: 64000, used: 1584, free: 62416, max new split: 114688
 No. of 1-blocks: 29, 2-blocks: 7, max blk sz: 18, max free sz: 3888
```

This looks bad ("max new split" is much lower on 5.2.2), but "max new split" is showing the largest single free chunk of RAM, and there is often more than one large free chunk. If we measure the total free RAM for both then it's really close:

5.0.4:

```
>>> import esp32
>>> sum(h[1] for h in esp32.idf_heap_info(esp32.HEAP_DATA))
163256
```

5.2.2:

```
>>> import esp32
>>> sum(h[1] for h in esp32.idf_heap_info(esp32.HEAP_DATA))
163284
```

... if you have another measurement that looks different, or some code that runs out of memory on V5.2.2 but works fine on V5.0.4 then please open a new issue with the details.

robert-hh, 2024-09-25:
> I've submitted a candidate fix in the linked PR,

Where exactly? I do not see it in the linked PR.

agatti, 2024-09-25:
@robert-hh I believe it's #15910.

projectgus, 2024-09-25:
> Where exactly? I do not see it in the linked PR.

Have replied over on the PR: https://github.com/micropython/micropython/pull/15910#issuecomment-2375446782

xuancong84, 2024-09-27:
Thanks @projectgus !
Can explain to me what is the difference among `gc.mem_free()`, `micropython.mem_info()` and `esp32.idf_heap_info(esp32.HEAP_DATA)`. There are quite a number of functions that tells free memory, very confusing to me.

projectgus, 2024-10-01:
> Can explain to me what is the difference among gc.mem_free(), micropython.mem_info() and esp32.idf_heap_info(esp32.HEAP_DATA). There are quite a number of functions that tells free memory, very confusing to me.

Very happy to explain, but would start by recommending this documentation section: https://docs.micropython.org/en/latest/library/esp32.html#esp32.idf_heap_info - especially the "Note" box underneath.

Is it still unclear after reading this? If so, we should probably improve this documentation section.

PR #15910 (2024-10-10): esp32: Fix hang in taskYIELD() on riscv CPUs when IRQs disabled.
### Summary

 Closes #15846, fixing a regression introduced in #15476.

The hang occurs because the esp32 port was calling "from ISR" port-layer functions to set/clear the interrupt mask. FreeRTOS kernel therefore doesn't know the CPU is in a critical section. In `taskYIELD()` the riscv port layer [blocks after yielding until it knows the yield has happened](https://github.com/espressif/esp-idf/blob/v5.2.2/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c#L652), and would block indefinitely if IRQs are disabled (until INT [... cut]
closed 2024-10-10 by dpgeorge