micropython/micropython #15852: Guru Meditation Error

Karthikeyan-tuc, 2024-09-16:
### Port, board and/or hardware

esp32

### MicroPython version

 v1.24.0-preview.309.gf1bdac375.dirty on 2024-09-16

### Reproduction

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.24.0-preview.309.gf1bdac375.dirty on 2024-09-16
IDF version : v5.4-dev-2744-g59e1838270
Machine     : Generic ESP32 module with ESP32

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40001277  PS      : 0x00060830  A0      : 0x800d96dd  A1      : 0x3ffcf0b0  
A2      : 0x00000000  A3      : 0x3f401b84  A4      : 0x3ffcf100  A5      : 0x3ffcf150  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x800da434  A9      : 0x3ffcf060  
A10     : 0x00000000  A11     : 0x3ffc4138  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x3ffc4160  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40001274:0x3ffcf0b0 0x400d96da:0x3ffcf0c0 0x400d9772:0x3ffcf100 0x400df763:0x3ffcf180 0x400e6ab1:0x3ffcf1b0 0x400e6b79:0x3ffcf1d0 0x4008569e:0x3ffcf1f0 0x400df6fb:0x3ffcf290 0x400e6ab1:0x3ffcf2c0 0x400e7360:0x3ffcf2e0 0x4008564d:0x3ffcf350 0x400df6fb:0x3ffcf3f0 0x400e6ab1:0x3ffcf460 0x400e6b79:0x3ffcf480 0x4008569e:0x3ffcf4a0 0x400df6fb:0x3ffcf540 0x400e6ab1:0x3ffcf570 0x400855fb:0x3ffcf590 0x400df6fb:0x3ffcf630 0x400e6ab1:0x3ffcf660 0x400855fb:0x3ffcf680 0x400df6fb:0x3ffcf720 0x400e6ab1:0x3ffcf770 0x400e6ac6:0x3ffcf790 0x400f481b:0x3ffcf7b0 0x400f4b73:0x3ffcf840 0x400d8492:0x3ffcf870

ELF file SHA256: 24f435bf1

Rebooting...
ets Jul 29 2019 12:21:46

### Expected behaviour

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.24.0-preview.309.gf1bdac375.dirty on 2024-09-16
IDF version : v5.4-dev-2744-g59e1838270
Machine     : Generic ESP32 module with ESP32

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40001277  PS      : 0x00060830  A0      : 0x800d96dd  A1      : 0x3ffcf0b0  
A2      : 0x00000000  A3      : 0x3f401b84  A4      : 0x3ffcf100  A5      : 0x3ffcf150  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x800da434  A9      : 0x3ffcf060  
A10     : 0x00000000  A11     : 0x3ffc4138  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x3ffc4160  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40001274:0x3ffcf0b0 0x400d96da:0x3ffcf0c0 0x400d9772:0x3ffcf100 0x400df763:0x3ffcf180 0x400e6ab1:0x3ffcf1b0 0x400e6b79:0x3ffcf1d0 0x4008569e:0x3ffcf1f0 0x400df6fb:0x3ffcf290 0x400e6ab1:0x3ffcf2c0 0x400e7360:0x3ffcf2e0 0x4008564d:0x3ffcf350 0x400df6fb:0x3ffcf3f0 0x400e6ab1:0x3ffcf460 0x400e6b79:0x3ffcf480 0x4008569e:0x3ffcf4a0 0x400df6fb:0x3ffcf540 0x400e6ab1:0x3ffcf570 0x400855fb:0x3ffcf590 0x400df6fb:0x3ffcf630 0x400e6ab1:0x3ffcf660 0x400855fb:0x3ffcf680 0x400df6fb:0x3ffcf720 0x400e6ab1:0x3ffcf770 0x400e6ac6:0x3ffcf790 0x400f481b:0x3ffcf7b0 0x400f4b73:0x3ffcf840 0x400d8492:0x3ffcf870

ELF file SHA256: 24f435bf1

Rebooting...
ets Jul 29 2019 12:21:46

### Observed behaviour

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.24.0-preview.309.gf1bdac375.dirty on 2024-09-16
IDF version : v5.4-dev-2744-g59e1838270
Machine     : Generic ESP32 module with ESP32

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40001277  PS      : 0x00060830  A0      : 0x800d96dd  A1      : 0x3ffcf0b0  
A2      : 0x00000000  A3      : 0x3f401b84  A4      : 0x3ffcf100  A5      : 0x3ffcf150  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x800da434  A9      : 0x3ffcf060  
A10     : 0x00000000  A11     : 0x3ffc4138  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x3ffc4160  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40001274:0x3ffcf0b0 0x400d96da:0x3ffcf0c0 0x400d9772:0x3ffcf100 0x400df763:0x3ffcf180 0x400e6ab1:0x3ffcf1b0 0x400e6b79:0x3ffcf1d0 0x4008569e:0x3ffcf1f0 0x400df6fb:0x3ffcf290 0x400e6ab1:0x3ffcf2c0 0x400e7360:0x3ffcf2e0 0x4008564d:0x3ffcf350 0x400df6fb:0x3ffcf3f0 0x400e6ab1:0x3ffcf460 0x400e6b79:0x3ffcf480 0x4008569e:0x3ffcf4a0 0x400df6fb:0x3ffcf540 0x400e6ab1:0x3ffcf570 0x400855fb:0x3ffcf590 0x400df6fb:0x3ffcf630 0x400e6ab1:0x3ffcf660 0x400855fb:0x3ffcf680 0x400df6fb:0x3ffcf720 0x400e6ab1:0x3ffcf770 0x400e6ac6:0x3ffcf790 0x400f481b:0x3ffcf7b0 0x400f4b73:0x3ffcf840 0x400d8492:0x3ffcf870

ELF file SHA256: 24f435bf1

Rebooting...
ets Jul 29 2019 12:21:46

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2024-09-16:
Please provide information about what you tried to achieve best with an Python code example to reproduce the error. Without that help is unlikely.

dpgeorge, 2024-09-18:
```
MPY version : v1.24.0-preview.309.gf1bdac375.dirty on 2024-09-16
IDF version : v5.4-dev-2744-g59e1838270
```
It looks like you're using a modified version of MicroPython and an unsupported IDF version.  Please use standard MicroPython and IDF v5.2.2.

closed 2024-09-18 by dpgeorge

Karthikeyan-tuc, 2024-09-19:
> ```
> MPY version : v1.24.0-preview.309.gf1bdac375.dirty on 2024-09-16
> IDF version : v5.4-dev-2744-g59e1838270
> ```
> 
> It looks like you're using a modified version of MicroPython and an unsupported IDF version. Please use standard MicroPython and IDF v5.2.2.

Thank you for this information.custom firmware.bin created successfully and working good.