micropython/micropython #15858: PIO mov instruction does not have operation functions

sk8board, 2024-09-16:
### Port, board and/or hardware

rp2, Raspberry Pi Pico

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

### Problem Statement

The PIO mov instruction does not have operation functions

### Reproduction

### Documentation

Raspberry Pi RP2040 datasheet shows the mov instruction accepts three inputs: destination, operation, and source.

[image]

MicroPython only accepts two inputs and omits operation, as shown here:

https://github.com/micropython/micropython/blob/master/ports/rp2/modules/rp2.py#L165C9-L165C16

[image]

This means the important operations of the move instruction, like invert and reverse are not available in MicroPython.

Invert and reverse are key operations of PIO. 

### Expected behaviour

The expectation is for invert and reverse operations of the mov instruction within PIO to operate as designed by Raspberry Pi.

### Observed behaviour

Invert and reverse operations of the mov instruction within PIO are not available.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

rkompass, 2024-09-17:
It should have it.
It's documented: [here](https://docs.micropython.org/en/latest/library/rp2.html)
And iirc, I already had used it successfully.

The thing seems to be that `invert()` or `reverse()` are processed separately before going into the` mov() `processing.

sk8board, 2024-09-17:
Thank you for your reply. This is very helpful.

I would like to make a suggestion to help ensure others do not make this same mistake. The screenshots below show that finding this information is not intuitive. Maybe a large hyperlink could be implemented that is obvious to help enforce that additional PIO information is available at the hyperlink?

[image]

[image]

closed 2024-09-17 by sk8board