micropython/micropython #15868: PIC16 ports does not compile (without modification)

SirTobias, 2024-09-17:
### Port, board and/or hardware

PIC16

### MicroPython version

Micropython master branch, Microchip xc16-gcc v2.10 for PIC16

### Reproduction

1. Check out master branch
2. Cross-compile, if needed
3. Go to ports/pic16, either make or make BOARD=dsPICSK leads to the same issue

### Observed behaviour

With micropython (master branch, PIC16 port), XC16-gcc v2.10, I get on an Ubuntu 22.04 LTS:

with either ```make``` or ``m̀ake BOARD=dsPICSG```

`GEN build/genhdr/qstr.split                                                                                                                                                                                        
python3 ../../py/makeqstrdefs.py split qstr build/genhdr/qstr.i.last build/genhdr/qstr _                                                                                                                           
touch build/genhdr/qstr.split                                                                              
GEN build/genhdr/qstrdefs.collected.h                                                                      
python3 ../../py/makeqstrdefs.py cat qstr _ build/genhdr/qstr build/genhdr/qstrdefs.collected.h            
QSTR updated                                                                                               
GEN build/genhdr/qstrdefs.generated.h                                                                      
cat ../../py/qstrdefs.h qstrdefsport.h build/genhdr/qstrdefs.collected.h | sed 's/^Q(.*)/"&"/' | /opt/microchip/xc16/v2.10/bin/xc16-gcc -E  -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -I. -I../.. -Ibuild -I/opt/microchi
p/xc16/v2.10/include -I/opt/microchip/xc16/v2.10/support/dsPIC33F/h -Wall -Werror -std=gnu99 -nostdlib -mcpu=33FJ256GP506 -mlarge-code  -O1 -DNDEBUG - | sed 's/^\"\(Q(.*)\)\"/\1/' > build/genhdr/qstrdefs.preproc
essed.h                                                                                                                                                                                                            
python3 ../../py/makeqstrdata.py build/genhdr/qstrdefs.preprocessed.h > build/genhdr/qstrdefs.generated.h                                                                                                          
GEN build/genhdr/moduledefs.split                                                                          
python3 ../../py/makeqstrdefs.py split module build/genhdr/qstr.i.last build/genhdr/module _               
touch build/genhdr/moduledefs.split                                                                        
GEN build/genhdr/moduledefs.collected                                                                      
python3 ../../py/makeqstrdefs.py cat module _ build/genhdr/module build/genhdr/moduledefs.collected        
Module registrations updated                                                                               
GEN build/genhdr/moduledefs.h                                                                              
python3 ../../py/makemoduledefs.py build/genhdr/moduledefs.collected > build/genhdr/moduledefs.h           
GEN build/genhdr/root_pointers.split                                                                       
python3 ../../py/makeqstrdefs.py split root_pointer build/genhdr/qstr.i.last build/genhdr/root_pointer _   
touch build/genhdr/root_pointers.split                                                                     
GEN build/genhdr/root_pointers.collected                                                                   
python3 ../../py/makeqstrdefs.py cat root_pointer _ build/genhdr/root_pointer build/genhdr/root_pointers.collected
Root pointer registrations updated                                                                         
GEN build/genhdr/root_pointers.h                                                                                                                                                                                   
python3 ../../py/make_root_pointers.py build/genhdr/root_pointers.collected > build/genhdr/root_pointers.h 
mkdir -p build/extmod/                                                                                     
mkdir -p build/py/                                                                                         
mkdir -p build/shared/libc/                                                                                
mkdir -p build/shared/readline/                                                                            
mkdir -p build/shared/runtime/                                                                             
CC ../../py/mpstate.c                                                                                      
/opt/microchip/xc16/v2.10/bin/xc16-gcc  -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -I. -I../.. -Ibuild -I/opt/microchip/xc16/v2.10/include -I/opt/microchip/xc16/v2.10/support/dsPIC33F/h -Wall -Werror -std=gnu99 -nostdl
ib -mcpu=33FJ256GP506 -mlarge-code  -O1 -DNDEBUG -c -MD -MF build/py/mpstate.d -o build/py/mpstate.o ../../py/mpstate.c || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troublesho
oting\033[0m"; false)                                                                                                                                                                                              
In file included from ../../py/mpstate.h:33:0,                                                                                                                                                                     
                 from ../../py/mpstate.c:27:                                                               
../../py/misc.h: In function 'mp_clz_mpi':                                                                 
../../py/misc.h:383:5: error: size of unnamed array is negative ` 

When I comment out the assertion in py/misc.h:383, then it is OK. Has anybody have an idea?

Thank you very much in advance.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

agatti, 2024-09-18:
Weird, I've also had to deal with this recently.  This seems to be working for me:

```diff
diff --git a/ports/pic16bit/Makefile b/ports/pic16bit/Makefile
index 6d061514f..dda48fa3c 100644
--- a/ports/pic16bit/Makefile
+++ b/ports/pic16bit/Makefile
@@ -7,7 +7,7 @@ QSTR_DEFS = qstrdefsport.h
 include $(TOP)/py/py.mk
 include $(TOP)/extmod/extmod.mk
 
-XCVERSION ?= 1.35
+XCVERSION ?= 2.10
 XC16 ?= /opt/microchip/xc16/v$(XCVERSION)
 CROSS_COMPILE ?= $(XC16)/bin/xc16-
 
@@ -31,7 +31,7 @@ CFLAGS += -O1 -DNDEBUG
 endif
 
 LDFLAGS += --heap=0 -nostdlib -T $(XC16)/support/$(PARTFAMILY)/gld/p$(PART).gld -Map=$@.map --cref -p$(PART)
-LIBS += -L$(XC16)/lib -L$(XC16)/lib/$(PARTFAMILY) -lc -lm -lpic30
+LIBS += -L$(XC16)/lib -L$(XC16)/lib/$(PARTFAMILY) -lc99-elf -lm-elf -lc99-pic30-elf
 
 SRC_C = \
 	main.c \
diff --git a/ports/pic16bit/mpconfigport.h b/ports/pic16bit/mpconfigport.h
index a2d607fb2..d80f7edb9 100644
--- a/ports/pic16bit/mpconfigport.h
+++ b/ports/pic16bit/mpconfigport.h
@@ -93,7 +93,3 @@ typedef int mp_off_t;
 #define MICROPY_MPHALPORT_H "pic16bit_mphal.h"
 #define MICROPY_HW_BOARD_NAME "dsPICSK"
 #define MICROPY_HW_MCU_NAME "dsPIC33"
-
-// XC16 toolchain doesn't seem to define these
-typedef int intptr_t;
-typedef unsigned int uintptr_t;
diff --git a/py/misc.h b/py/misc.h
index 96b13c151..8ac80bb7b 100644
--- a/py/misc.h
+++ b/py/misc.h
@@ -380,6 +380,18 @@ static inline bool mp_check(bool value) {
 
 // mp_int_t can be larger than long, i.e. Windows 64-bit, nan-box variants
 static inline uint32_t mp_clz_mpi(mp_int_t x) {
+    #ifdef __XC16__
+    mp_uint_t mask = MP_OBJ_WORD_MSBIT_HIGH;
+    mp_uint_t zeroes = 0;
+    while (mask != 0) {
+        if (mask & (mp_uint_t)x) {
+            break;
+        }
+        zeroes++;
+        mask >>= 1;
+    }
+    return zeroes;
+    #else
     MP_STATIC_ASSERT(sizeof(mp_int_t) == sizeof(long long)
         || sizeof(mp_int_t) == sizeof(long));
 
@@ -389,6 +401,7 @@ static inline uint32_t mp_clz_mpi(mp_int_t x) {
     } else {
         return mp_clzll((unsigned long long)x);
     }
+    #endif
 }
 
 #endif // MICROPY_INCLUDED_PY_MISC_H
```

SirTobias, 2024-09-18:
Thank you, that worked. I also had to adapt the Makefile according to the changes regarding the C library.

closed 2024-09-18 by SirTobias