micropython/micropython #15877: Portenta H7 Ethernet does not connect

razvanphp, 2024-09-19:
### Port, board and/or hardware

stm32 on Portenta H7 + Vision Shield ETH

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Arduino Portenta H7 with STM32H747

### Reproduction

```
import network
import socket
import pyb

# Initialize the LAN interface with PHY parameters
lan = network.LAN()

lan.config(trace=8, low_power=False)
lan.ifconfig('dhcp')

# Activate the LAN interface
lan.active(True)

print("MAC Address:", lan.config("mac").hex())

# Wait until the link is up
while not lan.isconnected():
    pass

print("Network configuration:", lan.ifconfig())

# Get the IP address of www.google.com
try:
    ip_address = socket.getaddrinfo('www.google.com', 80)[0][-1][0]
    print("IP address of www.google.com:", ip_address)
except Exception as e:
    print("Error:", e)
```

### Expected behaviour

We should get the DHCP IP address.

### Observed behaviour

```
15:27:36.010 -> Traceback (most recent call last):
15:27:36.010 ->   File "main.py", line 9, in <module>
15:27:36.010 -> OSError: timeout waiting for DHCP to get IP address
15:27:37.495 -> MicroPython v1.23.0 on 2024-06-02; Arduino Portenta H7 with STM32H747
15:27:37.529 -> Type "help()" for more information.
15:27:37.529 -> >>> 
```

### Additional Information

I actually tried with setting a static IP instead, like this:
```
lan.ifconfig(('192.168.10.231', '255.255.255.0', '192.168.10.1', '8.8.8.8'))
```
And somehow it gets the DHCP lease successfully, but not using the specified IP.

It also works if I just call: `lan.ifconfig()`

but it does not work with: `lan.ifconfig('dhcp')`
```
15:39:14.146 ->   File "main.py", line 8, in <module>
15:39:14.204 -> OSError: timeout waiting for DHCP to get IP address
```

In `OpenMV v4.5.9` this trick does not work, I could not start the network.LAN there at all. CC @kwagyeman, not sure what is different, since the MP version is the same 🙁 

Can it be because of the MAC address? I see a generic one being assigned (`02:37:47:12:32:bb`).

This should be in another issue, but it would be nice to be able to set the MAC address:
```
mac_bytes = b'\xA8\x61\x0A\xAE\x44\xE0'
network.LAN.config(trace=8, low_power=False, mac=mac_bytes)
```

### Code of Conduct

Yes, I agree

kwagyeman, 2024-09-19:
@razvanphp - I can confirm that the latest firmware doesn't work.

Can you let Arduino know about the issue and they will assign us to fix it.

@iabdalkader - For visibility.

razvanphp, 2024-09-20:
Thanks for the quick answer!

> Can you let Arduino know about the issue and they will assign us to fix it.

Not sure how and where can I do that.

How else can I help? In which source code is the most probable way to check?

kwagyeman, 2024-09-20:
Start a thread on their forum:

https://forum.arduino.cc/c/hardware/portenta/pro-family/168

Or email their help support.

kwagyeman, 2024-09-20:
The issue is likely something buried in MicroPython as the Ethernet is completely controlled by that.

razvanphp, 2024-09-23:
> @razvanphp - I can confirm that the latest firmware doesn't work. This is a regression as this worked previously.

Not sure if this ever worked, I tried all those firmwares for H7 and all of them fail in the same way:
- ARDUINO_PORTENTA_H7-20220618-v1.19.1.dfu
- ARDUINO_PORTENTA_H7-20230426-v1.20.0.dfu
- ARDUINO_PORTENTA_H7-20231005-v1.21.0.dfu
- ARDUINO_PORTENTA_H7-20240222-v1.22.2.dfu
- ARDUINO_PORTENTA_H7-20240602-v1.23.0.dfu
- ARDUINO_PORTENTA_H7-20240918-v1.24.0-preview.315.g8feb714b4.dfu

It's not possible to set a static IP (it gets `dhcp` instead) and when trying `dhcp` string as parameter, it throws the timeout error. If you point me into the right file, maybe I can figure it out myself and contribute this out, should it bere somewhere here:
https://github.com/micropython/micropython/blob/3ca01eccae528a74fec8b1731f472d41018e5225/extmod/network_lwip.c#L79-L84

razvanphp, 2024-09-23:
BTW, I also tried with:
```python
lan.ipconfig(dhcp4=False, addr4=('192.168.10.133', '255.255.255.0'))
```
but still the same behaviour...

I noticed that `ifconfig` is deprecated in favour of `ipconfig`, but those two documentations are not in sync:
1. https://docs.micropython.org/en/latest/library/network.html#network.AbstractNIC.ifconfig
2. https://docs.micropython.org/en/latest/library/network.LAN.html#network.LAN.ifconfig

kwagyeman, 2024-09-23:
@razvanphp - Yeah, it did previously work. Have you reported the issue to Arduino yet? We have a support contract with them to work on this. Once they assign us the work, we can fix it.

Note, we support the Nicla through our firmware. You should be pulling it from here: https://github.com/openmv/openmv/releases

Try firmware v4.5.3 and older: https://github.com/openmv/openmv/releases/tag/v4.5.2

MicroPython typically gets fixes our firmware receives afterwards as things can take a while to merge upstream.

razvanphp, 2024-09-24:
> Have you reported the issue to Arduino yet? We have a support contract with them to work on this. Once they assign us the work, we can fix it.

Yes, on the forum [here](https://forum.arduino.cc/t/portenta-h7-micropython-no-ethernet/1304342) and also through contact-us form.

Tried also another older version of OpenMV, same thing:
```
OpenMV v4.5.2; MicroPython v1.20-omv-r22; PORTENTA-STM32H747
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 9, in <module>
OSError: timeout waiting for DHCP to get IP address
>
```
```
OpenMV v4.5.1; MicroPython v1.20-omv-r17; PORTENTA-STM32H747
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 9, in <module>
OSError: timeout waiting for DHCP to get IP address
>
```

the difference is that now, OpenMV behaves like vanilla micropython, in the sense that it works with (only) DHCP, if passing nothing to `lan.ifconfig()`, but when passing a tuple for a static IP, it still gets DHCP, so I think this is a bug upstream anyway.

kwagyeman, 2024-09-24:
Ah, I never checked this with non dhcp. Yes, probably a big upstream. We don't touch the Ethernet code on the OpenMV Cam.

razvanphp, 2024-09-28:
Ok, mistery solved!

It seems that the order of `lan.active(True)` and `lan.ifconfig([tuple])` matters, in the sense that calling active resets the interface to DHCP:
https://github.com/micropython/micropython/blob/17d82344581ad3a76033fae54c5d3304e17f185f/ports/stm32/network_lan.c#L73-L90
https://github.com/micropython/micropython/blob/17d82344581ad3a76033fae54c5d3304e17f185f/ports/stm32/eth.c#L852-L864
https://github.com/micropython/micropython/blob/17d82344581ad3a76033fae54c5d3304e17f185f/ports/stm32/eth.c#L777-L801

Not sure if adding this to documentation would be beneficial, but this was a waste of time for everybody, I'm sorry for that!

closed 2024-09-28 by razvanphp

kwagyeman, 2024-10-02:
@razvanphp - Thanks for debugging this. You should send a documentation update to MicroPython about this for their Ethernet module: https://github.com/micropython/micropython/blob/master/docs/library/network.LAN.rst