micropython/micropython #15892: Unable to use both SD card and display when they share SPI pins. (Erratic behavior from both)

echo-lalia, 2024-09-22:
### Port, board and/or hardware

ESP32 (Lilygo TDeck with ESP32-S3)

### MicroPython version

Tested with:
- MicroPython v1.22.2 on 2024-02-22; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
- MicroPython v1.23.0 on 2024-06-02; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
- MicroPython v1.23.0 on 2024-06-02; Generic ESP32S3 module with ESP32S3
- MicroPython v1.24.0-preview 335 gb08ddbba5

### Reproduction

I know this is a lot, but the behavior of the issue is hard for me to describe, so I tried writing a short script that can demonstrate the issues I'm having, with multiple tests:

``` Py
from machine import SPI, Pin, SDCard
import framebuf
import random
import os

# ~~~~~~~~~~~~~~ SETUP: ~~~~~~~~~~~~~~

# Enable TDeck peripherals:
enable = Pin(10, Pin.OUT, value=1)

# spi pins:
sck = Pin(40)
miso = Pin(38)
mosi = Pin(41)

# sdcard select
sdcard_cs = Pin(39)

# display pins
display_cs = Pin(12, Pin.OUT, value=1)
display_dc = Pin(11, Pin.OUT, value=1)
backlight = Pin(42, Pin.OUT, value=1)

# dummy/test image:
fbuf = framebuf.FrameBuffer(bytearray(320*240*2), 320, 240, framebuf.RGB565)

# helpers for the tests below:
def make_display():
    return SPI(1, baudrate=80_000_000, sck=sck, mosi=mosi, miso=miso)

def make_sd():
    return SDCard(slot=2, sck=sck, miso=miso, mosi=mosi, cs=sdcard_cs)

# Display must have already been initialized elsewhere for this simple function to work:
def write_to_display(display, text, deinit=True):
    fbuf.text(
        text,
        random.randint(0,300),
        random.randint(0,230),
        65535,
    )
    display_cs.off()
    display.init()
    display.write(fbuf)
    if deinit:
        display.deinit()
    display_cs.on()

# ~~~~~~~~~~~~~~ TEST0: ~~~~~~~~~~~~~~
def test0():
    """SD mounted after display init."""
    # sd first:
    sd = make_sd()
    display = make_display()

    # If this is the first time, OsError(16):
    # (If the code is run twice, it mounts correctly)
    os.mount(sd, '/sd')
    os.chdir("sd")
    # Display write does nothing:
    write_to_display(display, "testing!")
    # UnicodeError():
    print(os.listdir())
    # REPL may or may not also disconnect.
    

# ~~~~~~~~~~~~~~ TEST1: ~~~~~~~~~~~~~~
def test1():
    """SD read is corrupted when display is used."""
    # display first:
    display = make_display()
    sd = make_sd()

    # SDCard mounts successfully:
    os.mount(sd, '/sd')
    # Display writes successfully:
    write_to_display(display, "First write")
    os.chdir("sd")
    # Display writes successfully:
    write_to_display(display, "Second write")
    # UnicodeError():
    print(os.listdir())
    # REPL may or may not also disconnect.

# ~~~~~~~~~~~~~~ TEST2: ~~~~~~~~~~~~~~
def test2():
    """Display does nothing when not deinitialized."""
    # order does not matter:
    display = make_display()
    sd = make_sd()

    # SDCard mounts successfully:
    os.mount(sd, '/sd')
    # Display does nothing:
    write_to_display(display, "First write", deinit=False)
    os.chdir("sd")
    # Display does nothing:
    write_to_display(display, "Second write", deinit=False)
    # SDCard is read with no issue:
    print(os.listdir())

# ~~~~~~~~~~~~~~ TEST3: ~~~~~~~~~~~~~~
def test3():
    """SD can not be mounted after display deinit."""
    # display first:
    display = make_display()
    sd = make_sd()
    
    # display does nothing:
    write_to_display(display, "Testing!")
    # OSError(16):
    os.mount(sd, '/sd')

# ~~~~~~~~~~~~~~ TEST4: ~~~~~~~~~~~~~~
def test4():
    """SD can not be mounted after display deinit."""
    # sdcard first:
    sd = make_sd()
    display = make_display()
    
    # display writes successfully (usually):
    write_to_display(display, "Testing!")
    # OSError(16):
    os.mount(sd, '/sd')
    

test0()

```

### Expected behaviour

Regardless of difficulty, I would expect it to be somehow possible to use the SDCard and display alongside each other. However, I can not find any approach that makes the behavior stable, and so I think there may be something unexpected happening under the surface.

### Observed behaviour

If an SDCard object and an SPI object for the display, exist at the same time, one or both of them always seems to stop working in unexpected ways. The exact issues that pop up depend on the order the objects are created/accessed with.

If I `init()` and `deinit()` the display every time I write to it, then:
- If the SDCard is created before the display:
  - _The SDCard can not be mounted._  
  - _The display may or may not be written to successfully._

- If the display is created before the SDCard:
  - If the SDCard is mounted before writing to the display:
    - _The display can be written to with no issue, throws a `UnicodeError()` when reading SD contents._
  - If the display is written to before mounting the SDCard:
    - _The SDCard can not be mounted (`OSError(16)`)_

If I don't `deinit()` the display after writing to it, then:
- _The SDCard has no issues, but the display never updates._

*(Skipping the `init()` doesn't seem to do anything to this behavior, either.)*

### Additional Information

I'm using a Lilygo TDeck, which has all of the SPI devices (Specifically, an ST7789 display and a micro-sdcard reader) connected to the same SPI pins.

I did also test the above code using an M5Stack Cardputer (Which is the only other device I have to test this on, and unfortunately does have the same ESP32-S3 controller). And the behaviour seemed to be the same, except the `UnicodeError` never happens, and instead some garbled file names get printed.

### Code of Conduct

Yes, I agree

ricksorensen, 2024-09-22:
see discussion #12633 . 

Is the problem two different SPI drives (1&2) trying to control the same IO pins?  The python library `sdcard.py` takes an SPI instance and a CS as arguments instead of the pins and may help with the conflict.

echo-lalia, 2024-09-22:
> Is the problem two different SPI drives (1&2) trying to control the same IO pins? 

Thanks for the response! 

I *believe* that they are using the same SPI device, because the SPI IDs start at 1, and on the ESP32, the SDcard slots are 0-1 for the SD-specific hardware, and 2-3 use SPI ports. (Plus, if I used for example SPI 2 and SD slot 2, then I get an error that the device is already in use).

I appreciate you linking the discussion! For my project, I'm trying to support multiple devices, and so introducing an additional module that is only used by one device wouldn't be ideal. But, it's definitely better than nothing :)

peterhinch, 2024-09-22:
The 80MHz baudrate is very high. Have you tried reducing it?

echo-lalia, 2024-09-22:
> Have you tried reducing it?

Oh good point, sorry for leaving that out!

Yes the display alone works fine at 40mhz, and so I've tried it like that as well. The behavior didn't seem to change at all, so I put it back up to 80mhz and forgot about it when I made this issue.

sufyanaslam198, 2024-09-25:
I am running a similar setup to you but with a rp2040 (pico W) where 1 SPI bus is shared between a SD card and ST7789 display (seed studio round display). I do not have an esp32 to replicate your scenario but  my setup differs from yours in the following ways:

- I am using the sd card library from micropython lib https://github.com/micropython/micropython-lib/tree/394cbfc98a333dd1d4db35fb69379c72c30337f3/micropython/drivers/storage/sdcard
- I am explicitly turning the chip select pins high before initializing anything and after every SPI call. For example, after a write to display:
    `write_to_display(display, "First write", deinit=False)`
    `display_cs.high() `

Try testing the above and see if it helps.

echo-lalia, 2024-11-06:
> see discussion #12633 .

>  I am using the sd card library from micropython lib https://github.com/micropython/micropython-lib/tree/394cbfc98a333dd1d4db35fb69379c72c30337f3/micropython/drivers/storage/sdcard

I was able to get the SDCard working alongside the display by copying the micropython-lib sdcard module, and adding a `self.spi.init(baudrate=self.baudrate)` to every SPI write, (and doing the same in the display driver).

Thanks for the help everyone!

I'm still keeping the issue open because I feel the built-in SDCard implementation should be capable of working in this configuration. If any maintainers disagree, though, they can feel free to close it :)

peterhinch, 2024-11-07:
As I understand it this was a deliberate design decision. It is up to the application designer to implement bus sharing where required. This is to minimise code and maximise performance. CircuitPython adopts a different approach.

closed 2024-11-07 by peterhinch