micropython/micropython #15894: urequests.get causing Memory Fragmentation problem in RP2404 web server after 60h runtime

MartinH-Armadeus, 2024-09-22:
### Port, board and/or hardware

RP2040

### MicroPython version

MicroPython v1.24.0-preview.334.g7e14680a8 and  v1.23

### Reproduction

run frequent urequests.get() for more than 60 hours on a busy system

### Expected behaviour

The webser applicatio is expected to run infinetly

### Observed behaviour

My application is frequently (once a minute) doing a http request to my server. This is running fine for 50...60 hours, but then I reproducibly get an Exception:

```py
    def get_dect200_response(self):
        status_url = f'{self.FRITZBOX_URL}/webservices/homeautoswitch.lua?sid={self.sid}&switchcmd=getswitchstate&ain={self.AIN}'
        try:
            response = urequests.get(status_url)
            return response

        except OSError  as e:
            if e.errno == 12:  # ENOMEM
                print("m", end="")
                myserver.memory_errors += 1
                return None
            else:
                print("EXCEPTION Status Reqest: ",e.errno)
                return None

```
There is a timer interrupt running aside, which is running the     gc.collect()   every 2 seconds. 

Reguarly, I am getting  OSError 12, which I try to ignore. This already seems to ba a matter of some memory framentation.
But finally I the urequest.get() is getting an exception where the e.errno is None. In v1.23, it was similar, except the the urequests.get() did never return.
 

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-09-23:
Without the context of the rest of your program, or a minimum (and complete) reproduction of the issue, there's probably not much that can be done short of venturing some general advice about dealing with memory fragmentation.

Hammering `gc.collect()` will accomplish roughly nothing (you only do this for *performance* reasons, not memory reasons, ie: to control *when* the collect happens not *if*) and you should read and liberally apply the tricks detailed here: https://docs.micropython.org/en/latest/reference/constrained.html

dpgeorge, 2024-09-23:
If you are only getting `OSError` then it's not a memory fragmentation problem.

The main source of `OSError` from `requests.get()` would be from the lwIP TCP/IP stack.  It has a fixed amount of memory and never fragments, due to the way it's allocated.

You must take care to always close the results from the requests get.  Like `response.close()`, and wrap it in a try-finally.  Otherwise the socket remains open and that's a way to trigger `OSError`.

> But finally I the urequest.get() is getting an exception where the e.errno is None.

In this case please print the full error like `print(repr(er))` so we can see exactly where the error comes from.

sosi-deadeye, 2024-09-25:
Or use a context manager:

```python
class closing:
    """
    Context manager to call close() on object before leaving the context
    """
    def __init__(self, obj):
        self.obj = obj

    def __enter__(self):
        return self.obj

    def __exit__(self, exc_type, exc_obj, exc_tb):
        self.obj.close()
```

Use of the context manager:
```python
def get_dect200_response(self):
        status_url = f'{self.FRITZBOX_URL}/webservices/homeautoswitch.lua?sid={self.sid}&switchcmd=getswitchstate&ain={self.AIN}'

        try:
            with closing(requests.get(status_url)) as response:
                return response
        except: ...
```

Is it ok to add a context manager to the Response class? Actually it doesn't support the context manager protocol.