micropython/micropython #15900: PPP connection stuck when trying make simple HTTP request

VasylZakharko, 2024-09-24:
### Port, board and/or hardware

rp2 pico, rp2 pico w

### MicroPython version

master branch as 3e085c3aa67c8517678440d4b37754ee555bb809

### Reproduction

```
import machine, time, network, socket
import requests
import ntptime

def http_get_test():
   # Lookup the server address, for the given family and socket type.
   ai = socket.getaddrinfo(b"google.com", 80, socket.AF_INET, socket.SOCK_STREAM)
   print("Address infos:", ai)

   # Select the first address.
   ai = ai[0]
   print("Address info a[-1]:", ai[-1])
    
   try:
     s.connect(ai[-1])
     request = b"GET / HTTP/1.1\r\nHost: google.com\r\nUser-Agent: MicroPython\r\nConnection: close\r\n\r\n"
     print(s.send(request))
     while True:
        print(s.read(16)) # hangs here
   finally:
     s.close()

def ppp_test():
   uart = machine.UART(0, 115200)

   uart.write(b"atz\r\n")
   time.sleep(1)
   print(uart.readline())
   print(uart.readline())

   uart.write(b"at\r\n")
   time.sleep(1)
   print(uart.readline())
   print(uart.readline())

   uart.write(b"AT+CNMP=2\r\n")
   time.sleep(3)    
   print(uart.readline())
   print(uart.readline())
   
   uart.write(b'AT+CGDCONT=1,"IP","internet"\r\n')
   time.sleep(3)
   print(uart.readline())
   print(uart.readline())

   uart.write(b"ATDT*99#\r\n")
   time.sleep(3)
   print(uart.readline())
   print(uart.readline())

   modem = network.PPP(uart)
   modem.connect(security=network.PPP.SEC_PAP, user=b"", key=b"")

   while not modem.isconnected():
      print('waiting to connect')
      time.sleep(1)
      
   print('PPP Connected, IP:', modem.ifconfig())
   time.sleep(1)

   ntptime.settime() # sometimes getting timeout here
   
   rtc = machine.RTC()
   print(rtc.datetime())

   http_get_test() # hangs

ppp_test()
```

### Expected behaviour

Expected to return HTML data for http://google.com

### Observed behaviour

code stucks at reading data from socket:
```
try:
     s.connect(ai[-1])
     request = b"GET / HTTP/1.1\r\nHost: google.com\r\nUser-Agent: MicroPython\r\nConnection: close\r\n\r\n"
     print(s.send(request))
     while True:
         print(s.read(16)) # hangs here
finally:
     s.close()

```

### Additional Information

Added PPP config flag:
`#define MICROPY_PY_NETWORK_PPP_LWIP 1
`

### Code of Conduct

Yes, I agree

VasylZakharko, 2024-09-24:
Enabled some extra logging into pppos, maybe it could be helpful:
`
>>> ppp()
b'atz\r\r\n'
b'OK\r\n'
b'at\r\r\n'
b'OK\r\n'
b'AT+CGDCONT=1,"IP","internet"\r\r\n'
b'OK\r\n'
ppp phase changed[0]: phase=0
ppp_connect[0]: holdoff=0
ppp phase changed[0]: phase=3
pppos_connect: unit 0: connecting
ppp_start[0]
ppp phase changed[0]: phase=6
ppp_send_config[0]
pppos_send_config[0]: out_accm=FF FF FF FF
ppp_recv_config[0]
pppos_recv_config[0]: in_accm=FF FF FF FF
ppp: auth protocols: PAP=1 CHAP=0 CHAP_MD5=0
pppos_write[0]: len=24
ppp_start[0]: finished
pppos_input[0]: got 33 bytes
pppos_input[0]: Dropping ACCM char <13>
pppos_input[0]: Dropping ACCM char <13>
pppos_input[0]: Dropping ACCM char <10>
pppos_input[0]: Dropping ACCM char <13>
pppos_input[0]: Dropping ACCM char <10>
waiting to connect
pppos_input[0]: got 99 bytes
pppos_write[0]: len=12
pppos_input[0]: got 51 bytes
pppos_write[0]: len=28
ppp_netif_set_mtu[0]: mtu=1500
ppp_send_config[0]
pppos_send_config[0]: out_accm=0 0 0 0
ppp_recv_config[0]
pppos_recv_config[0]: in_accm=0 0 0 0
ppp phase changed[0]: phase=7
pppos_write[0]: len=10
pppos_input[0]: got 32 bytes
PAP authentication succeeded
ppp phase changed[0]: phase=9
pppos_write[0]: len=26
pppos_write[0]: len=18
waiting to connect
pppos_input[0]: got 34 bytes
pppos_write[0]: len=26
waiting to connect
waiting to connect
waiting to connect
waiting to connect
waiting to connect
pppos_write[0]: len=18
waiting to connect
pppos_write[0]: len=26
waiting to connect
waiting to connect
pppos_input[0]: got 38 bytes
pppos_write[0]: len=14
pppos_write[0]: len=26
pppos_input[0]: got 38 bytes
pppos_write[0]: len=8
Could not determine remote IP address: defaulting to 10.64.64.64
sifup[0]: err_code=0
local  IP address 10.109.25.152
remote IP address 10.64.64.64
primary   DNS address 10.74.32.5
secondary DNS address 10.74.227.5
ppp phase changed[0]: phase=10
PPP Connected, IP: ('10.109.25.152', '255.255.255.255', '10.64.64.64', '10.74.32.5')
pppos_netif_output[0]: proto=0xhx, len = 33
pppos_input[0]: got 81 bytes
ppp_input[0]: ip in pbuf len=76
Address infos: [(2, 1, 0, '', ('216.58.208.196', 80))]
Connect address: ('216.58.208.196', 80)
pppos_netif_output[0]: proto=0xhx, len = 33
pppos_input[0]: got 49 bytes
ppp_input[0]: ip in pbuf len=44
+-+-+-+-+-+-+-+-+-+-+-+-+-+- tcp_input: flags -+-+-+-+-+-+-+-+-+-+-+-+-+-+
tcp_parseopt: MSS
SYN-SENT: ackno 6510 pcb->snd_nxt 6510 unacked  6509
pppos_netif_output[0]: proto=0xhx, len = 33
pppos_netif_output[0]: proto=0xhx, len = 33
pppos_input[0]: got 49 bytes
ppp_input[0]: ip in pbuf len=44
+-+-+-+-+-+-+-+-+-+-+-+-+-+- tcp_input: flags -+-+-+-+-+-+-+-+-+-+-+-+-+-+
pppos_netif_output[0]: proto=0xhx, len = 33
pppos_input[0]: got 45 bytes
ppp_input[0]: ip in pbuf len=40
+-+-+-+-+-+-+-+-+-+-+-+-+-+- tcp_input: flags -+-+-+-+-+-+-+-+-+-+-+-+-+-+
tcp_receive: ACK for 6550, unacked->seqno 6510:6550
tcp_receive: removing 6510:6550 from pcb->unacked
pppos_write[0]: len=18
pppos_write[0]: len=18
pppos_input[0]: got 256 bytes
pppos_input[0]: got 32 bytes
pppos_write[0]: len=18
pppos_input[0]: got 256 bytes
pppos_input[0]: Dropping bad fcs 0xhx proto=0xhx
pppos_input_drop: pbuf len=288, addr 2000e000
pppos_input[0]: got 32 bytes
pppos_write[0]: len=18
pppos_input[0]: got 256 bytes
pppos_input[0]: Dropping bad fcs 0xhx proto=0xhx
pppos_input_drop: pbuf len=288, addr 2000e000
pppos_input[0]: got 32 bytes
pppos_write[0]: len=18
pppos_write[0]: len=18
pppos_write[0]: len=18
pppos_write[0]: len=18
IPV6CP: timeout sending Config-Requests
pppos_input[0]: got 256 bytes
pppos_input[0]: Dropping bad fcs 0xhx proto=0xhx
pppos_input_drop: pbuf len=288, addr 2000e000
pppos_input[0]: got 32 bytes
`

dpgeorge, 2024-09-25:
> pppos_input[0]: got 256 bytes
> pppos_input[0]: Dropping bad fcs 0xhx proto=0xhx

It looks like when it receives 256 bytes then the errors start to occur.  256 bytes is the default UART RX buffer size.  So I would assume that bytes are lost.

You need more UART buffering and/or hardware flow control enabled (which would probably require sending a special AT command to turn it on at the modem side).

Try `uart = machine.UART(0, 115200, rxbuf=1000)`.

VasylZakharko, 2024-09-25:
Hi Damien,

Thank you very much much for advise! I got sockets working with PPP!

Kind regards,
Vasyl.