micropython/micropython #15939: Web server stops responding - This site can't be reached

ploughpuff, 2024-09-30:
### Port, board and/or hardware

Raspberry Pi Pico W with RP2040

### MicroPython version

MicroPython v1.23.0 on 2024-06-02;

### Reproduction

1. Copy code from GitHub gist to new file: https://gist.github.com/ploughpuff/4a23743b8bc8dc6c0eda9448902c8cc7
2. Replace the SSID and PASSWD with your own WiFi AP details
3. Run code and wait for local IP address to be revealed (e.g. 192.168.x.x)
4. From a computer on the same LAN, enter address into your favourite browser.  Observe page refreshing once per second
5. From another computer on the same LAN, or your phone connected to same WiFi AP, enter address into browser.  Observe page refreshing once per second.

Demo code taken mostly from: https://electrocredible.com/raspberry-pi-pico-w-web-server-asynchronous-micropython/

### Expected behaviour

Both browsers should refresh the page contents indefinitely.

### Observed behaviour

Both browsers refresh the page ok to start with, but after a couple of minutes they stop and both browsers display "This site can’t be reached".

_Note: Actual displayed text is browser dependent.  I use Brave (PC) and Chrome (Mobile)._

Screen grab of Thonny and Browser after issues occurs:
[image]

### Additional Information

Pico-W is connected and powered via USB from my computer.
Computer is hardwired to (broadband provider) PlusNet modem/router.
Pico-W connects via WiFi to same PlusNet modem/router.
Mobile connects via WiFi to same PlusNet modem/router.

### Code of Conduct

Yes, I agree

bixb922, 2024-10-02:
I believe you need to call ```.close() ``` before ```.wait_closed()```

If you need a web server, consider using Microdot https://github.com/miguelgrinberg/microdot/
Microdot is quite small and does a good job

projectgus, 2024-10-03:
> I believe you need to call `.close() ` before `.wait_closed()`

Oddly enough, the `wait_closed()` implementation in MicroPython currently does call `close()`: https://github.com/micropython/micropython/blob/17d82344581ad3a76033fae54c5d3304e17f185f/extmod/asyncio/stream.py#L19-L21

However I agree in this case it is actually better to replace `wait_closed()` with `close()`, in case this changes in the future. This kind of very simple HTTP Server is sending a "close-delimited response body" (HTTP/1.0 style), so the server needs to close the socket or else the client won't know when the response is finished.

I don't think this is the root cause for this problem, though.

dpgeorge, 2024-10-03:
Thanks for the bug report.

I tried to reproduce the issue but cannot.  Here is my set up:
- Pico W board running MicroPython v1.23.0 on 2024-06-02.
- That board, a PC and a phone all connected to the same LAN.
- Running the code from the gist in the OP.
- The PC and mobile accessing the Pico W's webpage, refreshing automatically every second.

I ran it for 1 hour without issue, both browsers continue to access the page.

-------------

I see from your screenshot, the REPL output in the bottom right corner, the following:
- Your browsers seem to access favicon.ico many times, not just once for the first HTTP request.
- The log stops with an empty "Request" line.

I think the last point shows the issue: the first `await reader.readline()` returns the empty bytes object which means that the connection was closed (without sending any data).  Then the server code gets stuck forever in the "Skip HTTP request headers" because no data will ever be returned from `readline()`.

The above needs to happen a few times (4 with the default backlog of 5) before the server is blocked completely from new connections.

I'm able to force the above situation by manually opening a connection to the Pico W and immediately closing it.  That leads to a message `Request: b''` at the REPL and if I do this 4 times then the PC/phone accessing the Pico W do issue a timeout message.

So the server code needs to be improved so that it doesn't sit forever reading headers when the port has been closed.  You could try simply breaking out of that waiting loop if `b''` is returned from `readline()`.

dpgeorge, 2024-10-03:
Note that the networking examples https://github.com/micropython/micropython/blob/master/examples/network/http_server.py and https://github.com/micropython/micropython/blob/master/examples/network/https_server.py in this repository do check for `b''` return value from `readline()`:
```py
                while True:                                                    
                    h = client_s.readline()                                    
                    if h == b"" or h == b"\r\n":                               
                        break                                                  
                    print(h)                                                   
```
(Although these examples are not asyncio based, but the same concept will work with asyncio.)

ploughpuff, 2024-10-03:
Amazing support thank you everyone.  I modified the skip headers loop as suggested and have not reproduce my issue in 1 hr of testing.  Reverting that one change and my issues comes back after a couple of mins, so I've good confidence we found the problem.

I did spend the last couple of days playing with microdot and utemplate but my webpage only needs to display a few temperatures and humidity values so would prefer to keep the page I have created and spend time on the control strategy and not web servers.

Once again thank you.

projectgus, 2024-10-08:
@ploughpuff Sounds like this has been taken care of, so I'm going to close this issue now. Thanks for the confirmation!

closed 2024-10-08 by projectgus