micropython/micropython #15943: `natmod`: allow static `.bss` variables

vshymanskyy, 2024-10-01:
### Description

The [natmod documentation](https://docs.micropython.org/en/latest/develop/natmod.html) says:
> static BSS variables are not supported; workaround: use global BSS variables

This is widely used in existing code bases and requires [a lot of patching](https://github.com/vshymanskyy/lbm-micropython/blob/main/extra/lbm-4.5.0.patch).

However, this limitation only affects certain platforms:

```
┌──────┬───────┬───────┬──────────┬──────────┬─────────────┬─────────────┬──────────┬─────────────┐
│      │ x86   │ x64   │ armv6m   │ armv7m   │ armv7emsp   │ armv7emdp   │ xtensa   │ xtensawin   │
├──────┼───────┼───────┼──────────┼──────────┼─────────────┼─────────────┼──────────┼─────────────┤
│ test │ OK    │ x     │ x        │ x        │ x           │ x           │ OK       │ OK          │
└──────┴───────┴───────┴──────────┴──────────┴─────────────┴─────────────┴──────────┴─────────────┘
```

Maybe it's possible to improve `mpy_ld` to get rid of this limitation. This would greatly simplify things.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-10-02:
+1

Though I think presently natmod has no concept of setup or teardown. At runtime there needs to be some means to allocate, initialise and - crucially - relocate static storage since there is - afaik - no way on these embedded platforms of referring to it in a position-independent manner?

vshymanskyy, 2024-10-02:
I'm not aware of such limitations, i was able malloc and free memory as usual.

Gadgetoid, 2024-10-02:
It's a bit more complicated than that. (excuse the train of thoughtestness of this, since I'm plumbing the very depths of my memory about how we (mostly @Daft-Freak) tackled this on 32blit)

I was wrong about there being "no concept of setup", it's right there in the natmod docs that MicroPython requires natmods to include a GOT (Global Offset Table) which is patched at load time to make sure your natmod can find its own functions and data (or handle long jumps aiui) from wherever it happens to be loaded into RAM.

The problem - I think - is that data and static .bss are not PIC by default (and are *not* included in the GOT) and that the magic `-mno-pic-data-is-text-relative` switch (which we use successfully for the 32blit SDK, where PIC user code is loaded and patched at runtime much like in MicroPython but... also not) might not work in all circumstances.

As such, those sections cannot be reloc patched and initialised (because MicroPython has no knowledge of their existence) and thus your program will go looking in the wrong place.

This stuff must be supported both by `mpy_ld` *and* by the import machinery, *and* requires the compiler to be able to produce PIC data/static .bss.

It's mentioned in passing here: https://github.com/orgs/micropython/discussions/12811#discussioncomment-7399671

I suspect the reason it hasn't been attempted is somewhere between "it's incredibly heady stuff that nobody (including me) actually understands" and "it's (maybe?) not possible to support on *all* platforms therefore it's easier to support on none."

That said, I'm with you on this- it's annoying that it's not supported (the limitations made me bounce off natmod pretty quickly because it effectively excludes C++ and/or using Pico SDK features) and since even that above comment is a year old, it might be time to try reevaluating how possible it is.

I guess my first steps to exploring what's up here would be to:

* undo your patches
* build with `-mno-pic-data-is-text-relative`
* see what `mpy_ld` has to say about this, and look about patching in support for data/static .bss if it's not just... implicit somehow
* navigate https://github.com/micropython/micropython/blob/master/py/persistentcode.c (where *I think* the actual relocation patching and data initialisation happens) and ensure data/static .bss are allocated and initialised

vshymanskyy, 2024-10-02:
[static-bss-test.zip](https://github.com/user-attachments/files/17235603/static-bss-test.zip)

```log
--- Building x86 ---
init...
testing operations...
GLOBAL: global buffer
STATIC: static buffer
STATIC LOCAL: static local buffer

--- Building x64 ---
init...
testing operations...
GLOBAL: global buffer
STATIC: static buffer
STATIC LOCAL: static local buffer

--- Building armv6m ---
--- Plug armv6m board and press enter ---
init...
testing operations...^C                                    <<<< hangs
--- Building armv7m ---
--- Plug armv7m board and press enter ---
init...
testing operations...
GLOBAL: global buffer
STATIC: static buffer
STATIC LOCAL: static local buffer

--- Building armv7emsp ---
--- Plug armv7emsp board and press enter ---
init...
testing operations...
GLOBAL: global buffer
STATIC: static buffer
STATIC LOCAL: static local buffer

--- Building armv7emdp ---
--- Plug armv7emdp board and press enter ---
init...
testing operations...
GLOBAL: global buffer
STATIC: static buffer
STATIC LOCAL: static local buffer

--- Building xtensa ---
--- Plug xtensa board and press enter ---
init...
testing operations...
GLOBAL: global buffer
STATIC: millis: %d                                         <<<< unrelated string
STATIC LOCAL: static local buffer

--- Building xtensawin ---
--- Plug xtensawin board and press enter ---
init...
testing operations...
GLOBAL: global buffer
STATIC: millis: %d                                         <<<< unrelated string
STATIC LOCAL: static local buffer

```

vshymanskyy, 2024-10-02:
So the status of actually running such modules is:
```
┌──────┬───────┬───────┬──────────┬──────────┬─────────────┬─────────────┬──────────┬─────────────┐
│      │ x86   │ x64   │ armv6m   │ armv7m   │ armv7emsp   │ armv7emdp   │ xtensa   │ xtensawin   │
├──────┼───────┼───────┼──────────┼──────────┼─────────────┼─────────────┼──────────┼─────────────┤
│ test │ OK    │ OK    │ x        │ OK       │ OK          │ OK          │ x        │ x           │
└──────┴───────┴───────┴──────────┴──────────┴─────────────┴─────────────┴──────────┴─────────────┘
```