micropython/micropython #15944: FrameBuffer Initialisation fails with ValueError on `memoryview`

corranwebster, 2024-10-01:
### Port, board and/or hardware

RP2; Unix (but I believe this affects any board with framebuffer support)

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

### Reproduction

Run the code below in the REPL, or run a script with the same code in it:
``` python
import array
import framebuf
buf = array.array('H', bytearray(2*8*8))
sub_buf = memoryview(buf)[4:]
fbuf = framebuf.FrameBuffer(sub_buf, 4, 8, framebuf.RGB565, 8)
```

### Expected behaviour

I expected the code to work without error.

### Observed behaviour

I get an unexpected `ValueError` with no explanation:
``` python-console
>>> import array
>>> import framebuf
>>> buf = array.array('H', bytearray(2*8*8))
>>> sub_buf = memoryview(buf)[4:]
>>> fbuf = framebuf.FrameBuffer(sub_buf, 4, 8, framebuf.RGB565, 8)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: 
```

### Additional Information

The "what am I actually trying to do" here is to create a frame buffer on a subsection of another buffer using a memory view to avoid copying data, either:
- to clip out a sprite from a larger sprite-sheet buffer to display using `FrameBuffer.blit`
- to clip drawing to a sub-region of a larger image (for this you _have_ to have a memory view into the same underlying buffer)

This is likely related to, but different from #15571, as that is an issue with bit-alignment.  As with that issue, I think that the lines at fault are here: https://github.com/micropython/micropython/blob/17d82344581ad3a76033fae54c5d3304e17f185f/extmod/modframebuf.c#L317-L319

If you look at the underlying buffers, you have something like the following:
```
....xxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
```
where `.` are values in the underlying buffer that are skipped by the memoryview, `x` are values that should be visible in the framebuffer, and `s` are values which are skipped by the stride of 8.

It appears that what the `FrameBuffer` code is expecting is the following:
```
....xxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssssxxxx
ssss
```
even though the last 4 values are never written to.

## Work-around

This can be worked around by allocating extra memory to the underlying buffer, although that's fiddly for the first use-case if the data is loaded from an existing sprite sheet buffer, eg. loaded from a file.

### Code of Conduct

Yes, I agree

Gadgetoid, 2024-10-01:
Insofar as I can tell, you're effectively asking for FrameBuf to allow a buffer that's `(stride - width) * bytes_per_pixel` shorter than it would ordinarily be?

I think it's reasonable to claim this is a bug in the calculation, albeit something of an edge case. Since the stride represents the pixels *skipped* and not pixels written to then there is no risk of running off the end of the buffer.

The docs suggest that stride _"defaults to width but may need adjustments when implementing a FrameBuffer within another larger FrameBuffer or screen"_, so this is clearly an intended use case.

@jimmo you added this validation, what do you think?

corranwebster, 2024-10-01:
Yes, I think that the fact I was using a memoryview is perhaps orthogonal, other than perhaps not having control over the allocation of the original buffer.  I haven't tried but I expect that this would also fail with a regular array that is short the last run of skipped values.

Things are a bit more complex for pixel formats which are less than one byte per pixel.  I _think_ that the array can be short by up to ```floor((stride - width) * bytes_per_pixel))``` in those cases, because you need those few extra bits to pad out the byte.

smurfix, 2024-10-04:
I *think* that changing the test to this
```
if (((height_required - 1) * stride + width) * bpp / 8 > bufinfo.len) 
```
might fix the problem.

dpgeorge, 2024-10-22:
Fixed by d1574de3b6a050ef07c39a131203305f731d39d8

closed 2024-10-22 by dpgeorge