micropython/micropython #15963: mpremote: Behaviour is confusing with multiple mountpoints

peterhinch, 2024-10-04:
### Description

Follow on from https://github.com/micropython/micropython/pull/11777#issuecomment-2390338451.

To the naive user of a system with multiple mounted filesystems the behaviour of
```
$ mpremote cp foo.py :  
```
seems nondeterministic, because the destination of the copy can only be established with knowledge of the state of the target (its current directory). The same applies to
```
$ mpremote ls : 
```
The following change would provide a deterministic and more discoverable interface.
1. Where there is only one mounted filesystem, behaviour remains exactly as currently.
2. Where there are multiple filesystems behaviour is as follows:

`mpremote ls :` produces a list of mountpoints (as per `ls : /`).  
`mpremote cp foo.py :` throws an exception because you cannot write to '/'.   
`mpremote cp foo.py :/sd/` works as at present.

While this is a breaking change, an inability to write to '/' will not surprise anyone familiar with the MP directory structure.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

dpgeorge, 2024-10-30:
Thanks for raising this issue.  I agree that there's an issue to solve here, to make it clearer when there are multiple mount points and you don't start at the root.  And you have made a clear case for the change.

It being a breaking change is the biggest issue to consider.

But also there are some subtleties.  For example:
```
$ mpremote exec "import os;os.chdir('lib')" ls :
```
That will currently chdir to `/lib` (or `/flash/lib` on stm32 boards) and then the `ls :` will list the lib directory.

How would that behave with your proposed change?

----

Here are some ideas for alternatives:
1. For those ports like stm32 that mount the flash at `/flash`, change them to mount it at the root `/`.  Then all ports are consistent and the root is writable.  That's quite a big change but maybe the right way to do it, and can be done as part of MicroPython 2.0.
2. Make it so that ls (and maybe all filesystem operations) show the current directory first?  Eg:

```
$ mpremote ls
ls :/flash
         454 boot.py
          34 main.py
```

andrewleech, 2024-10-30:
I think they are both very good suggestions myself. Agree that moving /flash to / is a breaking change that will need to be managed but in the name of consistency across ports it feels like the correct thing to move towards. 

The improvement to `ls` looks really nice in the suggestion there, I think that's a great improvement idea!

peterhinch, 2024-10-30:
In my suggestion `ls`, `cp` etc. would require absolute host paths, and hence be independent of the host's state. The dependence on host state leads to behaviour which, to the uninitiated, seems broken. It also needs care when scripting `mpremote`.

> How would that behave with your proposed change?

The behaviour of `ls :` would be independent of current directory.

There is a case for a non-writable root, as per `rshell`. The user explicitly specifies an unambiguous destination. (e.g. `/pyboard/`, '/flash/' or '/sd/').

dpgeorge, 2024-10-31:
> In my suggestion `ls`, `cp` etc. would require absolute host paths

OK, that's clear.  So essentially all filenames that reference a remote board/device are absolute filenames.

Then something like `mpremote edit :main.py` will also no longer work on pyboard, but must instead be `mpremote edit :/flash/main.py`

andrewleech, 2024-10-31:
I for one do like to be able to support relative paths as well as absolute... though I'm not sure how best to provide appropriate visibility of this to users. To support relative better it'd be handy to add `mpremote fs cd <path>` though I realise that's counter to your suggestions @peterhinch 

Maybe as an interim fix to improve consistency '~' support (aka home) could be added which points to the default boot directory, aka `/` or `/flash` depending on port? Then at least `mpremote edit :~/main.py` would always work and be unambigious?

peterhinch, 2024-10-31:
@andrewleech That's a good idea. Shortcuts to points in the filesystem don't conflict with my suggestion so why not add another symbol that maps onto the host's current working directory? Then you could do something like `mpremot edit :^/subdir/foo.py`.

robert-hh, 2024-10-31:
Why not a single dot instead of ^

Josverl, 2024-11-02:
I like the `.` for cwd in the MCU path as that is simple to learn, and does not change the behavior from the current state.

I think the actual issue is not in mpremote, but in that different ports use different default directories after clean boots, and for some depending on  the ability to mount an SD 
Id argue that to get a consistent behavior for mpremote, that Micropython  would need to be consistent first