micropython/micropython #15967: Access SDCARD with LVGL MicroPython 1.23.0

spienz, 2024-10-06:
### Port, board and/or hardware

ESP32 ESP32-S3-WROOM-1-N4R8

### MicroPython version

LVGL MicroPython 1.23.0 on 2024-10-06; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

https://www.elecrow.com/esp32-display-5-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html

### Reproduction

import machine, os, vfs, network, time, wlan_connect

sd = machine.SDCard(slot=2, width=1, sck=12, miso=13, mosi=11, cs=10, freq=20000000)
vfs.mount(sd, '/sd') # mount

### Expected behaviour

Mount sd sdard

### Observed behaviour

Traceback (most recent call last):
  File "main.py", line 7, in <module>
TypeError: extra keyword arguments given

### Additional Information

with old sdcard access version and old micropython version it was working
cs = machine.Pin(10, machine.Pin.OUT)
# Intialize SPI peripheral (start with 1 MHz)
spi = machine.SPI(2,
                  baudrate=100000,
                  polarity=0,
                  phase=0,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(12),
                  mosi=machine.Pin(11),
                  miso=machine.Pin(13))
# Initialize SD card
sd = sdcard.SDCard(spi, cs)

### Code of Conduct

Yes, I agree

andrewleech, 2024-10-06:
Hi, the error message you post refers to line 7 of your main.py, it's not entirely clear which command that refers to. Can you try to run each line manually on the repl to check for sure exactly which command is failing? 

For reference, the esp32 sdcard driver is defined here so this is the definite list of args it supports: https://github.com/micropython/micropython/blob/master/ports%2Fesp32%2Fmachine_sdcard.c#L172

Keep in mind the firmware you're using is clearly a third party modified code by that vendor, so it's not certain exactly what they've changed so there's a limit to how much the community here can do. You should also reach out to the vendor for support in case the things that are breaking are from changes they've made.

spienz, 2024-10-06:
Hy Andrew,

with this code it was working.
But SDCARD isn't available in this build.

# Assign chip select (CS) pin (and start it high)
cs = machine.Pin(10, machine.Pin.OUT)
# Intialize SPI peripheral (start with 1 MHz)
spi = machine.SPI(2,
                  baudrate=100000,
                  polarity=0,
                  phase=0,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(12),
                  mosi=machine.Pin(11),
                  miso=machine.Pin(13))
# Initialize SD card
sd = sdcard.SDCard(spi, cs)

# OR this simpler initialization code should works on Maker Pi Pico too...
#sd = sdcard.SDCard(machine.SPI(1), machine.Pin(15))

os.mount(sd, '/sd')

so i tried it with that, in REPL i became the error:
I tried almost every combination

>>> sd = machine.SDCard(slot=2, width=1, sck=12, miso=13, mosi=11, cs=10, freq=20000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given

The main Reason is that elecrow wan't use the suggested wiring

https://docs.micropython.org/en/latest/library/machine.SDCard.html

The Micropython is fresh compiled using 

https://github.com/lvgl-micropython/lvgl_micropython/tree/main

I tried other componets like Wifi and it is working with that

Regards 

Sebastian

andrewleech, 2024-10-10:
On official micropython I just ran the following on repl:
``` python
import machine
sd = machine.SDCard(slot=2, width=1, sck=12, miso=13, mosi=11, cs=10, freq=20000000)
```
and it ran without throwing an error. 
The problem might be related to some of the patches being done in the lvgl_micropython repo, you might have better luck raising this issue over there?

jonnor, 2025-03-03:
Not reproducible.

closed 2025-03-03 by jonnor