micropython/micropython #15984: build error undeclared 'USB_SERIAL_JTAG_PACKET_SZ_BYTES

songzhishuo, 2024-10-09:
### Port, board and/or hardware

esp32c6

### MicroPython version

master 197becbdc

### Reproduction

 make -j

### Observed behaviour

/home/argon/Desktop/mpy/micropython/ports/esp32/usb_serial_jtag.c: In function 'usb_serial_jtag_tx_strn':
/home/argon/Desktop/mpy/micropython/ports/esp32/usb_serial_jtag.c:101:17: error: 'USB_SERIAL_JTAG_PACKET_SZ_BYTES' undeclared (first use in this function); did you mean 'USB_SERIAL_JTAG_BDATA_BITS'?
  101 |         if (l > USB_SERIAL_JTAG_PACKET_SZ_BYTES) {
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 USB_SERIAL_JTAG_BDATA_BITS
/home/argon/Desktop/mpy/micropython/ports/esp32/usb_serial_jtag.c:101:17: note: each undeclared identifier is reported only once for each function it appears in
[ 96%] Building C object esp-idf/main_esp32c6/CMakeFiles/__idf_main_esp32c6.dir/__/machine_touchpad.c.obj
[ 96%] Building C object esp-idf/main_esp32c6/CMakeFiles/__idf_main_esp32c6.dir/__/machine_pin.c.obj
make[3]: *** [esp-idf/main_esp32c6/CMakeFiles/__idf_main_esp32c6.dir/build.make:3297：esp-idf/main_esp32c6/CMakeFiles/__idf_main_esp32c6.dir/__/usb_serial_jtag.c.obj] 错误 1
make[3]: *** 正在等待未完成的任务....
[ 96%] Building C object esp-idf/main_esp32c6/CMakeFiles/__idf_main_esp32c6.dir/__/machine_dac.c.obj
make[3]: 离开目录“/home/argon/Desktop/mpy/micropython/ports/esp32/build-ESP32_GENERIC_C6”
make[2]: *** [CMakeFiles/Makefile2:4546：esp-idf/main_esp32c6/CMakeFiles/__idf_main_esp32c6.dir/all] 错误 2
make[2]: 离开目录“/home/argon/Desktop/mpy/micropython/ports/esp32/build-ESP32_GENERIC_C6”
make[1]: *** [Makefile:136：all] 错误 2
make[1]: 离开目录“/home/argon/Desktop/mpy/micropython/ports/esp32/build-ESP32_GENERIC_C6”
make failed with exit code 2, output of the command is in the /home/argon/Desktop/mpy/micropython/ports/esp32/build-ESP32_GENERIC_C6/log/idf_py_stderr_output_55067 and /home/argon/Desktop/mpy/micropython/ports/esp32/build-ESP32_GENERIC_C6/log/idf_py_stdout_output_55067
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-10-09:
`make` by itself will build the `ESP32_GENERIC` board, which is just esp32, not the C6.

So you must be doing something else to build a C6 board.

And please make sure you are using IDF 5.2.2.

songzhishuo, 2024-10-11:
thank you dpgeorge, I made a little mistake, my idf version is 5.0, Idf 5.2.2 make is ok

closed 2024-10-11 by songzhishuo