micropython/micropython #15995: docs: websocket

LaySoft, 2024-10-10:
### Documentation URL

https://docs.micropython.org/

### Description

Where is the built in websocket module documentation?

### Code of Conduct

Yes, I agree

jonnor, 2024-10-11:
Missing it seems! 
The code can be found at https://github.com/micropython/micropython/blob/master/extmod/modwebsocket.c
and there are some tests for it here https://github.com/micropython/micropython/blob/master/tests/extmod/websocket_basic.py

I believe that contributions that would provide documentation for this would be welcomed. The documentation setup is under docs/ in the main repository.

LaySoft, 2024-10-12:
The code [you link](https://github.com/micropython/micropython/blob/master/extmod/modwebsocket.c) to may have something to do with websocket, but it's certainly not a complete, usable implementation.

A websocket implementation must contain some form of strings like "Sec-WebSocket" or "258EAFA5-E914-47DA-95CA-C5AB0DC85B11" and the code above doesn't have any.

Other implementations, e.g. [nodemcu](https://github.com/nodemcu/nodemcu-firmware/blob/release/app/websocket/websocketclient.c), swoole contain them, they implement a real websocket implementation, they work, I use them effectively.

So I don't even understand why this "something" is called a websocket...

jonnor, 2024-10-12:
The module is used as part of WebRepl (and was written for it). So there you will find the strings you refer to. https://github.com/micropython/micropython-lib/blob/394cbfc98a333dd1d4db35fb69379c72c30337f3/micropython/net/webrepl/webrepl.py#L9

It is probably one of the few usage examples for this module. Note that it is a WebSocket *server* implementation. For a Websocket client there are other options, such as in aiohttp_ws: https://github.com/micropython/micropython-lib/blob/394cbfc98a333dd1d4db35fb69379c72c30337f3/python-ecosys/aiohttp/aiohttp/aiohttp_ws.py#L51

There is also a WebSocket server implementation in MicroDot, as far as I understand, https://microdot.readthedocs.io/en/latest/index.html

LaySoft, 2024-10-13:
Thanks for the clarification, now I understand that the built-in websocket implementation is server. I need a client. I could get started with the aiohttp_ws module you mentioned, my first attempts were successful.