micropython/micropython #15996: REPL with telnet

LaySoft, 2024-10-10:
### Port, board and/or hardware

ESP32-C3

### MicroPython version

v1.23.0
https://micropython.org/resources/firmware/ESP32_GENERIC_C3-20240602-v1.23.0.bin

### Reproduction

The documentation says, i need to simple telnet to the device ip address.
https://docs.micropython.org/en/v1.9.3/wipy/wipy/tutorial/repl.html
But the telnet client does not connect to the port 23.

### Observed behaviour

Telnet connection on the port 23.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

andrewleech, 2024-10-11:
That docs link is for an obsolete version (1.9.4) running on a particular port/board called wipy. Current micropython / other ports don't have telnet built in by default.

The standard way of getting to the repl is USB/serial or via webrepl on boards with it enabled (esp32 probably does)

Alternatively you could try installing a telnet server like https://github.com/cpopp/MicroTelnetServer and try that out (I haven't tried that myself)

LaySoft, 2024-10-12:
I tried the telnet server you recommended, it starts, I can connect to it via port 23, but I can't type, the keystrokes don't appear. However, if I do something on another serial terminal, it shows up here on telnet.

If I press ctrl + ], I get a "telnet>" prompt, I can type, but any command I issue "?Invalid command" is the response, and then I can't type again, I have to press ctrl + ] again.

It's all so confusing, surely this is how it's supposed to work, or am I messing something up?

andrewleech, 2024-10-12:
I don't know, as I said I've never used that server. 

If I want repl over network the official method is webrepl

robert-hh, 2024-10-12:
Telnet as documented works with the CC3200 port. It was never implemented for other ports.. The utelnet.py script used to work on some ports like esp8266, but is not officially supported and not maintained by the author. So I'm closing this issue.

closed 2024-10-12 by robert-hh