micropython/micropython #16008: Using ntptime.settime() while _thread is running can break sleep timing

GM-Script-Writer-62850, 2024-10-14:
### Port, board and/or hardware

Raspberry Pi PICO W

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

### Reproduction

- _thread_work() - measure how long time.sleep takes
- wifi() - connect to wifi and reconnect if it does down
  -  wifi_auth just has the wifi login variables
- setTime() - this sets the clock using ntptime
  - this was unreliable in micropython 1.2, hence the try blocks
- wait4clock() - just some busy work for asyncio, can be used to start process that depend on the clock being set
```
import _thread
import ntptime
import asyncio
import time

sleep=asyncio.sleep_ms
#ntptime.host="10.0.0.69"
wlan = 0

def _thread_work():
	while 1:
		then=time.ticks_ms()
		time.sleep(1)
		then=time.ticks_diff(time.ticks_ms(),then)
		print("Core 1:",then)

async def wifi():
	from wifi_auth import ssid, password
	import network

	global wlan
	wlan = network.WLAN(network.STA_IF)
	wlan.active(True)
	wlan.config(pm = 0xa11140)# Power management is very very bad, ping time is around 1000x worse and packet loss insane
	wlan.connect(ssid, password)

	while 1:
		while 1:
			wstat=wlan.status()
			if wstat < 0 or wstat >= 3:
				break
			print('Waiting for WiFi connection...')
			await sleep(1000)

		if wlan.status() == 3:
			status = wlan.ifconfig()
			print('Wifi Connected; ip =',status[0])
			while wlan.isconnected():
				await sleep(30000)
			print('WiFi Down')
		else:
			print("Failed to connect to wifi; retry in 30 seconds")
			await sleep(30000)
		wlan.connect(ssid, password)
#END wifi()

async def setTime():
	#ntptime.settime() failure: [Errno 110] ETIMEDOUT
	#ntptime.settime() failure: overflow converting long int to machine word
	print('Attempt to set time')
	success=0
	try:
		print("calling ntptime.settime()")
		ntptime.settime()
		success=1
	except OverflowError:
		# it is not going to work
		print("overflow error; settime is borked")
	except OSError:
		print("ntptime.settime() failure")
	if not success:
		await sleep(10000)
		asyncio.create_task(setTime())
	else:
		print("END of setTime() process")
#END setTime()

async def wait4clock():
	delay=2000
	while time.time() < 1640995200: # Wait for clock to set or a 1 year...)
		print("Waiting for clock...")
		await sleep(3000)
	print('Clock set:',time.time())
	#_thread.start_new_thread(_thread_work,())
#END wait4clock()

async def main():
	asyncio.create_task(wait4clock())
	asyncio.create_task(wifi())
	while not wlan or not wlan.isconnected():
		then=time.ticks_ms()
		await sleep(1000)
		then=time.ticks_diff(time.ticks_ms(),then)
		print("Core 0:",then,"Waiting for wifi")
	asyncio.create_task(setTime())
	while 1:
		then=time.ticks_ms()
		await sleep(1000)
		then=time.ticks_diff(time.ticks_ms(),then)
		print("Core 0:",then)

_thread.start_new_thread(_thread_work,())

asyncio.run(main())
```

### Expected behaviour

Calls to `time.sleep(1)` and `asyncio.sleep_ms(1000)` should be near 1 second long

### Observed behaviour

output of example code:
```
Waiting for clock...
Waiting for WiFi connection...
Core 1: 1000
Core 0: 1000 Waiting for wifi
Waiting for WiFi connection...
Core 1: 1000
Core 0: 1001 Waiting for wifi
Waiting for WiFi connection...
Core 1: 1000
Waiting for clock...
Core 0: 1003 Waiting for wifi
Wifi Connected; ip = 10.0.0.190
Core 1: 1000
Core 0: 1001 Waiting for wifi
Attempt to set time
calling ntptime.settime()
END of setTime() process
Core 1: 22526
Core 0: 22522
Clock set: 1728875328
Core 1: 5324
Core 0: 5324
Core 1: 7473
Core 0: 7474
Core 1: 17208
Core 0: 17208
Core 1: 5319
Core 0: 5319
Core 1: 24678
Core 0: 24678
Core 1: 5325
Core 0: 5325
Core 1: 24678
Core 0: 24678
Core 1: 5330
Core 0: 5330
Core 1: 24673
Core 0: 24673
Core 1: 5325
Core 0: 5325
Core 1: 24684
Core 0: 24684
Core 1: 5319
Core 0: 5319
Core 1: 24682
Core 0: 24683
Core 1: 5324
Core 0: 5324
Core 1: 1939
Core 0: 1939
Core 1: 5734
Core 0: 5734
Core 1: 5836
Core 0: 5836
Core 1: 11162
Core 0: 11162
Core 1: 5329
Core 0: 5330
Core 1: 7059
Core 0: 7059
Core 1: 13925
Core 0: 13926
Core 1: 3687
Core 0: 3687
Core 1: 5325
Core 0: 5325
Core 1: 7269
Core 0: 7269
Core 1: 17413
Core 0: 17413
Core 1: 5325
Core 0: 5325
Core 1: 1018
Core 0: 1018
Core 1: 23655
Core 0: 23655
Core 1: 5324
Core 0: 5324
```

### Additional Information

- Note this does not have a 100% reproduction rate, _feels_ near 50%
- This is a regression since micropython 1.2 (rp2-pico-w-20230426-v1.20.0.u2f)
- Note that Thonny is set to **NOT** synchronize device's real time clock (Tools -> Options-> Interpreter)

### Code of Conduct

Yes, I agree

sosi-deadeye, 2024-10-14:
I wrote a little test to use it with Thonny and without ntptime.

```python
"""
Test works with Thonny, because the rtc is set to year 2000
and 5 seconds after to year 2024. ntptime is not involved.
"""
import asyncio
import machine
import math
import time
import sys

print(sys.version, sys.platform)

rtc = machine.RTC()
print("Welcome to the Year 2000", end="\n\n")
rtc.init((2000,1,1,0,0,0,0,0))
print("Unix-Time:", time.time(), end="\n\n")

async def time_travel(ev: asyncio.Event):
    await asyncio.sleep(10)
    print("\nTime travel to 2024")
    rtc.init((2024,1,1,0,0,0,0,0))
    print("Unix-Time:", time.time(), end="\n\n")
    await asyncio.sleep(10)
    ev.set()

async def ping(ev: asyncio.Event, statistics: list):
    rtc = machine.RTC()
    last = None
    
    while not ev.is_set():
        current = time.ticks_ms()
        
        if last is not None:
            delta_ms = time.ticks_diff(current, last)
            print("Duration:", delta_ms, "ms")
            statistics[rtc.datetime()[0]].append(delta_ms)

        last = current
        await asyncio.sleep(1)

async def main():
    stop_event = asyncio.Event()
    stats = {2000: [], 2024: []}
    asyncio.create_task(time_travel(stop_event))
    await ping(stop_event, stats)
    return stats

def show_stats(stats):
    def stdev(values):
        size = len(values)
        mean = sum(values) / size
        u = sum((mean - value) ** 2 for value in values) / (size - 1)
        return math.sqrt(u)
    
    for year, values in stats.items():
        print(f"{year}:", stdev(stats[year]))
    
    print()
    print("Raw-Data:", stats)

if __name__ == "__main__":
    stats = asyncio.run(main())
    print("\nStats:")
    show_stats(stats)
    
```

Output on ESP32-S3:

```
MPY: soft reboot
3.4.0; MicroPython v1.24.0-preview.409.g82e69df33 on 2024-10-10 esp32
Welcome to the Year 2000

Unix-Time: 0

Duration: 1009 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms

Time travel to 2024
Unix-Time: 757382400

Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms
Duration: 1010 ms

Stats:
2000: 0.3333333
2024: 0.0

Raw-Data: {2000: [1009, 1010, 1010, 1010, 1010, 1010, 1010, 1010, 1010], 2024: [1010, 1010, 1010, 1010, 1010, 1010, 1010, 1010, 1010, 1010]}
MicroPython v1.24.0-preview.409.g82e69df33 on 2024-10-10; Generic ESP32S3 module with ESP32S3
Type "help()" for more information.
```

I haven't a RP2040 here, so I could not test the code yet.
`asyncio` is written in Python and `asyncio.sleep_ms` uses `time.ticks_ms` and `time.ticks_diff`.
If `time.ticks` / `time.ticks_ms` is affected, this is a huge problem.

`ntptime.settime` uses the RTC to set the time: https://github.com/micropython/micropython-lib/blob/master/micropython/net/ntptime/ntptime.py#L68

AFAIK should `ticks` and `RTC` be independent of each other.

peterhinch, 2024-10-14:
Please see [my reply in Discussions](https://github.com/orgs/micropython/discussions/15927#discussioncomment-10933443). I suggest a substantial rewrite with the aim of not starting `asyncio` until system time is stable. The script should have this general starting procedure:
1. Connect to WiFi with synchronous code.
2. Set system time with synchronous code - see [these notes](https://github.com/peterhinch/micropython-samples/blob/master/README.md#414-ntp-time) on error conditions.
3. Start the thread.
4. Start asyncio. Run a task to maintain WiFi connection in case of outages and a task to measure `await asyncio.sleep(1)`.

GM-Script-Writer-62850, 2024-10-14:
> I wrote a little test to use it with Thonny and without ntptime.

i patched this to work with my pico and added `_thread` and was unable to replicate it
```
"""
Test works with Thonny, because the rtc is set to year 2021
and 5 seconds after to year 2024. ntptime is not involved.
"""
import asyncio
import machine
import math
import time
import sys
import _thread

print(sys.version, sys.platform)

rtc = machine.RTC()
print("Welcome to the Year 2021", end="\n\n")
rtc.datetime((2021, 1, 1, 4, 0, 0, 19, 0))
print("Unix-Time:", time.time(), end="\n\n")

async def time_travel(ev: asyncio.Event):
    await asyncio.sleep(10)
    print("\nTime travel to 2024")
    rtc.datetime((2024,1,1,0,0,0,0,0))
    print("Unix-Time:", time.time(), end="\n\n")
    await asyncio.sleep(10)
    ev.set()

async def ping(ev: asyncio.Event, statistics: list):
    rtc = machine.RTC()
    last = None
    
    while not ev.is_set():
        current = time.ticks_ms()
        
        if last is not None:
            delta_ms = time.ticks_diff(current, last)
            print("Duration:", delta_ms, "ms")
            statistics[rtc.datetime()[0]].append(delta_ms)

        last = current
        await asyncio.sleep(1)

async def main():
    stop_event = asyncio.Event()
    stats = {2021: [], 2024: []}
    asyncio.create_task(time_travel(stop_event))
    await ping(stop_event, stats)
    return stats

def show_stats(stats):
    def stdev(values):
        size = len(values)
        mean = sum(values) / size
        u = sum((mean - value) ** 2 for value in values) / (size - 1)
        return math.sqrt(u)
    
    for year, values in stats.items():
        print(f"{year}:", stdev(stats[year]))
    
    print()
    print("Raw-Data:", stats)

def _thread_work():
    i=0
    while i < 20:
        then=time.ticks_ms()
        time.sleep(1)
        then=time.ticks_diff(time.ticks_ms(),then)
        print("Core 1:",then)
        i+=1

if __name__ == "__main__":
    _thread.start_new_thread(_thread_work,())
    stats = asyncio.run(main())
    print("\nStats:")
    show_stats(stats)

```

> Please see [my reply in Discussions](https://github.com/orgs/micropython/discussions/15927#discussioncomment-10933443). I suggest a substantial rewrite with the aim of not starting `asyncio` until system time is stable. The script should have this general starting procedure:

if setting the clock while using asyncio is the issue why am i unable to replicate it without using `_thread`? furthermore why does not not happen with micropython 1.20